realsenseai/realsense-ros #1830: Build realsense_ros using catkin build command

NishanthARao, 2021-04-29:
I have to install the ```realsense_ros``` package in a workspace that is already built using the command ```catkin_build```. I cannot install ```realsense_ros``` package in another workspace. How do I perform the operations ```catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release && catkin_make install``` using ```catkin_build``` command? 

I tried ```catkin build --catkin-make-args -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release``` but I am not sure if it builds properly. 

Can someone please help me out? Thanks.

MartyG-RealSense, 2021-04-29:
Hi @NishanthARao  My understanding is that you could clone realsense-ros to a workspace folder of your own choosing instead of creating a new 'catkin_ws/src/' folder.  A custom location makes existing installation guide instructions harder to follow though, as **cd** directory-change instructions will correspond to the recommended 'catkin_ws/src/' setup process.  And so if you are using a custom path and follow a guide's directory-change instructions exactly then you could end up in the wrong directory on your custom path when inputting catkin commands.  So that is something that you will need to be careful about to ensure that you are in the correct directory for inputting a particular command.

If you need to use **catkin_build** instead of catkin_make and do so on a custom path then the discussion in the link below may be helpful to you.

https://github.com/IntelRealSense/realsense-ros/issues/1432

NishanthARao, 2021-05-01:
@MartyG-RealSense I see. The discussion that you linked to, was very helpful. Thanks a lot. It would also be better if the procedure to build using ```catkin build``` command was put up in the documentation.

closed 2021-05-01 by NishanthARao