realsenseai/realsense-ros #1831: colcon build error

wuhaoqi-1, 2021-04-29:
[image]
`In file included from /home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/include/realsense2_camera/realsense_node_factory.h:10:0,
                 from /home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/src/realsense_node_factory.cpp:4:
/home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/include/realsense2_camera/base_realsense_node.h:13:10: fatal error: diagnostic_updater/diagnostic_updater.hpp: No such file or directory
 #include <diagnostic_updater/diagnostic_updater.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:0:
/home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/include/realsense2_camera/base_realsense_node.h:13:10: fatal error: diagnostic_updater/diagnostic_updater.hpp: No such file or directory
 #include <diagnostic_updater/diagnostic_updater.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
In file included from /home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/include/realsense2_camera/t265_realsense_node.h:3:0,
                 from /home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/home/jetson-nx/ros2_ws/src/realsense-ros/realsense2_camera/include/realsense2_camera/base_realsense_node.h:13:10: fatal error: diagnostic_updater/diagnostic_updater.hpp: No such file or directory
 #include <diagnostic_updater/diagnostic_updater.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o] Error 1
make[1]: *** [CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< realsense2_camera [1.10s, exited with code 2]

Summary: 2 packages finished [4.49s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
`

RealSenseSupport, 2022-02-07:
@wuhaoqi-1  What's your ROS2 distribution? Please install diagnostic-updater using "sudo apt-get install ros-$ROS_DISTRO-diagnostic-updater" and colcon build later.

HainingLuo, 2022-03-28:
Hi, could I ask if this issue has been resolved? I got same error just now.

I am using ROS Melodic and Dashing together and I have installed ros-dashing-diagnostic-updater. However, it did not solve the issue.

My installation process was not exactly the same as instructed since I have ROS melodic on the side. Instead of doing `sudo apt-get install python3-rosdep -y`, I did `sudo apt-get install python-rosdep -y`, and the dependency installation ended successfully. Could this be the reason why I am getting this error?

Many thanks!

ShaoxiongYao, 2023-03-08:
I tried `sudo apt-get install ros-foxy-diagnostic-updater` (I am using `foxy`) and the problem is resolved.

vickhung, 2025-07-23:
im having the same issue with humble. on jetson orin nx. is this issue covered yet?

MartyG-RealSense, 2025-07-23:
Hi @vickhung  Have you tried installing the diagnostic updater for Humble with the command `sudo apt-get install ros-humble-diagnostic-updater` and then performing colcon build afterwards, please?

closed 2026-10-23 by MartyG-RealSense