realsenseai/realsense-ros #1834: Pointcloud with no texture?

JeremieBourque1, 2021-04-29:
Hi,

Is it possible for the d455 to publish the pointcloud without any texture (just XYZ) in order to cut down on the size of the cloud?

Thanks

MartyG-RealSense, 2021-04-29:
Hi @JeremieBourque1   My understanding is that a texture-free point cloud can be generated by editing the launch file arg **pointcloud_texture_stream** to be defined as **RS2_STREAM_ANY** instead of its default of RS2_STREAM_COLOR

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L46

If you do not want to edit the launch file then you could add the instruction below to the end of your roslaunch instruction:

**pointcloud_texture_stream:=RS2_STREAM_ANY**

For example:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY**

JeremieBourque1, 2021-04-29:
I tried it and using RS2_STREAM_ANY replaces the `rgb` field by the `intensity` field but the bandwith used by the topic is the same for both so I am not getting the size reduction I was hoping for

MartyG-RealSense, 2021-04-29:
If you do not need color, do you see improvement in overall bandwidth usage if you add **enable_color:=false** to the roslaunch instruction to not publish the color topic?

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_ANY enable_color:=false**

JeremieBourque1, 2021-04-29:
Disabling color didn't have an impact on the bandwith

MartyG-RealSense, 2021-04-30:
Is reducing the resolution or FPS speed of the depth topic an option for you, please?

doronhi, 2021-05-02:
Hi @JeremieBourque1 
Setting the flag **pointcloud_texture_stream:=RS2_STREAM_ANY** as @MartyG-RealSense mentions should have reduced the BW. There is a bug regarding this and I just push #1842 to fix it.
However, when testing you should consider the fact that by default, a pointcloud with color texture covers only the FOV (field-of-view) where color points are available which is smaller than the FOV of the depth sensor. So by requesting a pointcloud without texture you get a pointcloud that covers more space and therefore may be larger than the one with color texture.
A valid comparison will be to also add the flag **allow_no_texture_points:=true** - then you will compare the same number of points and see the true difference in the BW (That is, after applying #1842).
Also, I would like to point out that the 3 coordinate values, x, y, z are allocated with 4 bytes each. The texture is allocated with 4 bytes per pixel. The expected reduction in size is therefore around 25%.

JeremieBourque1, 2021-05-03:
I tried with the latest commit and with `allow_no_texture_points:=true` but I am actually getting a higher BW when using `RS2_STREAM_ANY`.
With `RS2_STREAM_COLOR` I am getting around 200 MB/s while with `RS2_STREAM_ANY` I am getting 225 MB/s.
I am using `rostopic bw` to measure the bandwith.

doronhi, 2021-05-05:
I wonder if that has something to do with a ceiling of bandwidth. Could you try with lower fps? Also, it may be easier to compare when using **ordered_pc:=true** as the size should remain constant between frames.

MartyG-RealSense, 2021-05-17:
Hi @mszuyx   Apologies for the late response.  Do you still require assistance with your question, please?  Thanks!

mszuyx, 2021-05-17:
Yes, can you simply make some comments on the following topics:

"using uncolored pointcloud will consume more CPU"

"how much would the "ordered_pc" option impact the processing speed?"

Thank you !!

MartyG-RealSense, 2021-05-17:
1.  I could not find any references that backed up the claim in the provided link that using RS2_STREAM_ANY would result in a doubling of the number of point cloud points and hence an increase in CPU usage.

2.  I do not have information about how the **ordered_pc** setting in the RealSense ROS wrapper impacts processing speed, so I will refer that to @doronhi

doronhi, 2021-05-18:
Based on [the D435 spec](https://www.intelrealsense.com/wp-content/uploads/2020/06/Intel-RealSense-D400-Series-Datasheet-June-2020.pdf) the Depth FOV HD is 87x58 degrees (page 55), and the Color FOV is 69x42 degrees. The depth image covers an area approximately 1.74 times larger than the color image. 
The pointcloud is built around the depth image. When we use the **RS2_STREAM_COLOR** flag, we force the pointcloud to consist of only 1/1.74 of the points we have in the depth image.

Pointing a D435 camera such that the pointcloud seems to be pretty filled, I see the followings:
with **RS2_STREAM_COLOR**: around 196018 points, each takes 20 bytes -> message size: ~3920360
with **RS2_STREAM_ANY**: around 386548 points of 16 bytes -> message size: ~6184768
The number of points we have when no color is needed is about twice the number when color is not. Considering that all those points are being copied one by one to the pointcloud2 message, a matching increase in CPU usage can be expected.
The size rate, in this case, is 6184768 / 3920360 = 1.57, meaning that without color we get a message size around 1.5 times larger than with colors. In other scenarios, the numbers will vary but doubling the number of points sometimes seems possible.

When using **ordered_pc** the size of the message is fixed. In the previous example, I used a depth resolution of 848x480 yielding an image of 407040 points. Although many of them have invalid values that is still the number of points in the pointcloud2 message. In other scenarios, where we are not looking at an almost completely filled image, the ordered pointcloud will still be of size 407040 points while the unordered pointcloud will contain much less. All the points are still copied to the pointcloud2 message and therefore, the processing speed, in the publisher, and maybe in the receiver end as well, as the used BW will behave accordingly.

closed 2021-05-27 by JeremieBourque1