realsenseai/realsense-ros #1836: How to display a ros bag

JohnVorwald, 2021-04-29:
A 435i camera was used to take video during a test.  The camera was calibrated and the  data was recorded.  I receive the error Failed to create ros reader: Invalid file format, file does not contain topic "/file_version" nor "/FILE_VERSION".

MartyG-RealSense, 2021-04-30:
Hi @JohnVorwald   Did you record the bag in the RealSense SDK using a tool such as the **RealSense Viewer** and then try to play back the bag in ROS?  Or record in ROS and play back in the Viewer?  

If so, there are differences in the RealSense SDK's bag file-format compared to the ROS rosbag format that may cause compatibility issues.

https://github.com/IntelRealSense/librealsense/issues/3020

 If you are using ROS then the **rosbag record** ROS tool can be used to record camera data in a pure ROS rosbag.

http://wiki.ros.org/rosbag/Commandline#rosbag_record

JohnVorwald, 2021-05-04:
Hi @MartyG-RealSense 

The bag was recorded with a python code.  I am trying to generate the 3d point cloud.  

rosbag info XYZ.bag
path:        ...
version:     2.0
duration:    1:47s (107s)
start:       ...
end:        ...
size:        ...
messages:    ...
compression: ...
types:       geometry_msgs/Point [4a842b65f413084dc2b10fb484ea7f17]
             sensor_msgs/Image   [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu     [6a62c6daae103f4ff57a132d6f95cec2]
topics:      ...
             /realsense/cam0_image      3225 msgs    : sensor_msgs/Image  
             /realsense/cam1_image      3225 msgs    : sensor_msgs/Image  
             /realsense/depth_image     3226 msgs    : sensor_msgs/Image  
             /realsense/imu/data_raw    3224 msgs    : sensor_msgs/Imu

I also have the files generated by the real sense camera calibration.  Can you provide guidance on reading the existing ros bag images and converting to 3d point cloud, using information in the camera calibration files.

MartyG-RealSense, 2021-05-05:
A more direct approach to generating a point cloud from a bag may be to use the SDK Python example program **opencv_pointcloud_viewer.py** and edit it to use a bag as its camera data source instead of a live camera.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

The **read_bag_example.py** example program that you mentioned above provides a reference for what a Python equivalent use of the bag-accessing code looks like.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py#L41-L49

The highlighted code could potentially be adapted for **opencv_pointcloud_viewer.py**, replacing its live stream setup with the bag-reading code based around ** rs.config.enable_device_from_file**

JohnVorwald, 2021-05-06:
Hi @MartyG-RealSense 

When I run read_bag_example.py  

python3 read_bag_example.py -i 12in-05hz.bag

I get this ouput

Traceback (most recent call last):
  File "read_bag_example.py", line 49, in <module>
    pipeline.start(config)
RuntimeError: Failed to resolve request. Request to enable_device_from_file("12in-05hz.bag") was invalid, Reason: Failed to create ros reader: Error reading from file: wanted 4 bytes, read 0 bytes

Here's the top of the bag file

john@john-desktop:~/projects/python$ head 12in-05hz.bag 
#ROSBAG V2.0

My "guess" is that the pipeline wants to read a topic, or information that's not present in the ros bag file.

MartyG-RealSense, 2021-05-06:
@JohnVorwald   You are correct that **Failed to resolve request** means that the SDK could not support or could not understand a configuration that was requested in the script.

By default, line 46 of read_bag_example.py expects a bag with a depth stream with 30 FPS speed.  The notes on this line remind to alter the script depending on the stream contents of the specific bag file that the script is accessing.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py#L46

In the link below, a RealSense user adapted the script to access a **color** stream of resolution 1280x720 at 6 FPS.

https://stackoverflow.com/questions/66151659/intel-realsense-unable-to-read-bag-file

I should highlight to D455 camera users reading this in future that 5 FPS should be used as the minimum FPS for the D455 as it does not support 6 FPS.

JohnVorwald, 2021-05-06:
@MartyG-RealSense 
I have other sensors in addition to the realsense camera.  Can you provide a sample ros python code to record realsense and instrumentation data, assume the instrument data is either a single value like temperature or a 3 value geometry point (x, y, z).  And then a corresponding playback python script.  I believe the realsense images are stored at 30 hz, I'm not sure about the other sensors, but there two or three recordings between images.  Also, can you provide more detail on activating the higher resolution, 6 hz, image mode.

MartyG-RealSense, 2021-05-06:
I would think that if you are recording multiple types of device and each of them have a ROS topic associated with them then an appropriate solution would be to use **rosbag record -all** to record all topics that are currently being published even if in multiple ROS terminals.

http://wiki.ros.org/rosbag/Commandline#rosbag_record
http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data

You likely could not playback such a bag in librealsense applications though as the SDK's bag format is designed only for camera data.

https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

If the bag is recorded in ROS then you should be able to play its topics back in ROS with **rosbag play** though.

http://wiki.ros.org/rosbag/Commandline#rosbag_play

In regard to the 6 Hz comment, do you mean that you wish to launch with a custom color stream in ROS with 1280x720 resolution and 6 FPS please?  If so then since ROS wrapper version 2.2.22 onwards you would need to define a roslaunch instruction that provides three factors - color stream width, color stream height and color stream FPS.  

https://github.com/IntelRealSense/realsense-ros/issues/1751#issuecomment-798834284

JohnVorwald, 2021-05-06:
My question regarding 6 Hz is what resolutions does realsense 435i support, and how to activate a different resolution, for the next test sequence.  Regarding the ros bag record / play, what you're proposing is what we did.  So now I have a several ros bags of images and other measurements data and also I have camera calibration files.  How do I use the camera calibration files to calculate the point cloud from the stereo images.  I can loop over the bag topics, so I have cam0 and cam1 numpy image matrix pairs.  How can I use realsense library to process image pairs and output the pointcloud.  The point cloud is needed, along with the other measured data, to calculate the desired results.  The purpose of the test was to show that we could use the point cloud, but I'm not able to calculate the point cloud.

http://wiki.ros.org/rosbag/Commandline#rosbag_record
http://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data

IntelRealSense/librealsense#3020 (comment)<https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766>

http://wiki.ros.org/rosbag/Commandline#rosbag_play

#1751 (comment)<https://github.com/IntelRealSense/realsense-ros/issues/1751#issuecomment-798834284>

MartyG-RealSense, 2021-05-07:
Pages 63 and 64 of the current edition of the PDF data sheet document for the 400 Series cameras list the resolution and FPS combinations that are supported by D435i.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

In regard to setting different resolutions for different test sequences, you could design a range of alternative roslaunch instructions to use that each have a different stream configuration, or create a set of custom launch files.  I am not aware of a method to change the resolution dynamically after launch during runtime though.

Your question above about using numpy image matrix pairs sourced from a bag to generate a point cloud is outside of my ROS knowledge.  I will refer it to @doronhi the RealSense ROS wrapper developer to see whether he has suggestions about what you want to achieve

JohnVorwald, 2021-05-07:
Thank you for the reference.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

closed 2021-05-18 by MartyG-RealSense