realsenseai/realsense-ros #1837: How to change origin position of point cloud in rviz?

zeldamax, 2021-04-30:
My Intel RealSense D435i is mounted on a Denso Cobotta (on joint 5). When I visualize the point cloud, it starts projecting outwards from xyz = (0, 0, 0), which is not the real position of the RealSense camera. So I want to change the origin position of the point cloud so that it matches the position of the camera mounted on the robot. I've attached two images for visualization. If the robot gets moved by MoveIt, the position of the camera changes. That means that the origin position of the point cloud will have to adjust. Can I achieve this by using the ros tf package and if so how? Thank you

[image]
[image]

zeldamax, 2021-04-30:
Ok thank you very much for your explanation. I now know that the depth origin corresponds to the center of the left infrared sensor. I also just saw this in the Intel RealSense Viewer Application. But is there any way I can parent the depth origin in rviz to another object so it no longer is at 0,0,0 but at a different position instead?

MartyG-RealSense, 2021-04-30:
I'm not certain about this question, though the reference at the link below may be helpful.

http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch#Origins

doronhi, 2021-05-02:
It seems like you have a frame called "camera position". You need to link it to the camera_link frame of the camera.
Look for the `static_transform_publisher` command in [rs_d400_and_t265.launch](https://github.com/IntelRealSense/realsense-ros/blob/7d8fa6eeab9c9eb1e5c818dcb2065c9d9e64639c/realsense2_camera/launch/rs_d400_and_t265.launch#L49) for reference.

<node pkg="tf" type="static_transform_publisher" name="t265_to_d400" args="0 0 0 0 0 0 /$(arg tf_prefix_camera1)_link /$(arg tf_prefix_camera2)_link 100"/>

zeldamax, 2021-05-10:
I now managed to create a launch file which uses the `static_transform_publisher` to link the camera_link to the fifth joint of the robot:
`
<launch>
   <include file="$(find denso_robot_bringup)/launch/cobotta_bringup.launch">
    <arg name="sim" value="false"/>
    <arg name="ip_address" value="192.168.178.222"/>
  </include>
  <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
    
  </include>
  <node pkg="tf2_ros" type="static_transform_publisher" name="camera_mount" args="0 0 0 0 0 0 J5 camera_link"/>
</launch>
`
This seems to be working. Thanks a lot for your help. I am now facing a different sort of problem, the two packages seem to be using different time. While the Intel Realsense uses the real / ros time, the Denso Cobotta uses simulation / gazebo time. This means that I still can't see the point cloud at the desired position. I tried to solve this by changing the timing systems accordingly (e.g. using `<param name="/use_sim_time" value="false"/>` in the launch file, but haven't had any luck so far. I will post about it if I find a solution.

MartyG-RealSense, 2021-05-10:
If you wanted the RealSense ROS node to use Simulation Time instead of ROS time to match up to the simulation, I wonder if you would actually set **use_sim_time** to True in the RealSense ROS launch file, or set it by adding **use_sim_time:=true** to the end of the RealSense ROS roslaunch instruction.  for example:

**roslaunch realsense2_camera rs_rgbd.launch use_sim_time:=True**

MartyG-RealSense, 2021-05-16:
Hi @zeldamax  Do you require further assistance with this case, please?  Thanks!

zeldamax, 2021-05-18:
I managed to get it working now. The solution for simulation mode was exactly as you said @MartyG-RealSense. (by using `roslaunch realsense2_camera rs_rgbd.launch use_sim_time:=True`) Thank you for your support and @doronhi of course aswell. I also got it working for real mode using the following launch file:
```
<launch>
   <param name="/use_sim_time" value="false"/>
   <include file="$(find denso_robot_bringup)/launch/cobotta_bringup.launch">
    <arg name="sim" value="false"/>
    <arg name="ip_address" value="192.168.178.222"/>
    <arg name="send_format" value="0"/>
    <arg name="recv_format" value="2"/>
  </include>
  <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
    
  </include>
  <node pkg="tf2_ros" type="static_transform_publisher" name="camera_mount" args="-0.05 -0.04 0.027 0 -1.57079632679 0 J5 camera_link"/>
</launch>
```

closed 2021-05-18 by zeldamax

MartyG-RealSense, 2021-05-19:
Great news @zeldamax - thanks very much for the update and for sharing your code   :)