realsenseai/realsense-ros #1838: Plans to make realsense-ros node a lifecycle node? [ROS2]

anaelle-sw, 2021-04-30:
Hello,

in our use case, we regularly plug / un-plug cameras (D435) to our system. We need to adapt the number of realsense-ros nodes running according to the current number of embedded cameras, at runtime without re-launching our whole ROS application. 

Previously, with ROS1, we used to simply kill / launch the realsense-ros nodes with system calls.

ROS2 offers the possibility to use LifecycleNodes ([ROS2 design documentation here](https://design.ros2.org/articles/node_lifecycle.html) and [a demo here](https://github.com/ros2/demos/blob/foxy/lifecycle/README.rst)). We would be super interested in using realsense-ros nodes as lifecycle nodes, especially for two reasons:
- easily activate / deactivate the realsense-ros nodes at runtime (while our system is in some kind of a sleep mode)
- easily configure / unconfigure a realsense-ros node at runtime (depending of the number of cameras being added / removed to our system)

(By "activate / deactivate / configure / unconfigure" I am directly referring to the lifecycle node's states transitions (see doc link above), and by "easily", I mean that these transitions can be triggered by an external node of our own, which would be able to ask the lifecycle realsense-ros node to change from a state to another)

Well my question is: are you considering to make realsense-ros a ROS2 lifecycle node? If so, any idea of the time it will take until it would be implemented? 
If this is not part of your plans for now, would you still accept a pull request to implement this? Even if, obviously, there would be a lot of discussions to have about the design of it, before implementing anything. 
I am just asking to know if lifecycle nodes are something you could consider at some point, or if there is just no room for lifecycle nodes in realsense-ros development.

Thank you!

MartyG-RealSense, 2021-04-30:
Hi @anaelle-sw  Thanks very much for your question!  Support for lifecycle nodes in the RealSense ROS2 wrapper is a feature that has previously been suggested by a RealSense ROs community member in the link below.

https://github.com/IntelRealSense/librealsense/issues/5825#issuecomment-669345438

In regard to whether lifecycle node support is part of official future plans for the RO2 wrapper or whether a PR would be the preferable option, I will refer that question to @doronhi the RealSense ROS wrapper developer.

doronhi, 2021-05-02:
Hi @anaelle-sw 
Making realsense2_camera a lifecycle node is indeed in our intentions. It is currently planned for the last quarter of 2021 but currently not prioritized very high. Your indication that this feature is useful may boost it up.
As you say, it requires a design change. Any discussion and requirements description is very much appreciated.

anaelle-sw, 2021-05-06:
Hi, thanks to both of you for your answers! 

> Any discussion and requirements description is very much appreciated.

Well for the moment, the three main features that we would appreciate with lifecycle nodes are:
- Launching a realsense-ros node (but not configuring it, just keeping it un-configured). Then, when we connect a camera at runtime, configuring the realsense-ros node for this new camera
- Un-configuring a realsense-ros node at runtime (but not killing it, just keeping it un-configured) when we disconnected the associated camera
- Activating / de-activing a realsense-ros node to start / stop all streams (it could be quite equivalent to the "/enable" service already existing)

doronhi, 2021-05-09:
Hi,
Thank you for your comments and requirement description.
In the current version, if you start the node with the camera detached, the node just hangs there. When the camera is attached it is automatically detected, configured, and started. When the camera disconnects the node stays there, waiting for the camera to reconnect. Is that the behavior you wish to keep under the life-cycle structure?

doronhi, 2021-05-11:
As said, there is still some time before work will begin. For now, I internally logged your remarks under DSO-17089.
Thanks for the input.

MartyG-RealSense, 2021-06-14:
Hi @anaelle-sw   As lifecycle node plans for the ROS wrapper are set for some time in the future and your feedback has been logged by @doronhi  - is it okay to close this case please?  Thanks!

anaelle-sw, 2021-06-16:
Hello, yes sure, thanks for support

MartyG-RealSense, 2021-06-16:
You are very welcome @anaelle-sw - thanks very much for the update!

doisyg, 2021-09-22:
Hello,
Any update on this issue ? Is it still planned for the end of the year ?

imstevenpmwyca, 2022-01-05:
Hello @doronhi,
Is there any update regarding this lifecycle functionality? Would you share with us its current state or road-plan?
Thank you very much and kind regards!

SamerKhshiboun, 2023-03-22:
Hi @anaelle-sw, @imstevenpmwyca,
My name is Samer, and I am a developer in the RealSense SW Team.
As I got to this very interesting issue, I started to investigate the Lifecycle concept in ROS2 and estimating the effort we need to do these changes.
I will update asap.
Thank you.

alonsnir, 2023-07-23:
Any progress on this one?

SamerKhshiboun, 2023-07-23:
Hi @alonsnir ,
Sorry for delayed update, but the lifecycle is one of our major features that is planned to be completed by end of year.
Implementation should start by end of Q3 and finish by end of Q4.

nwn-cpr, 2024-01-25:
Hi @SamerKhshiboun, is there any update on the timeline for this feature?

quantumxt, 2024-06-07:
Hi there, wondering if there's any update regarding the lifecycle feature?

quantumxt, 2024-06-11:
Hi @SamerKhshiboun,

We are interested in utilizing the realsense-ros nodes as ROS2 Lifecycle Nodes to accommodate the dynamic addition and removal of Realsense cameras within our system at runtime. Previously, it was projected to be completed by Q4 2023 (https://github.com/IntelRealSense/realsense-ros/issues/1838#issuecomment-1646757580). However, there have been no updates since July 2023.

Given this context, could you please provide an update on the status of this feature?

Thank you.

SamerKhshiboun, 2024-06-11:
++ @Nir-Az @Arun-Prasad-V 

Hi @quantumxt 
This feature was planned for Q4 2023, but unfortunately was postponed due to other high priority projects.
We will have an internal developers sync on this feature next week, after it I can hopefully give you an update on the status and on the deadlines.

There are multiple questions that we have to answers in our team:
- should be in an independent branch ? and how should we handle future PRs ? should we merge each change into this branch also ?
- what is the best design inside our code to make this feature works ? need to decide which code parts are responsible for each block in the diagram
[image]
-  how much effort it will take ? does it need redesign of our code ?
-  how to handle 3rd party packages which doesn't support the lifecycle node ? e.g. image_transport ?

@quantumxt if you are already familiar with Lifecycle Nodes and are using other packages which support it, your inputs are more than welcome.

quantumxt, 2024-06-12:
Hi @SamerKhshiboun,

The use of Lifecycle Nodes is currently being explored, and we are still learning how to best utilize it. Below are some proposed implementation approaches for consideration:

* Implement the Lifecycle node as a standalone node (separate from the default node), to ensure backwards compatibility and optional use.

* Modify the current launch file to include a new parameter (to select the Lifecycle node), allowing users to opt-in to using the Lifecycle node.

* Introduce a Node Manager (optional) to orchestrate state transitions, which can be triggered automatically or through service calls.

* During state transitions:
  * Configuring: Initialize the camera publisher.
  * Activate: Activate the publisher & start publishing the camera feed, pointcloud, etc.
  * Deactivate: Stop the publisher.
  * Cleanup: Destroy the publisher.
 
These are some of our preliminary ideas, hope it would help.

quantumxt, 2024-07-11:
Hi @SamerKhshiboun, is there any update on the status of this feature?

Nir-Az, 2025-03-20:
Hi guys,
Glad to update that this feature is WIP and should be available soon (when building from source).
Stay tuned..

Nir-Az, 2025-03-23:
Feature was merged,
Please follow the readme for more details and try it .
Any feedback is welcomed :)
#3319

closed 2025-03-23 by Nir-Az