realsenseai/realsense-ros #1839: How to set exposure and framerate on T265 from ROS launch file?

shanekelly, 2021-05-01:
Hello,

I am wondering how to manually set the exposure and framerate for the T265 from a ROS launch file. I am able to set the exposure (though, not the framerate) from the Realsense Viewer, but I need to do so from a ROS launch file.

I am using [this](https://github.com/IntelRealSense/realsense-ros/files/6409198/t265.launch.txt) launch file and have tried to set the framerate via the `fisheye_fps` parameter, but when I do so I get [this](https://github.com/IntelRealSense/realsense-ros/files/6409200/output.txt) output with errors.

I have been unable to find a similar parameter for setting the exposure.

Any help would be greatly appreciated!

shanekelly, 2021-05-06:
@doronhi @MartyG-RealSense @RealSenseCustomerSupport @RealSenseSupport
Any advice on this question would be much appreciated!

doronhi, 2021-05-09:
Hi,
Regarding the fps, T265 only support 30Hz so it can't be changed with librealsense2 or the realsense2_camera wrapper.
There is an issue with setting the exposure manually within the launch file for T265 as have already been reported in #1049.
I see now that that issue was not taken care of. I put it now back on (DSO-17076), to see if it can be handled within librealsense2 library and if not, a workaround can be created within the realsense2_camera. The solution that now comes to my mind is keeping a list of options that failed to be set and give them another try after all the other options have been set.
Since fixing the librealsense2 will happen no sooner than the next release (at least 2-3 weeks) you could try implementing the suggested workaround (or another) for now.

closed 2026-10-23 by MartyG-RealSense