realsenseai/realsense-ros #1843: "control_transfer returned error" occurring at high frequency with multiple cameras

victorll998, 2021-05-03:
Hi everyone,

I have encountered similar cases related [#1162](https://github.com/IntelRealSense/realsense-ros/issues/1162#issue-598240927) but with multiple cameras (two d455 and one t265) in my system.

I have two d455 and t265 running on Intel NUC (i7). I tested with the cables inbox and a powered USB hub. When only one d455 and t265 connected, the control_transfer returned error has refrained after a few seconds when nodes initialize. But when I tried to connect two d455 and t265, one of the d455 would retain the control transfer error at a high frequency while the other d455 is fine.

The errors come at an average rate of 3 sec, which I think it is not normal. Does anyone have an idea about it?

camera 1 continue to give me the error 
```
SUMMARY
========

PARAMETERS
 * /cam_1/realsense2_camera/accel_fps: -1
 * /cam_1/realsense2_camera/accel_frame_id: cam_1_accel_frame
 * /cam_1/realsense2_camera/accel_optical_frame_id: cam_1_accel_optic...
 * /cam_1/realsense2_camera/align_depth: False
 * /cam_1/realsense2_camera/aligned_depth_to_color_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_fisheye_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_infra1_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_infra2_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/allow_no_texture_points: False
 * /cam_1/realsense2_camera/base_frame_id: cam_1_link
 * /cam_1/realsense2_camera/calib_odom_file: 
 * /cam_1/realsense2_camera/clip_distance: -2.0
 * /cam_1/realsense2_camera/color_fps: -1
 * /cam_1/realsense2_camera/color_frame_id: cam_1_color_frame
 * /cam_1/realsense2_camera/color_height: -1
 * /cam_1/realsense2_camera/color_optical_frame_id: cam_1_color_optic...
 * /cam_1/realsense2_camera/color_width: -1
 * /cam_1/realsense2_camera/confidence_fps: -1
 * /cam_1/realsense2_camera/confidence_height: -1
 * /cam_1/realsense2_camera/confidence_width: -1
 * /cam_1/realsense2_camera/depth_fps: -1
 * /cam_1/realsense2_camera/depth_frame_id: cam_1_depth_frame
 * /cam_1/realsense2_camera/depth_height: -1
 * /cam_1/realsense2_camera/depth_optical_frame_id: cam_1_depth_optic...
 * /cam_1/realsense2_camera/depth_width: -1
 * /cam_1/realsense2_camera/device_type: 
 * /cam_1/realsense2_camera/enable_accel: False
 * /cam_1/realsense2_camera/enable_color: True
 * /cam_1/realsense2_camera/enable_confidence: True
 * /cam_1/realsense2_camera/enable_depth: True
 * /cam_1/realsense2_camera/enable_fisheye1: False
 * /cam_1/realsense2_camera/enable_fisheye2: False
 * /cam_1/realsense2_camera/enable_fisheye: False
 * /cam_1/realsense2_camera/enable_gyro: False
 * /cam_1/realsense2_camera/enable_infra1: False
 * /cam_1/realsense2_camera/enable_infra2: False
 * /cam_1/realsense2_camera/enable_infra: False
 * /cam_1/realsense2_camera/enable_pointcloud: False
 * /cam_1/realsense2_camera/enable_pose: False
 * /cam_1/realsense2_camera/enable_sync: False
 * /cam_1/realsense2_camera/filters: spatial,temporal,...
 * /cam_1/realsense2_camera/fisheye1_frame_id: cam_1_fisheye1_frame
 * /cam_1/realsense2_camera/fisheye1_optical_frame_id: cam_1_fisheye1_op...
 * /cam_1/realsense2_camera/fisheye2_frame_id: cam_1_fisheye2_frame
 * /cam_1/realsense2_camera/fisheye2_optical_frame_id: cam_1_fisheye2_op...
 * /cam_1/realsense2_camera/fisheye_fps: -1
 * /cam_1/realsense2_camera/fisheye_frame_id: cam_1_fisheye_frame
 * /cam_1/realsense2_camera/fisheye_height: -1
 * /cam_1/realsense2_camera/fisheye_optical_frame_id: cam_1_fisheye_opt...
 * /cam_1/realsense2_camera/fisheye_width: -1
 * /cam_1/realsense2_camera/gyro_fps: -1
 * /cam_1/realsense2_camera/gyro_frame_id: cam_1_gyro_frame
 * /cam_1/realsense2_camera/gyro_optical_frame_id: cam_1_gyro_optica...
 * /cam_1/realsense2_camera/imu_optical_frame_id: cam_1_imu_optical...
[... 271 lines cut ...]
[ INFO] [1620023267.370824259]: Expected frequency for depth = 30.00000
[ INFO] [1620023267.386364551]: Expected frequency for color = 30.00000
[ INFO] [1620023267.396254492]: setupStreams...
[ INFO] [1620023267.417498629]: insert Depth to Stereo Module
[ INFO] [1620023267.417546521]: insert Color to RGB Camera
[ INFO] [1620023267.466236248]: SELECTED BASE:Depth, 0
[ INFO] [1620023267.471884995]: RealSense Node Is Up!
 03/05 16:27:47,550 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ WARN] [1620023267.680723885]: 
 03/05 16:27:47,681 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,731 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,782 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,832 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,883 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 03/05 16:27:47,933 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:47,985 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:48,035 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:48,085 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:53,244 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:55,295 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:55,345 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:56,396 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:56,446 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/05 16:27:58,498 WARNING [140139598698240] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

RealSenseSupport, 2021-05-12:
@victorll998 Do you mean that two D455 cameras connected to one USB hub?

victorll998, 2021-05-12:
@RealSenseSupport Yes

victorll998, 2021-05-12:
I also have done the same tests when I connect each d455 to thunderbolt3. The warnings are relieved compared to use the USB hub but still are coming at a disturbingly high rate.

RealSenseSupport, 2021-05-18:
@victorll998 Do you mean you still get frequent "control_transfer returned error" if you connect each D455 to host USB port directly? Do you still get such issue if you reduce RGB resolution to 848x480?

norman1479, 2021-05-18:
Hi All, we get the same message a few dozen times from each camera node after start-up, on industrial computers with 2, 3 or 4 cameras each, all attached to host USB ports. Resolutions used for both color and depth are 848x480.

Some of the output from lspci:

`00:00.0 Host bridge: Intel Corporation 8th Gen Core Processor Host Bridge/DRAM Registers (rev 07)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 07)
000:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian 
01:00.2 USB controller: NVIDIA Corporation TU104 USB 3.1 Host Controller (rev a1)`

victorll998, 2021-05-19:
@RealSenseSupport When I connect each D455 to the host USB port (thunderbolt3) I still get frequent "control_transfer returned error". I have the same issue even though I reduce RGB resolution to 848x480.

I also found this issue could be related to auto-exposure because I can see the frequent "control_transfer returned error" when I change the ambient light. To rule out possibilities of camera hiccup, could you take a look and see if you can reproduce the same error?

RealSenseSupport, 2021-05-31:
@victorll998 So you will get "control_transfer returned error" when connect two D455 to the host. But no such issue if connect one D455, right? 

How did you install librealsense and ROS wrapper? By apt-get install or build from source?

RealSenseSupport, 2021-06-09:
@victorll998 Could you please update so that we can provide further support on this? Thanks!

RealSenseSupport, 2021-07-21:
@victorll998 Do you still have issue? Looking forward to your update. Thanks!

RealSenseSupport, 2021-07-28:
@victorll998 Could you please update? Please note that the issue will be closed if we don't hear from you for another 7 days. Thanks!

RealSenseSupport, 2021-08-18:
@victorll998 Sorry we don't hear from you for weeks. This issue will be closed. Please feel free to create another one if you still have any issues or questions. Thanks!

closed 2021-08-18 by RealSenseSupport