realsenseai/realsense-ros #1844: L515 - get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Protocol error

htawadegetample, 2021-05-05:
Hello Everyone,
I am seeing the "failed Last Error: Protocol error" when I launch l515 using ROS node in a docker container. It doesn't come up directly after the launch but randomly after a while. I am running the l515 using ROS melodic on ubuntu 18.04 using a docker container. The kernel version is 5.4.0-67-generic. We have seen that it gets resolved by disconnecting and re-connecting the cable to the camera and relaunching the container. Following are the logs and my docker file.

PARAMETERS
 * /camera_0/realsense2_camera/accel_fps: 250
 * /camera_0/realsense2_camera/accel_frame_id: camera_0_accel_frame
 * /camera_0/realsense2_camera/accel_optical_frame_id: camera_0_accel_op...
 * /camera_0/realsense2_camera/align_depth: False
 * /camera_0/realsense2_camera/aligned_depth_to_color_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/allow_no_texture_points: False
 * /camera_0/realsense2_camera/base_frame_id: camera_0_link
 * /camera_0/realsense2_camera/calib_odom_file: 
 * /camera_0/realsense2_camera/clip_distance: -2.0
 * /camera_0/realsense2_camera/color_fps: 30
 * /camera_0/realsense2_camera/color_frame_id: camera_0_color_frame
 * /camera_0/realsense2_camera/color_height: 480
 * /camera_0/realsense2_camera/color_optical_frame_id: camera_0_color_op...
 * /camera_0/realsense2_camera/color_width: 640
 * /camera_0/realsense2_camera/confidence_fps: 30
 * /camera_0/realsense2_camera/confidence_height: 480
 * /camera_0/realsense2_camera/confidence_width: 640
 * /camera_0/realsense2_camera/depth_fps: 30
 * /camera_0/realsense2_camera/depth_frame_id: camera_0_depth_frame
 * /camera_0/realsense2_camera/depth_height: 240
 * /camera_0/realsense2_camera/depth_optical_frame_id: camera_0_depth_op...
 * /camera_0/realsense2_camera/depth_width: 320
 * /camera_0/realsense2_camera/device_type: 
 * /camera_0/realsense2_camera/enable_accel: False
 * /camera_0/realsense2_camera/enable_color: True
 * /camera_0/realsense2_camera/enable_confidence: True
 * /camera_0/realsense2_camera/enable_depth: True
 * /camera_0/realsense2_camera/enable_fisheye1: False
 * /camera_0/realsense2_camera/enable_fisheye2: False
 * /camera_0/realsense2_camera/enable_fisheye: False
 * /camera_0/realsense2_camera/enable_gyro: False
 * /camera_0/realsense2_camera/enable_infra1: False
 * /camera_0/realsense2_camera/enable_infra2: False
 * /camera_0/realsense2_camera/enable_infra: False
 * /camera_0/realsense2_camera/enable_pointcloud: True
 * /camera_0/realsense2_camera/enable_pose: False
 * /camera_0/realsense2_camera/enable_sync: False
 * /camera_0/realsense2_camera/filters: pointcloud
 * /camera_0/realsense2_camera/fisheye1_frame_id: camera_0_fisheye1...
 * /camera_0/realsense2_camera/fisheye1_optical_frame_id: camera_0_fisheye1...
 * /camera_0/realsense2_camera/fisheye2_frame_id: camera_0_fisheye2...
 * /camera_0/realsense2_camera/fisheye2_optical_frame_id: camera_0_fisheye2...
 * /camera_0/realsense2_camera/fisheye_fps: 30
 * /camera_0/realsense2_camera/fisheye_frame_id: camera_0_fisheye_...
 * /camera_0/realsense2_camera/fisheye_height: 480
 * /camera_0/realsense2_camera/fisheye_optical_frame_id: camera_0_fisheye_...
 * /camera_0/realsense2_camera/fisheye_width: 640
 * /camera_0/realsense2_camera/gyro_fps: 400
 * /camera_0/realsense2_camera/gyro_frame_id: camera_0_gyro_frame
 * /camera_0/realsense2_camera/gyro_optical_frame_id: camera_0_gyro_opt...
 * /camera_0/realsense2_camera/imu_optical_frame_id: camera_0_imu_opti...
 * /camera_0/realsense2_camera/infra1_frame_id: camera_0_infra1_f...
 * /camera_0/realsense2_camera/infra1_optical_frame_id: camera_0_infra1_o...
 * /camera_0/realsense2_camera/infra2_frame_id: camera_0_infra2_f...
 * /camera_0/realsense2_camera/infra2_optical_frame_id: camera_0_infra2_o...
[ WARN] [1620086846.574897221]: Device f0233041 is connected using a 2.1 port. Reduced performance is expected.
[ WARN] [1620086847.267373948]: Given stream configuration is not supported by the device!  Stream: Confidence, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: None
 * /camera_0/r[ INFO] [1620086846.333991602]: Initializing nodelet with 16 worker threads.
[ INFO] [1620086846.482937368]: RealSense ROS v2.2.21
[ INFO] [1620086846.482957925]: Built with LibRealSense v2.43.0
[ INFO] [1620086846.482969046]: Running with LibRealSense v2.43.0
[... 189 lines cut ...]
 04/05 15:03:41,813 ERROR [140636313298688] (error-handling.cpp:76) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
[ERROR] [1620140622.147375421]: The device has been disconnected!

=====================================================================================
Docker File

# Base image from docker hub
FROM ros:melodic-perception AS dev

# Install ros astra camera dependencies
RUN apt-fast install -qy ros-$ROS_DISTRO-rgbd-launch ros-$ROS_DISTRO-libuvc ros-$ROS_DISTRO-libuvc-camera ros-$ROS_DISTRO-libuvc-ros

# Install librealsense sdk
RUN apt-key adv --keyserver keys.gnupg.net --recv-key 6F3EFCDE
RUN add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
RUN apt-fast -qy install librealsense2-dkms
RUN apt-fast -qy install librealsense2-utils
RUN apt-fast -qy install librealsense2-dev
RUN apt-fast -qy install librealsense2-dbg
RUN apt-fast -qy  update
RUN apt-fast -qy upgrade

# Install realsense ros dependencies
RUN apt-fast -qy install ros-$ROS_DISTRO-ddynamic-reconfigure
RUN apt-fast -qy install ros-$ROS_DISTRO-diagnostics

RealSenseSupport, 2021-05-07:
Hi @htawadegetample 

Does the L515 function outside of docker container or are you seeing errors even when running the camera via the RealSense Viewer?

MartyG-RealSense, 2021-09-11:
Case closed due to no further comments received.

closed 2021-09-11 by MartyG-RealSense

MikhailBertrand, 2023-03-17:
For those who have faced these errors.

```
(librealsense-exception.h:52) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
(ds5-thermal-monitor.cpp:94) Error during thermal compensation handling: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such file or directory
```

In my case the issue was solved simply by updating the device's firmware.
You can easily update it via the prompt shown in the top right corner when opening realsense-viewer.

MartyG-RealSense, 2023-03-17:
Thanks so much @MikhailBertrand for sharing your experience with the RealSense community!

zp2546265641, 2024-06-14:
> Hi,
> 
> Will you be needing further help with this?
> 
> _If we don’t hear from you in 7 days, this issue will be closed._
> 
> Thanks

i have the same problem!!!can you help me?just because i use the docker hub

MartyG-RealSense, 2024-06-14:
Hi @zp2546265641  Which computer or computing device are you using, please? (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc).  Thanks!

zp2546265641, 2024-06-14:
> Hi @zp2546265641 Which computer or computing device are you using, please? (Raspberry Pi, Nvidia Jetson, desktop / laptop PC, etc). Thanks!

ubuntu20.04，ros noetic， laptop PC, yes，it is normal with usb connecting my computer,but when i use the docker hub,the same problen bring up

MartyG-RealSense, 2024-06-15:
@zp2546265641  Are you using the flag **--privileged** in your 'docker run' command and launching in sudo admin permissions mode **sudo docker run**, like the docker run command used on Noetic at https://github.com/IntelRealSense/realsense-ros/issues/2927

zp2546265641, 2024-06-17:
> @zp2546265641 Are you using the flag **--privileged** in your 'docker run' command and launching in sudo admin permissions mode **sudo docker run**, like the docker run command used on Noetic at #2927

where can i check and see if the flag "**docker run**" is set?

MartyG-RealSense, 2024-06-17:
**docker run** is not a flag, it is the command to launch Docker.  As far as I am aware, Docker would not be running if there was not a 'docker run' command.  The flags are the commands that start with - or -- that comes after the docker run command.  For example, with `sudo docker run -it --privileged` -it and --privilged are the flags.