realsenseai/realsense-ros #1845: Intel RealSense D455 not publishing infrared images

federicochiesa, 2021-05-05:
I have an Intel RealSense D455 and it's not publishing on the ```infra1``` and ```infra2``` topics, so I can't get the infrared cameras' video feed from ROS, while I can see it perfectly through ```realense-viewer```.
How can I solve this?

MartyG-RealSense, 2021-05-05:
Hi @federicochiesa  Could you try adding the instructions below to the end of your roslaunch instruction please to enable and publish the infrared topics:

**enable_infra1:=true enable_infra2:=true**

For example:

**roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true***

federicochiesa, 2021-05-15:
Thank you, that did it!

closed 2021-05-15 by federicochiesa

MartyG-RealSense, 2021-05-15:
Great news @federicochiesa - thanks for the update!