realsenseai/realsense-ros #1851: Windows installation pthread issue.

triplehoon, 2021-05-06:
I am now using windows ros-noetic.

I tried instruction on main page, but I get error message when I command "catkin_make clean".

`Traceback (most recent call last):
  File "C:/catkin_ws/build/catkin_generated/generate_cached_setup.py", line 22, in <module>
    code = generate_environment_script('C:/catkin_ws/devel/env.bat')
  File "C:\opt\ros\noetic\x64\lib\site-packages\catkin\environment_cache.py", line 63, in generate_environment_script
    env_after = ast.literal_eval(output.decode('utf8'))
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc7 in position 6268: invalid continuation byte
CMake Error at C:/opt/ros/noetic/x64/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(C:/opt/ros/noetic/x64/python.exe
  "C:/catkin_ws/build/catkin_generated/generate_cached_setup.py") returned
  error code 1
Call Stack (most recent call first):
  C:/opt/ros/noetic/x64/share/catkin/cmake/all.cmake:208 (safe_execute_process)
  C:/opt/ros/noetic/x64/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:58 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "C:/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed`

In CMakeErro.log

`
Determining if the include file pthread.h exists failed with the following output:
Change Dir: C:/catkin_ws/build/CMakeFiles/CMakeTmp

Run Build Command(s):C:/PROGRA~2/MICROS~2/2019/COMMUN~1/Common7/IDE/COMMON~1/MICROS~1/CMake/Ninja/ninja.exe cmTC_2dc49 && [1/2] Building C object CMakeFiles\cmTC_2dc49.dir\CheckIncludeFile.c.obj

FAILED: CMakeFiles/cmTC_2dc49.dir/CheckIncludeFile.c.obj 

C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1428~1.299\bin\Hostx64\x64\cl.exe  /nologo   /DWIN32 /D_WINDOWS /W3  /MDd /Zi /Ob0 /Od /RTC1 /showIncludes /FoCMakeFiles\cmTC_2dc49.dir\CheckIncludeFile.c.obj /FdCMakeFiles\cmTC_2dc49.dir\ /FS -c CheckIncludeFile.c

CheckIncludeFile.c(1): fatal error C1083: 포함 파일을 열 수 없습니다. 'pthread.h': No such file or directory
ninja: build stopped: subcommand failed.
`

I install pthread by using vcpkg, but It failed agian.  I think i have to change CMakeList.txt, but I don't know how...

MartyG-RealSense, 2021-05-06:
Hi @triplehoon  As a starting point in investigating your case, could you check a step by step installation guide in the link below that I made for a Noetic user on Windows when they were having problems with **catkin_make clean**, please?  Going through the steps may highlight whether you have missed a step

https://github.com/IntelRealSense/realsense-ros/issues/1366

MartyG-RealSense, 2021-05-12:
Hi @triplehoon  Do you require further assistance with this case, please?  Thanks!

triplehoon, 2021-05-13:
@MartyG-RealSense I just deciside to use in Linux enviroment. However thanks

closed 2021-05-13 by triplehoon