realsenseai/realsense-ros #1853: ROS2 Dashing Wrapper Fails colcon build

IndefiniteBen, 2021-05-07:
Following the installation instructions in [the README](https://github.com/IntelRealSense/realsense-ros/tree/ros2#installation-instructions) I followed the steps:
1. Install [ROS2 Dashing](https://docs.ros.org/en/dashing/Installation/Ubuntu-Development-Setup.html) using [this script](https://github.com/IndefiniteBen/installROS2) (FYI the link to the dashing instructions in the README is wrong)
2. Installed RealSense SDK from Debian package using [this script](https://github.com/jetsonhacks/installRealSenseSDK/blob/master/installLibrealsense.sh)
3. Clone ROS2 wrapper and install dependencies as explained in README
4. Try to use ` colcon build` and get the following error
```
nvidia@nvidia-desktop:~/ros2_ws$ colcon build
Starting >>> realsense2_camera_msgs
[Processing: realsense2_camera_msgs]                             
Finished <<< realsense2_camera_msgs [38.9s]                         
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [8.75s]                                                                
--- stderr: realsense2_camera                           
CMake Warning at CMakeLists.txt:101 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.45.0".

  The following configuration files were considered but not accepted:

    /usr/lib/aarch64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.41.0

CMake Error at CMakeLists.txt:103 (message):
  
   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
 
---
Failed   <<< realsense2_camera [11.9s, exited with code 1]

Summary: 2 packages finished [51.6s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
```
As I understand it, 2.41 is just the version installed from the Debian packages repo.
Therefore, my options are either build the SDK so I have 2.45, or clone a different wrapper such as [3.1.3](https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.3) but that says it is for Foxy, not Dashing. 

How can I best resolve this issue?

MartyG-RealSense, 2021-05-07:
Hi @IndefiniteBen  Support for Dashing in the RealSense ROS wrapper was added in ROS2 wrapper **3.1.5**.  From this version onwards, a single ROS2 wrapper provides support for Dashing, Eloquent and Foxy.  Before 3.1.5, Dashing was not supported and there were separate wrappers for Eloquent and Foxy.

The links in the readme are not correct but they are broken at the time of writing this because the Debian packages are missing, as described in te link below.

https://github.com/IntelRealSense/realsense-ros/issues/1814#issuecomment-823020133

If you are able to install librealsense 2.45.0 then you could install the Dashing-compatible 3.1.5 ROS wrapper by building it from source code.

Alternatively you could use an older ROS2 wrapper for Dashing that was produced by Intel but is unrelated to the current ROS wrapper and is no longer supported.  Because support has ended for it, it will not work with recent RealSense camera models such as D455 and L515 if you have one of those camera models but should work with models such as D415 and D435 / D435i.

https://github.com/intel/ros2_intel_realsense

IndefiniteBen, 2021-05-07:
I'm using the L515 so can't use the old wrapper unfortunately.

If I understand correctly, to install 2.45 I need to wait for the Debian packages to be added? I see your comment asking for an update so I guess it's just a case of waiting, or I could build the SDK from source?

MartyG-RealSense, 2021-05-07:
If you build the SDK and the ROS2 wrapper from source then you should be able to use them immediately instead of waiting for Debian packages for 2.45.0.

I am not involved in handling L515 related cases, so if you require further assistance with this issue then I recommend closing this case and opening a new one with 'L515' included in the message title so that the appropriate RealSense team member can assist you.  Good luck!

IndefiniteBen, 2021-05-07:
Okay, understood. Thanks for your help!

I'll close this issue once I open a new one. Have you considered using issue templates to prevent this kind of wrong assignment?

MartyG-RealSense, 2021-05-07:
Such a message is not suitable for the forums.  Thanks very much for the suggestion though   :)

IndefiniteBen, 2021-05-07:
Closing to replace with #1854 for L515

closed 2021-05-07 by IndefiniteBen