realsenseai/realsense-ros #1854: ROS2 Wrapper Fails colcon build for ROS2 Dashing with L515

IndefiniteBen, 2021-05-07:
This is a copy of #1853 that specifies the camera model used in the title.

Following the installation instructions in [the README](https://github.com/IntelRealSense/realsense-ros/tree/ros2#installation-instructions) I followed the steps:
1. Install [ROS2 Dashing](https://docs.ros.org/en/dashing/Installation/Ubuntu-Development-Setup.html) using [this script](https://github.com/IndefiniteBen/installROS2)
2. Installed RealSense SDK from Debian package using [this script](https://github.com/jetsonhacks/installRealSenseSDK/blob/master/installLibrealsense.sh)
3. Clone ROS2 wrapper and install dependencies as explained in README
4. Try to use ` colcon build` and get the following error
```
nvidia@nvidia-desktop:~/ros2_ws$ colcon build
Starting >>> realsense2_camera_msgs
[Processing: realsense2_camera_msgs]                             
Finished <<< realsense2_camera_msgs [38.9s]                         
Starting >>> realsense2_camera
Starting >>> realsense2_description
Finished <<< realsense2_description [8.75s]                                                                
--- stderr: realsense2_camera                           
CMake Warning at CMakeLists.txt:101 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.45.0".

  The following configuration files were considered but not accepted:

    /usr/lib/aarch64-linux-gnu/cmake/realsense2/realsense2Config.cmake, version: 2.41.0

CMake Error at CMakeLists.txt:103 (message):
  
   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases
 
---
Failed   <<< realsense2_camera [11.9s, exited with code 1]

Summary: 2 packages finished [51.6s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
```
The summary of the #1853 replies is as follows:
* Support for Dashing in the RealSense ROS wrapper was added in 3.1.5 (supports v2.43 of SDK)
* The Debian packages are missing for v2.45 of the SDK, see [issue 1814](https://github.com/IntelRealSense/realsense-ros/issues/1814#issuecomment-823020133)
* Possible solutions:
  * Wait for RealSense SDK v2.45 Debian packages
  * Build the SDK to get v2.45

RealSenseCustomerSupport, 2021-07-14:
----------------------------------------------

Please try installing the latest SDK 2.48.0, wrapper 3.2.2, and f/w 1.5.8.1 and let me know if you are still having issues.

RealSenseCustomerSupport, 2021-08-02:
----------------------------------------------

@IndefiniteBen, have you tried the latest s/w and f/w combinations? Are you still experiencing this issue? If we don't hear back within 7 days we will assume the problem has been resolved and we will close the issue.