realsenseai/realsense-ros #1855: Filtering Edge Noise Points

senoa95, 2021-05-07:
Hello, 
I am using an L515 with ROS Melodic. I am having issues filtering out noise points that occur at the edge of objects. The noise levels in realsense-viewer is a lot less compared to what I see in rviz. Please see the comparison in the pictures. I have posted the side images to demonstrate. I am using the ROS default launch params to get the camera publishing in ROS.
[image]
realsense-viewer
[image]
rviz - axis is the camera pose.

RealSenseSupport, 2021-05-26:
In the RealSense Viewer there are some performance settings that may be turned on that smooths out the processing.  You can have a look in the settings menu similiar to the image here

[image]

The GLSL settings could be turned on in your configuration setup.  See if you turn off these settings if there is any effect on the image generated.

senoa95, 2021-06-07:
I did some checking and it appears that the noise was induced by config changes that I had in the ros node launch file. I tried the default launch file and the results are much better. Thank you everyone for checking and the responses.
 
Closing this issue.

closed 2021-06-07 by senoa95