realsenseai/realsense-ros #1858: very poor performance of d455 in pi 4B

roboticsai, 2021-05-09:
I have connected my d455 on raspberry pi 4B and ran the rs_launch.py ros2 launch file.
But after i visualize the image and point cloud in rviz2 the data is comming very slow such that it looks like i've waster my money. 
I have connected the d455 to the usb 3 port of pi 4B. 

Environment:
1. OS = ubuntu server 20.04
2. Ros version = ros2 foxy 
3. realsennse sdk = https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md
4. realsense-ros version = foxy
5. camera model = intel realsense d455

ros2 topic pointcloud hz output rate is 1.7 and image rate is 4. When i run the same in pc, pointcloud rate is 14 along with image and rviz visualization is also very real time.

MartyG-RealSense, 2021-05-09:
Hi @roboticsai   As a starting point in investigating your case, may I ask which librealsense SDK version and ROS2 Foxy wrapper version you have installed, please?

MartyG-RealSense, 2021-05-11:
Hi @roboticsai   It is important to know the exact librealsense and ROS wrapper version numbers, as a particular ROS wrapper version should be matched as closely as possible with a specific librealsense version - for example, librealsense SDK **2.45.0** and ROS2 wrapper **3.2.0**.  The librealsense version to match a wrapper with is listed in the release notes of a particular wrapper version.

https://github.com/IntelRealSense/realsense-ros/releases

[image]

If you are not certain which librealsense version is installed, you can find out by running the RealSense Viewer program and looking at the version number at the top of the Viewer window,

[image]

MartyG-RealSense, 2021-05-17:
Hi @roboticsai  Do you require further assistance with this case, please?  Thanks!

roboticsai, 2021-05-17:
Well, since the data like image, imu, gyro and pointcloud are ok on the dev board and the latency is only on thre remote computer. I think that will not cause problem for now. I installed https://github.com/ros2/rmw_cyclonedds/tree/foxy this rmw for dds also but still the same issue was comming i.e latency in data on remote computer (connnected with lan).

MartyG-RealSense, 2021-05-17:
I am reminded of how a RealSense user recently tried implementing the SDK's **rs-server** networking in ROS using ROS protocols instead of the networking system's default protocols and was experiencing average speeds of 2Hz on the depth image and 10Hz on color compared to normal performance with a non-networked camera.

https://github.com/IntelRealSense/realsense-ros/issues/1808

Another RealSense ROS user improved the performance on their particular system by setting RViz to use compressed data.

https://github.com/IntelRealSense/realsense-ros/issues/1510#issuecomment-839616982

roboticsai, 2021-05-23:
No thanks.

closed 2021-05-23 by roboticsai