realsenseai/realsense-ros #1859: Change D435i min depth parameter

richardrye, 2021-05-10:
I'm currently using a D435i depth camera on a mobile robot, wheel loader design, and have installed the SLAM package from: https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

I would like to increase the min camera detection parameter to 1 m as I have a piston that is being detected; effectively creating an occlusion zone. I actually added a parameter line that worked perfectly but accepted a realsense2 update that unfortunately wiped it. I didn't record what or where I added it and just can't recall.

Can you please provide the param details, including which file to add it to.

Thanks it's greatly appreciated.

MartyG-RealSense, 2021-05-10:
Hi @richardrye  I do not know a RealSense ROS parameter that would set a minimum distance, as the RealSense ROS wrapper does not have such a setting.  Instead, it can set a **maximum** distance with the **clip_distance** setting.

As you are using the D435i SLAM tutorial, I wonder whether you set a minimum distance in **rtabmap_ros** by using its **RangeMin** option, as described in the link below.

https://github.com/introlab/rtabmap_ros/issues/413

richardrye, 2021-05-10:
Thanks Marty,

I think it may indeed have been a parameter line with rtabmap_ros. Could you please advise if the following is correct to set the min detection camera range to 1 m.

<param name="Grid/RangeMin" type="double" value="1.0"/>

Also can you please specify which file I'm adding this to, I don't think the rtabmap.launch file would be the one.

Thanks.

Richard

introlab/rtabmap_ros#413<https://github.com/introlab/rtabmap_ros/issues/413>

MartyG-RealSense, 2021-05-10:
I have not personally used the RangeMin setting but I researched it and nearly all of the references that I found used the line that you provided above, with only the value being different.  So it looks okay to me.

In regard to where to use it, I found an example of it being called as part of a roslaunch of rtabmap.launch:

roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start --Grid/RangeMax 3 **--Grid/RangeMin 0.3** --Grid/DepthDecimation 4"

Whilst the link below provides an example of using RangeMin as a param name like you did above, in an rtabmap_ros launch file:

https://github.com/introlab/rtabmap_ros/issues/467#issuecomment-699666604

richardrye, 2021-05-10:
Thanks for your quick response and I've solved the issue.

Added  --Grid/RangeMin 1.0 to the opensource_tracking.launch file as shown below.

<arg name="args" value="--delete_db_on_start --Grid/RangeMin 1.0"/>

Thanks again, you were very helpful.

introlab/rtabmap_ros#467 (comment)<https://github.com/introlab/rtabmap_ros/issues/467#issuecomment-699666604>