realsenseai/realsense-ros #1860: (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_

tmd9936, 2021-05-10:
I used realsense l515 in jetson xavier 
And running roslaunch in docker

It works normally in the real environment

this is my docker command
```bash
$ docker run --runtime nvidia -it --rm \
   --network host \
   --privileged \
   --user=$(id -u $USER):$(id -g $USER) \
   --env="QT_X11_NO_MITSHM=1" \
   --env="DISPLAY" \
   --volume="/dev/:/dev/" \
   --volume="$PWD:/home/${USER}" \
   --volume="/etc/group:/etc/group:ro" \
   --volume="/etc/passwd:/etc/passwd:ro" \
   --volume="/etc/shadow:/etc/shadow:ro" \
   --volume="/etc/sudoers.d:/etc/sudoers.d:ro" \
   --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw" \
   ros:multi_cam_fusion03 bash

```

this is Log
```bash
NODES
  /
    cam_fusion_node (multi_cam_fusion/cam_fusion_node)
    fusion_save_list (multi_cam_fusion_save/fusion_save_list)
    lepton (lepton_camera/uvc-radiometry_node.py)
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [160]
[ INFO] [1620619149.739781054]: Initializing nodelet with 4 worker threads.
process[camera/realsense2_camera-2]: started with pid [161]
[ INFO] [1620619150.480000282]: RealSense ROS v2.2.21
[ INFO] [1620619150.480143163]: Built with LibRealSense v2.41.0
[ INFO] [1620619150.480217308]: Running with LibRealSense v2.41.0
[ INFO] [1620619150.515766473]:  
[ INFO] [1620619150.594620957]: Device with serial number f0190093 was found.

[ INFO] [1620619150.595441763]: Device with physical ID /sys/devices/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.0/video4linux/video0 was found.
[ INFO] [1620619150.595731461]: Device with name Intel RealSense L515 was found.
[ INFO] [1620619150.604496680]: Device with port number 2-3.1 was found.
[ INFO] [1620619150.604661929]: Device USB type: 3.2
[ INFO] [1620619150.613744942]: getParameters...
process[lepton-3]: started with pid [172]
[ INFO] [1620619150.768296574]: setupDevice...
[ INFO] [1620619150.770378510]: JSON file is not provided
[ INFO] [1620619150.772119035]: ROS Node Namespace: camera
[ INFO] [1620619150.773614694]: Device Name: Intel RealSense L515
[ INFO] [1620619150.775473909]: Device Serial No: f0190093
[ INFO] [1620619150.777205730]: Device physical port: /sys/devices/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.0/video4linux/video0
[ INFO] [1620619150.779184689]: Device FW version: 01.03.15.100
[ INFO] [1620619150.781621667]: Device Product ID: 0x0B64
[ INFO] [1620619150.781726404]: Enable PointCloud: Off
[ INFO] [1620619150.781811588]: Align Depth: Off
[ INFO] [1620619150.781853701]: Sync Mode: Off
[ INFO] [1620619150.782542122]: Device Sensors: 
[ INFO] [1620619150.786542088]: L500 Depth Sensor was found.
[ INFO] [1620619150.790651943]: RGB Camera was found.
[ INFO] [1620619150.791663759]: Motion Module was found.
[ INFO] [1620619150.792237843]: num_filters: 0
[ INFO] [1620619150.792338868]: Setting Dynamic reconfig parameters.
[ INFO] [1620619150.906239953]: Done Setting Dynamic reconfig parameters.
[ INFO] [1620619150.911280279]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1620619150.920403548]: color stream is enabled - width: 1920, height: 1080, fps: 15, Format: RGB8
[ INFO] [1620619150.921198146]: confidence stream is enabled - width: 640, height: 480, fps: 30, Format: RAW8
[ INFO] [1620619150.921380388]: setupPublishers...
[ INFO] [1620619150.929282015]: Expected frequency for depth = 30.00000
[ INFO] [1620619150.943875982]: Expected frequency for color = 15.00000
[ INFO] [1620619150.959972360]: Expected frequency for confidence = 30.00000
[ INFO] [1620619150.969014956]: setupStreams...
[ INFO] [1620619150.983716923]: insert Depth to L500 Depth Sensor
[ INFO] [1620619150.984342144]: insert Color to RGB Camera
[ INFO] [1620619150.984532769]: insert Confidence to L500 Depth Sensor
[ INFO] [1620619151.019939373]: SELECTED BASE:Depth, 0
process[fusion_save_list-4]: started with pid [193]
 10/05 03:59:11,203 WARNING [547197374848] (l500-depth.cpp:455) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.
[ INFO] [1620619151.306818502]: Fusion Save List Server Start!
process[cam_fusion_node-5]: started with pid [203]
[ INFO] [1620619151.726045448]: RealSense Node Is Up!
 10/05 03:59:11,740 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 10/05 03:59:11,756 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 10/05 03:59:11,759 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 10/05 03:59:11,761 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 10/05 03:59:11,862 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 10/05 03:59:11,926 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 10/05 03:59:11,993 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
 10/05 03:59:12,072 ERROR [547197374848] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: rnel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_

```

What is this and what cause this problem?

RealSenseSupport, 2021-05-21:
@tmd9936 

Have you had this type of issue outside of docker container?

RealSenseSupport, 2021-06-30:
Just wondering if the problem still persist with the latest release of RealSense SDK...
Thank you.

closed 2021-07-07 by RealSenseSupport