realsenseai/realsense-ros #1865: do we have something similar to rs2_deproject_pixel_to_point in ROS

yuan0623, 2021-05-11:
Hi,

I'm trying to obtain the 3D coordinates of a certain pixel using ROS topic `/camera/aligned_depth_to_color/image_raw`. 
I have the pixel coordinate (x_p, y_p) and the distance value of (x_p, y_p) obtained through `/camera/aligned_depth_to_color/image_raw`. My question is: Do we have anything similar to `rs2_deproject_pixel_to_point in ROS` in ROS wrapper so I can directly obtain the 3D coordinate? Which topic should I use to obtain `intrinsics`? Many thanks!

Best
Yuan

ravijo, 2021-05-14:
@yuan0623 

I think you have already found the answer based on the given references above. However, if you are still looking, please read the following.

I made a ROS package to do the same. Please check it out [ros_openpose](https://github.com/ravijo/ros_openpose).

I think what you are looking for is available inside the [ros_openpose](https://github.com/ravijo/ros_openpose) repository [here](https://github.com/ravijo/ros_openpose/blob/1c3dd67584e7ff4842d2021b76ae245b77615812/include/ros_openpose/cameraReader.hpp#L109-L145).

Cheers!

MartyG-RealSense, 2021-05-14:
Thanks very much @ravijo for your assistance to @yuan0623   :)

ravijo, 2021-05-14:
@MartyG-RealSense: Thanks for the kind words. It's my pleasure to help others.

yuan0623, 2021-05-18:
> @yuan0623
>
> I think you have already found the answer based on the given references above. However, if you are still looking, please read the following.
>
> I made a ROS package to do the same. Please [...]

Hi Ravijo,

Thanks for your help, I will try to use your package.

Best
Yuan

closed 2021-05-18 by yuan0623