realsenseai/realsense-ros #1866: Best approach for frame on demand?

Bart-van-Ingen, 2021-05-12:
Hi!

For my application, I am not in need of a constant stream of images from the realsense to the ros2 topic. 
I want to be able to get a depth image on command by sending a command over a topic named ```/shutter```. 

I think that there are two ways of approaching this:

1. Fork the realsense-ros repo and alter it such that the _syncer does not use the ```frame_callback``` directly anymore, but instead only fills a new class variable of the base realsense node of type ```rs2::frame``` named ```current_frame```. Then a subscription is made on the ```/shutter``` topic which will call the current implemented ```frame_callback``` using whatever frame is in ```current_frame```.

2. Use some form of message filter from [wiki.ros.org/message_filters](wiki.ros.org/message_filters), although I am not sure if this is applicable ... any suggestions?

What do you think would  be the best way to approach this?

MartyG-RealSense, 2021-05-12:
Hi @Bart-van-Ingen  There was a past case that discussed toggling the publishing of sensors on and off using a ROS node called **mux** to create two publishers, one that sends a topic's data and one that doesn't, and switch between them.

https://answers.ros.org/question/257561/stop-publishing-a-specific-topic/
http://wiki.ros.org/topic_tools/mux

Alternatively, during the past week RealSense ROS2 Foxy wrapper **3.1.6** introduced a service to toggle sensors on and off.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.6

Bart-van-Ingen, 2021-05-14:
Hi @MartyG-RealSense, thanks for the reply, and sorry for the late response!

I have been playing with the sensor toggle and it is almost what i am looking for!
What I have noticed while using it is that when I disable and enable the sensors, the rgb camera takes some time (around 2 seconds) to adjust to the lighting conditions.

Because I need to be able to get images directly when I publish a command on the ```shutter``` topic, without the need for adjustment, this feature is not sufficient. Is there some parameter i can use to mitigate this, or is there another approach I can try out?

MartyG-RealSense, 2021-05-14:
If you are using the camera with auto-exposure enabled then it can take several frames for the auto-exposure to settle down.  If the lighting conditions where you are using the camera are consistent throughout the day then you could instead disable auto-exposure and set a manual exposure value.  Manual exposure does not require a settling-down period.

Bart-van-Ingen, 2021-05-14:
ah yes that makes sense! I am looking at the variables that can be changed in ```rs_launch.py``` and I do not see an option to turn auto exposure off on launch ... is this something that is done through a .yaml file?

MartyG-RealSense, 2021-05-14:
The links below have information about configuring parameters with a yaml file in the RealSense ROS2 wrapper.

https://github.com/IntelRealSense/realsense-ros/issues/1640
https://github.com/IntelRealSense/librealsense/issues/8863

I carefully researched your question about how to disable **RGB** auto-exposure but could not locate the information.  So I will refer it to @doronhi the RealSense ROS wrapper developer.

Hi @doronhi  Could yo uprovide advice please to @Bart-van-Ingen about how to disable RGB auto-exposure in a ROS2 yaml file please?  Thanks!

Bart-van-Ingen, 2021-05-19:
Hi @doronhi have you had time yet to look at this question?

doronhi, 2021-05-20:
You can add the following line to the list of parameters in rs_launch.py:
`{'name': 'stereo_module.enable_auto_exposure', 'default': 'false', 'description': 'set rgb_camera auto_exposure'},`

Bart-van-Ingen, 2021-05-26:
great! that answers my question. i will close the issue.

closed 2021-05-26 by Bart-van-Ingen