realsenseai/realsense-ros #1870: How to get depth scale value using ros

ranjitkathiriya, 2021-05-12:
Hello there, 

I have a code snippet and I don't know how can I obtain this value in ROS.

```
# Start streaming
profile = pipeline.start(config)

# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)
```

How can I get first_depth_sensor -> get_depth_scale() value I think it is 0.001. Can I get this value using ROS? or I have to take this value as static?

Thank,
Ranjit

MartyG-RealSense, 2021-05-12:
Hi @ranjitkathiriya  Unless you need to change the depth scale value whilst your application is running, it is simpler to hard-code a fixed value of **0.001** (the default depth scale of the 400 Series cameras) into your application instead of retrieving the value withan SDK instruction in real-time as the value does not change unless deliberately altered. 

However, **Doronhi** the RealSense ROS wrapper developer explains in the link below why the depth scale cannot be altered in the RealSense ROS wrapper.

https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873

ranjitkathiriya, 2021-05-12:
I thought 0.001  first_depth_sensor() must be in camera_info =, but it's okay I will hardcode it. Thank!

closed 2021-05-13 by ranjitkathiriya