realsenseai/realsense-ros #1871: The poor pointcloud2

kiennguyen610, 2021-05-13:
Hi everyone,
I am facing a problem with the poor pointcloud2. I installed all dependency packages. But when I run the command: roslaunch realsense2_camera opensource_tracking.launch. The pointcloud2 is quite bad. When I put the camera in this corner, the pointclouds will save the images here but when I put the camera in the nearby corner, the pointclouds will not save the images in the previous corner. I try moving the camera slowly so the pointclouds are good. Now, I am in the drone mapping project, and when I run the drone slowly to get the best pointclouds, the drone can not enough lifting force. Has anyone had the same problem as me, can you give me a solution? Thanks.

kiennguyen610, 2021-05-13:
And I am using the D435i camera and I am testing on my laptop. Then, I will use Jetson Nano kit to replace my laptop.

MartyG-RealSense, 2021-05-13:
Hi @kiennguyen610   When you mention that when you go to one corner the images of the nearby corner are not saved, I wonder whether this is due to the camera's **field of view (FOV)**.  The field of view of the camera determines how much of a scene the camera can see at one time without turning left / right / up / down, similar to how a human might be able to see nearby details unless they turn their head.  So when the camera is in one corner, the other corner might be outside of its field of view.

It is possible to expand in RealSense ROS the size of the area that can be captured in a point cloud by adding an additional RealSense camera and combining their individual point clouds together automatically.  Intel have a guide for doing so in the link below.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

I note that you state that you can obtain a good point cloud when moving slowly.  So if both corners are actually inside the field of view of the camera at the same time, maybe the problem is that the point cloud image breaks if the camera is moved quickly?

kiennguyen610, 2021-05-13:
Thanks for your response. But I wonder if this is another solution for this. Because I don't have enough finance to buy a Intel Camera.

MartyG-RealSense, 2021-05-13:
Is your system that uses realsense2_camera opensource_tracking.launch based on Intel's D435i SLAM tutorial, please?

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

If it is then you may get better results if you add support for **localization** in rtabmap-ros to the project.

https://github.com/IntelRealSense/librealsense/issues/8692#issuecomment-809287786

kiennguyen610, 2021-05-13:
And When I move the camera around, I see that The IMU moves wildly.

kiennguyen610, 2021-05-13:
Yes, I followed the above link and I added 4 dependence packages. But, the result is not good.

MartyG-RealSense, 2021-05-13:
Could you test what results you get if you put the command below on the end of your roslaunch instruction:

**unite_imu_method:=copy**

For example:

**realsense2_camera opensource_tracking.launch unite_imu_method:=copy**

The opensource_tracking launch file sets unite_imu to 'linear_interpolation'.  Using 'copy' in the roslaunch instruction shoud overrule the setting inside the launch file.

kiennguyen610, 2021-05-13:
I will try this and I will notify you about the result.

kiennguyen610, 2021-05-13:
I tried that command, but the result is not better. And when I open the terminal, I see the error is : [ERROR] (2021-05-13 15:47:03.789) Rtabmap.cpp:1156::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 417 is ignored!
Are you know about it @MartyG-RealSense

MartyG-RealSense, 2021-05-13:
Thank you very much for trying.  There was a past case of this error in the link below, apparently caused by the camera moving too fast.

https://github.com/IntelRealSense/realsense-ros/issues/1635

I recall that after ROS wrapper version 2.2.15, the IMU topics were disabled in the ROS wrapper by default but the D435i SLAM article's documentation was not updated to reflect this.  So can you try this launch instruction next please:

**realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true**

Thanks very much for your patience!

kiennguyen610, 2021-05-13:
Thanks for your response immediately ! I checked and the SDK version is v2.4.5.0 and the Realsense ROS wrapper version is v2.3.0

MartyG-RealSense, 2021-05-13:
Thanks for the confirmation.  That is the correct librealsense SDK and RealSense ROS wrapper version matching.

kiennguyen610, 2021-05-13:
Thanks. So, I don't know where I was wrong that the Pointclouds is very bad.

MartyG-RealSense, 2021-05-13:
Have you been able to test launching it with the enable commands for gyro and accel mentioned above?

**realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true**

https://github.com/IntelRealSense/realsense-ros/issues/1871#issuecomment-840427150

kiennguyen610, 2021-05-13:
I tried that command and I don't know the reason why fixed [map] frame does not exit. And the mapping is not created.

kiennguyen610, 2021-05-13:
I closed the Rviz program and rerun it again. The camera frame is good. But the result is still not better. I am really impotent

MartyG-RealSense, 2021-05-13:
Have you got Fixed Frame set as **map** as shown in the Rviz image in the guide, please?

[image]

kiennguyen610, 2021-05-13:
Yes, I have set Fixed Frame as map as shown in the Rviz image in the guide. 
[image]

MartyG-RealSense, 2021-05-13:
So to summarise, the system does work but the main problem - as indicated at the start of the case - is that the point cloud breaks when the camera is moved quickly due to the IMU?

kiennguyen610, 2021-05-13:
Yes, I think so.

MartyG-RealSense, 2021-05-13:
And does the point cloud recover if you stop movement as soon as it begins breaking?

kiennguyen610, 2021-05-13:
It seems the image don't match together.

MartyG-RealSense, 2021-05-13:
The next logical test to do would be to attempt to calibrate the IMU if you have not done so already, using the Python tool provided by Intel.

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

kiennguyen610, 2021-05-13:
Thanks ! I will try it later. Now, I am testing automatically mode for drone

kiennguyen610, 2021-05-14:
Hi @MartyG-RealSense  ! I am so sorry to bother you again. But, I find that my project has a problem as below. I tried some solutions and calibrated the IMU. However, the problem is not solved. Did you catch the same problem ?  
[image]

MartyG-RealSense, 2021-05-14:
I have come across some of the errors in your image in past cases.  Of the available references, the one that I know of with the most useful advice is here:

https://github.com/IntelRealSense/realsense-ros/issues/1080

It suggests that when the "Not enough outliers" message occurs, the camera should be put in a stable position and have an "interesting" scene to view (i.e keep the camera looking at scenes with features).

https://github.com/IntelRealSense/realsense-ros/issues/1080#issuecomment-589037861

Is your Image topic set to **/camera/color/image/raw** like in the D435i SLAM guide?

Under the **TF > Frames** section, do you have only the **camera_link** and the **map** boxes ticked and no others?

[image]

kiennguyen610, 2021-05-17:
I am sorry for delay. But I set TF -> Frames -> camera_link and map and I also set camera/color/image_raw like above. And I wonder if how an "interesting" scene to view is ? Thanks for your assistant !

MartyG-RealSense, 2021-05-17:
I would say that an "interesting" scene would be one that is not a big empty space, one that has enough surfaces and objects in it to provide the camera with a decent amount of detail to analyze.  This could be a problem on a flying drone (I note your earlier mention of 'lifting force') if it is moving through wide open skies with no detectable objects within depth sensing range instead of an enclosed room.

kiennguyen610, 2021-05-17:
Thanks for your assistant again. But, I wonder whether I have to adjust any parameters in camera or rtabmap.launch to map. I tried to adjust some parameters in rs_camera.launch and rtabmap.launch but it seems not effective.

MartyG-RealSense, 2021-05-17:
You could try adjusting the minimum inliers - **MinInliers** - in RTABMAP using the guidance below, perhaps lowering the value to address the  **Not enough inliers** warnings in your log.

http://official-rtab-map-forum.67519.x6.nabble.com/Minimum-inliers-Where-to-set-this-option-td4585.html

kiennguyen610, 2021-05-18:
Hi @MartyG-RealSense , I have a question about the camera. I wonder if the D435i is not suitable for this project, because my friend uses the D415 camera and he creates a map well. I only install 4 packages with the introduction of the project, but it has errors. I take 2 weeks for fixing error but it is not effective.

MartyG-RealSense, 2021-05-18:
The key differences between the D415 and D435i - aside from the D415 not having an IMU - are:

-  A smaller field of view size on D415.  If the application reacts poorly to large empty spaces on the image, then a D415 image should be able to see less of that empty space.

-  The D415 has smaller infrared sensors compared to the 'wide' IR sensors on D435i.

-  The D415's optimal depth accuracy resolution is 1280x720, whilst the depth resolution that the D435i operates optimally at is 848x480

kiennguyen610, 2021-05-18:
Hi @MartyG-RealSense , I think the hardware of my project may not wrong. But it is my log when I run the comman : " roslaunch realsense2_camera opensource_tracking.launch " . You can check it for anything wrong. I will be very grateful if you help me 
" PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
[... 304 lines cut ...]
[ INFO] [1621351744.014918872]: rtabmap 0.20.9 started...
[ INFO] [1621351744.091932174]: Odom: quality=122, std dev=0.017466m|0.050171rad, update time=0.204247s
[ INFO] [1621351744.108060724]: rtabmap (1): Rate=1.00s, Limit=0.000s, Conversion=0.0132s, RTAB-Map=0.2573s, Maps update=0.0021s pub=0.0001s (local map=1, WM=1)
[ INFO] [1621351744.323706027]: Odom: quality=127, std dev=0.025045m|0.047716rad, update time=0.220512s
[ WARN] [1621351744.497315527]: Failed to meet update rate! Took 0.0067686330000000004617
[ WARN] (2021-05-18 22:29:04.506) OdometryF2M.cpp:529::computeTransform() Registration failed: "Not enough inliers 9/15 (matches=162) between -1 and 11" (guess=xyz=-0.006166,0.002106,-0.036566 rpy=-0.006502,-0.011953,0.000260)
[ WARN] (2021-05-18 22:29:04.506) OdometryF2M.cpp:290::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.006166,0.002106,-0.036566 rpy=-0.006502,-0.011953,0.000260), trying again without a guess.
[ WARN] (2021-05-18 22:29:04.751) OdometryF2M.cpp:539::computeTransform() Trial with no guess succeeded!
[ INFO] [1621351744.761934776]: Odom: quality=68, std dev=0.027458m|0.048989rad, update time=0.432037s
[... cut ...]
[ WARN] (2021-05-18 22:29:05.526) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/15 (matches=8) between -1 and 14"
[ INFO] [1621351745.529719927]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.155653s
[ WARN] (2021-05-18 22:29:05.667) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough features in images (old=2000, new=12, min=15)"
[ INFO] [1621351745.668123628]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.130290s "
Kien

MartyG-RealSense, 2021-05-18:
It looks as though your IMU topics (gyro and accel) are not enabled.  Try this roslaunch instruction:

**roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true**

kiennguyen610, 2021-05-18:
I changed opensource_tracking.launch like this and It seems my IMU topics (gyro and accel) are
[image]
 enabled

MartyG-RealSense, 2021-05-18:
Ok, great!  Does this make any difference to the problems that you have been experiencing, please?

kiennguyen610, 2021-05-18:
Yes, the default setting is rs_camera.launch not rs_d435_camera_with_model.launch so I changed it. Because I am using d435i camera. But, my problem like this :

[image]
[image]
As you see, 2 image from camera is side by side. But the map is not created. So, I don't know the reason why like this

MartyG-RealSense, 2021-05-18:
Thanks for the images.  You have a yellow exclamation-mark warning on the **TF** section of the side-panel.  Can you expand it open and provide an image of what is inside please?

kiennguyen610, 2021-05-18:
This is :
[image]

MartyG-RealSense, 2021-05-18:
Can you show the list of tick-boxes under the **Frames** sub-category of **TF** please?  Like the image below from the D435i SLAM guide.

[image]

kiennguyen610, 2021-05-19:
I'm sorry about the delay. And, This is the list of tick-boxes under the Frames sub-category of TF
[image]

MartyG-RealSense, 2021-05-19:
You have only **camera_link** and **map** ticked, like in the D435i SLAM guide.  So that looks okay.

Your earlier list of items under **Frame** had a yellow warning beside **camera_bottom_screw_frame**.  

[image]

Does that particular warning get fixed if you tick the box beside **camera_bottom_screw_frame** in the list of tick-boxes under **Frames**?

kiennguyen610, 2021-05-19:
[image]
The warning is not fixed if I tick the box beside camera_bottom_screw_frame in the list of tick-boxes under Frames?

MartyG-RealSense, 2021-05-25:
Hi @kiennguyen610  Do you require further assistance with this case, please?  Thanks!

kiennguyen610, 2021-05-29:
Hi @MartyG-RealSense , thank you very much for your assistance. I fixed my problem. I was very stupid for some reason. So my problem was solved. You can closed the issue. Thanks !

MartyG-RealSense, 2021-05-29:
Excellent to hear that you found a solution @kiennguyen610  - thanks very much for the update!  As suggested, I will close the case.

closed 2021-05-29 by MartyG-RealSense