realsenseai/realsense-ros #1872: L515 : An error has occurred during frame callback: map::at

vinesmsuic, 2021-05-13:
# Problem

An Error log message generated in every frame and flood the shell output.
The `align_depth` topics are available and the image can be seen on `rqt_gui`.
Although this problem  did not cause much noticeable error on performance, I found it disturbing and it generates a huge size of log file.
This error only occur when `align_depth` is enabled.

# Environment
Device: Nvidia Jetson Xavier NX, RealSense LiDAR L515
OS: Ubuntu 18.04
ROS Version: ROS Melodic
RealSense SDK Version: LibRealSense SDK v2.41.0
realsense-ros Version: 2.2.21

# To reproduce the Error

My `rs_camera.launch` file
```xml
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="640"/>
  <arg name="confidence_height"   default="480"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="30"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="true"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="1280"/>
  <arg name="color_height"        default="720"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
[... 233 lines cut ...]
[ INFO] [1620918433.599546548]: Expected frequency for color = 30.00000
[ INFO] [1620918433.660883278]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1620918433.721179957]: Expected frequency for confidence = 30.00000
[ INFO] [1620918433.780981504]: setupStreams...
[ INFO] [1620918433.835060875]: insert Depth to L500 Depth Sensor
[ INFO] [1620918433.835536806]: insert Color to RGB Camera
[ INFO] [1620918433.835755907]: insert Infrared to L500 Depth Sensor
[ INFO] [1620918433.835974753]: insert Gyro to Motion Module
[ INFO] [1620918433.836117823]: insert Confidence to L500 Depth Sensor
 13/05 23:07:14,114 WARNING [547821773184] (ds5-motion.cpp:477) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1620918434.130634662]: SELECTED BASE:Depth, 0
[ INFO] [1620918434.187086562]: RealSense Node Is Up!
 13/05 23:07:14,426 WARNING [547376279936] (l500-depth.cpp:455) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.
 13/05 23:07:15,048 WARNING [547643969920] (l500-depth.cpp:455) UVC metadata payloads are not available for stream . Please refer to installation chapter for details.
[ERROR] [1620918435.104512032]: An error has occurred during frame callback: map::at
[ERROR] [1620918435.144468983]: An error has occurred during frame callback: map::at
[ERROR] [1620918435.163327119]: An error has occurred during frame callback: map::at
[ERROR] [1620918435.199078354]: An error has occurred during frame callback: map::at
[ERROR] [1620918435.244310161]: An error has occurred during frame callback: map::at
```
# Approach tried (but failed)

From #1348, using `roslaunch realsense2_camera rs_camera.launch enable_infra:=true` did not solve the issue. The error is still here.

Rebooting the NX without replugging the L515 will make the error message gone, but then the images in the `align_depth` topics are not showing anymore.

doronhi, 2021-05-20:
This error message seems like a bug. Before I start looking into it, could you confirm it happens with the latest version as well? (realsense2_camera version 2.3.0)

vinesmsuic, 2021-05-24:
Removed all the realsense packages, installed RealSenseSDK v2.45.0 and realsense-ros v2.3.0, now the error is gone and everything works perfectly.

Another issue I found is that the ddynamic-reconfigure  package must be cloned to src folder to get the realsense-ros successfully build. Using APT to install dynamic-reconfigure and ddynamic-reconfigure will not work at all.

closed 2021-06-29 by vinesmsuic