realsenseai/realsense-ros #1874: No cuda capable device is detected

sivaeinfo, 2021-05-18:
Actually i have tried to use the dope package in https://github.com/yehengchen/DOPE-ROS-D435

I have installed require packages and dependencies.
In this, they have given two command.
1.roslaunch realsense2_camera rs_rgbd.launch
2. roslaunch dope dope.launch

This gives following error
[image][image]

Please help me to sort out this issues.

MartyG-RealSense, 2021-05-18:
Hi @sivaeinfo   May I first confirm that you have a computer or computing device that is equipped with an Nvidia graphics GPU processor component.  This could include an **Nvida Jetson** board, or a computer that has an Nvidia graphics chip or plug-in video card.  CUDA requires that the computer has an Nvidia GPU component, as it does not support non-Nvidia GPUs.

sivaeinfo, 2021-05-18:
I am having computer system with Nvidia GT218 GeForce 210. I have uninstalled nividia driver 430 and reinstalling using the command (ref: https://askubuntu.com/questions/1105090/cant-change-to-nvidia-proprietary-driver-ubuntu-18-04)

sudo ubuntu-drivers autoinstall

It shows another error 

[image]
[image]

MartyG-RealSense, 2021-05-18:
This is technically an Ubuntu / Nvidia problem rather than a RealSense one.

However, if driver autoinstall failed and the driver that the 210 uses is **430** then you could try installing a specific driver with the command below:

**sudo apt install nvidia-430**

sivaeinfo, 2021-05-19:
I have install driver using
sudo apt install nvidia-driver-430
I want to install cuda 9.2 or more to match my pytorch.
I am not able to find proper way to install cuda 9.2 or more
https://medium.com/@aspiring1/installing-cuda-toolkit-10-0-and-cudnn-for-deep-learning-with-tensorflow-gpu-on-ubuntu-18-04-lts-f7e968b24c98

I think nvidia 430 GeForce 210 is not cuda enabled

MartyG-RealSense, 2021-05-19:
GeForce 210 is listed in specifications as having CUDA support, with 16 CUDA cores.  The official Nvidia specification page for the 210 is at the link below.

https://video-nvidia.com/en-us/series_200/nvidia-geforce-210.html

The **340** driver supports GeForce 200 Series, including 210.

https://www.nvidia.com/Download/driverResults.aspx/81761/en-us

[image]

I researched the question of 210 supporting modern CUDA versions.  The popular opinion was that 210 was too weak and that a more modern Nvidia graphics card should be used.  This may not be practical though if your computer is old too.

closed 2021-05-26 by sivaeinfo

MartyG-RealSense, 2021-05-26:
Okay, thanks very much @sivaeinfo for the update - good luck!