realsenseai/realsense-ros #1875: L515 - Depth Sensor Failure

rwbot, 2021-05-19:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
| **OS** | Ubuntu 18.04.05 LTS |
|**Name**|Intel RealSense L515|
|**Serial Number**|f0190830|
|**Firmware Version**|01.05.05.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003a96b5957|
|**Firmware Update Id**|0003a96b5957|
| **RealSense ROS**  | v2.3.0 |
| **ROS Distro** | Melodic |
| **ROS version** | 1.14.10 |

Unable to view depth information in both realsense-viewer and launching via ROS. Color+Motion work fine in realsense-viewer, but enabling Depth throws this error:
```
Timestamp: 1621399360298.776611
Severity: Error
Description: Fatal error occurred (16)
```
I've tried all the suggestions from the related issue https://github.com/IntelRealSense/realsense-ros/issues/1510 with no luck. Below is the output from `roslaunch realsense2_camera rs_camera.launch initial_reset:=true`

# With `initial_reset:=true`
```
rwbot@carowe:~$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/rwbot/.ros/log/c2f59f42-b856-11eb-a66a-a4fc7772dfcf/roslaunch-carowe-19701.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://carowe:45293/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
[... 119 lines cut ...]
[/camera/realsense2_camera_manager.BaseRealSenseNode::SetBaseStream]: SELECTED BASE:Depth, 0
 18/05 22:00:43,038 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:43,089 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[/camera/realsense2_camera_manager.BaseRealSenseNode::publishTopics]: RealSense Node Is Up!
[/camera/realsense2_camera_manager.BaseRealSenseNode::setBaseTime]: 
[/camera/realsense2_camera_manager.BaseRealSenseNode::setBaseTime]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 18/05 22:00:44,041 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:44,400 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[/camera/realsense2_camera_manager.BaseRealSenseNode::setupErrorCallback]: Hardware Notification:Fatal error occurred (16),1.6214e+12,Error,Hardware Error
 18/05 22:00:45,293 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:46,545 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:47,797 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:49,049 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:50,301 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:51,553 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:52,805 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:54,057 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:55,309 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 22:00:56,561 WARNING [139997839599360] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

Also, I made a rosbag [here](https://github.com/rwbot/realsense_l515_error_rosbag/blob/master/realsense_2021-05-18-23-02-12.bag) when I launched the node, and when i began the rosbag, an additional error was continuously thrown until the rosbag was terminated.
```
[/camera/realsense2_camera_manager.CompressedImage_<std::allocator<void> >::Ptr compressed_depth_image_transport::encodeCompressedDepthImage]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
```

RealSenseSupport, 2021-06-18:
@rwbot 

Did the camera function prior to the errors that you are seeing?  Does the camera work on any other system or it fails in the same way no matter what system/platform is in use?

You may need to submit a ticket here:  https://support.intelrealsense.com/hc/en-us/requests/new

If this is a HW issue, then a RMA may be necessary.

rwbot, 2021-06-21:
> @rwbot
>
> Did the camera function prior to the errors that you are seeing? Does the camera work on any other system or it fails in the same way no matter what system/platform is in use?
>
> You may need [...]

This particular camera hasn't ever worked, we've tried on multiple machines, same error.

RMA?

closed 2021-09-17 by MartyG-RealSense