realsenseai/realsense-ros #1877: NX+D455，Motion Module probelm

liangchen97, 2021-05-19:
Running commands with d455 on Ubuntu 18.04

roslaunch realsense2_ camera rs_ camera.launch

Error returned

Hardware Not ification:Motion Module Failure, 1.62133e + 12, error, hardware error or hardware not ification:Motion Module force pause,1.6214e+12,Error,Hardware Error

IMU related topics are missing in topic list

liangchen97, 2021-05-19:
Here are some details
/nvidia@nvidia-desktop:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/nvidia/.ros/log/761997da-b7b3-11eb-9e72-d8c0a650ee07/roslaunch-nvidia-desktop-17436.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:41941/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 85 lines cut ...]
[ INFO] [1621326704.309444165]: insert Depth to Stereo Module
[ INFO] [1621326704.309615269]: insert Color to RGB Camera   o998
[ INFO] [1621326704.444131963]: RealSense Node Is Up!
[ WARN] [1621326704.547723635]: 
 18/05 16:31:44,549 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:44,599 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:44,651 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:44,802 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:44,852 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:45,004 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:45,054 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:45,206 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 18/05 16:31:45,682 ERROR [547118023040] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 18/05 16:31:46,048 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 18/05 16:32:37,600 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 18/05 16:32:54,665 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 18/05 16:32:59,720 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 18/05 16:33:05,777 WARNING [547734151552] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

liangchen97, 2021-05-19:
vidia@nvidia-desktop:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/nvidia/.ros/log/b96ea40c-b863-11eb-8f2f-d8c0a650ee07/roslaunch-nvidia-desktop-1610.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:42523/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 72 lines cut ...]
[ INFO] [1621402406.876364859]: num_filters: 1
[ INFO] [1621402406.876493981]: Setting Dynamic reconfig parameters.
[ INFO] [1621402407.404387056]: Done Setting Dynamic reconfig parameters.
[ INFO] [1621402407.406507147]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1621402407.408383715]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1621402407.412266485]: setupPublishers...
[ INFO] [1621402407.429280430]: Expected frequency for depth = 30.00000
[ WARN] [1621402408.416962638]: Hardware Notification:Motion Module force pause,1.6214e+12,Error,Hardware Error
[ INFO] [1621402408.467724790]: Expected frequency for color = 30.00000
[ INFO] [1621402408.528112441]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1621402408.593906368]: setupStreams...
[ INFO] [1621402408.618243895]: insert Depth to Stereo Module
[ INFO] [1621402408.618522651]: insert Color to RGB Camera
[ INFO] [1621402408.722678124]: SELECTED BASE:Depth, 0
[ INFO] [1621402408.751175096]: RealSense Node Is Up!
[ WARN] [1621402408.845624590]: 
 19/05 13:33:28,847 WARNING [547717366144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 13:33:28,897 WARNING [547717366144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 13:33:29,049 WARNING [547717366144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 19/05 13:33:29,987 ERROR [547242471808] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 19/05 13:33:30,108 WARNING [547717366144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 13:33:30,201 WARNING [547717366144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 13:33:30,252 WARNING [547717366144] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

liangchen97, 2021-05-19:
my firmware version is 05.12.06.00. 
and min FW Version is 05.12.13.50.
My question is should I update the firmware? If so, to what version?

MartyG-RealSense, 2021-05-19:
Hi @liangchen97  The IMU topics (gyro and accel) are disabled by default.  You can enable them by adding the commands below to the end of your roslaunch instruction:

**enable_gyro:=true enable_accel:=true**

For example:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true**

**control_transfer returned** warnngs can be safely ignored if there are only a small number of them at launch, or they generate at a rate of 1 or 2 per minute after launch.  They are only a problem if they generate continuously, which can be an indication of a serious communication problem with the device.

If you are experiencing continuously generating **control_transfer returned** warnings, you could first try adding **initial_reset:=true** to your roslaunch instruction to perform a reset of the camera at launch:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true initial_reset:=true**

liangchen97, 2021-05-19:
The problem is solved! 

The warning disappears after "rs-fw-update"updates the firmware to 05.12.13.50

Then run roslaunch realsense 2_ camera rs_ camera.launch enable_ gyro:=true enable_ Accel: = true, I found IMU_ info

Thank you for your efficient and useful reply

MartyG-RealSense, 2021-05-19:
You are very welcome @liangchen97 - that is excellent news.  Thanks for the update!

closed 2021-05-25 by MartyG-RealSense