realsenseai/realsense-ros #1881: ROS wrapper input parameters: D435i

Nick-Kou, 2021-05-19:
I am trying to adjust the width and height parameters on launch in attempt to decrease the number of points published for each sensor_msgs/PointCloud2.msg.

Ubuntu 20.04
ROS noetic

nikolas@nikolas-SYS-5039A-I:~$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=320 infra_width:=320 color_width:=320 depth_height:=240 infra_height:=240 color_height:=240
... logging to /home/nikolas/.ros/log/f68d63f2-b8a2-11eb-b100-9fc7fe01b9f1/roslaunch-nikolas-SYS-5039A-I-58267.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nikolas-SYS-5039A-I:41601/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 240
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 320
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 240
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 320
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 69 lines cut ...]
[ INFO] [1621429607.003019745]: RGB Camera was found.
[ INFO] [1621429607.003212090]: Motion Module was found.
[ INFO] [1621429607.003253199]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1621429607.003275580]: Add Filter: pointcloud
[ INFO] [1621429607.003752059]: num_filters: 1
[ INFO] [1621429607.003778416]: Setting Dynamic reconfig parameters.
[ WARN] [1621429607.373638454]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1621429607.385426252]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1621429607.434412975]: Done Setting Dynamic reconfig parameters.
[ INFO] [1621429607.435393166]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1621429607.437647181]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1621429607.437707715]: setupPublishers...
[ INFO] [1621429607.440721735]: Expected frequency for depth = 30.00000
[ INFO] [1621429607.475907680]: Expected frequency for color = 30.00000
[ INFO] [1621429607.494954042]: setupStreams...
[ INFO] [1621429607.505955725]: insert Depth to Stereo Module
[ INFO] [1621429607.506057258]: insert Color to RGB Camera
 19/05 09:06:47,559 WARNING [139929002690304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1621429607.610282468]: SELECTED BASE:Depth, 0
[ INFO] [1621429607.621806457]: RealSense Node Is Up!
[ WARN] [1621429607.680588139]: 
 19/05 09:06:47,681 WARNING [139929002690304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 09:06:47,731 WARNING [139929002690304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 09:06:47,985 WARNING [139929002690304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 09:06:48,035 WARNING [139929002690304] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Nick-Kou, 2021-05-19:
It seems as after I insert the desired parameters for height and width the depth and color stream ignore the input parameters, as they are enabled with higher resolution:

[ INFO] [1621429607.435393166]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1621429607.437647181]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8

I was wondering if there is any parameter or setting I can configure to reduce the number of points in each sensor_msgs/PointCloud2.msg.

MartyG-RealSense, 2021-05-19:
Hi @Nick-Kou   From ROS wrapper 2.2.22 onwards, three factors must be specified in the roslaunch instruction to define a custom configuration - width, height and **FPS**.

If three factors are not provided then the custom configuration is deemed invalid and the default stream settings of the particular camera model are applied.

The **infra** topics are also disabled by default since wrapper version 2.2.15, so should be enabled in the roslaunch with **enable_infra:=true**

According to the data sheet document for the 400 Series cameras, the resolution 320x240 is only supported on the **RGB** stream.  A minimum resolution supported by RGB, depth and infrared is **424x240**

[image]

Overall, your roslaunch instruction may therefore look something like the one below:

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_infra:=true depth_width:=424 infra_width:=424 color_width:=424 depth_height:=240 infra_height:=240 color_height:=240 depth_fps:=30.0 color_fps:=30.0 infra_fps:=30.0**

Also note that if you need to enable IMU topics (accel and gyro) at sone point then you should add the further commands below to the roslaunch instruction:

**enable_accel:=true enable_gyro:=true**

Nick-Kou, 2021-05-19:
Thank you! After launching the following command:

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_infra:=true depth_width:=424 infra_width:=424 color_width:=424 depth_height:=240 infra_height:=240 color_height:=240 depth_fps:=30.0 color_fps:=30 infra_fps:=30

The number of points per message decreased significantly. Also, a temperature warning I was getting before, doesn't show anymore, so it is safe to say this configuration resolved those issues. However, I still receive this warning at least twice a minute: WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

For example:

nikolas@nikolas-SYS-5039A-I:~$ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=30 color_fps:=30
... logging to /home/nikolas/.ros/log/2c4dc12a-b8d6-11eb-b100-9fc7fe01b9f1/roslaunch-nikolas-SYS-5039A-I-66637.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nikolas-SYS-5039A-I:41183/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 240
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 424
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 240
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 424
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 101 lines cut ...]
 19/05 15:16:48,341 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:16:48,392 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:16:48,544 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:16:48,594 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:16:48,746 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:16:48,796 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:17:41,791 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:18:31,878 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 19/05 15:20:03,650 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:20:04,701 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:20:04,752 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/05 15:20:44,832 WARNING [140415365805824] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-05-19:
Great to hear that you had success @Nick-Kou 

If the **control_transfer returned** warning only appears at once or twice per minute after launch then it is safe to ignore it.

MartyG-RealSense, 2021-05-25:
Hi @Nick-Kou  Do you require further assistance with this case, please?  Thanks!

Nick-Kou, 2021-05-25:
Hi, @MartyG-RealSense  everything is resolved thanks.

MartyG-RealSense, 2021-05-25:
Excellent to hear, @Nick-Kou - I will therefore close this case.  Thanks again!

closed 2021-05-25 by MartyG-RealSense