realsenseai/realsense-ros #1882: Whenever I am using filters:=colorizer in RVIZ I am not able to see pointcloud

ranjitkathiriya, 2021-05-19:
Hey there,

> roslaunch realsense2_camera rs_rgbd.launch filters:=colorizer align_depth:=true

In this command, the point cloud is not visible. If I remove filters:=colorizer then point cloud is visible. I have checked in 2-3 version I have faced the same issue. Is there any way that I can see filters:=colorizer and point cloud both in rviz?

MartyG-RealSense, 2021-05-19:
i @ranjitkathiriya  If the problem that you are encountering is that you cannot achieve a color pointcloud in RViz, a RealSense user found that the solution that worked for them was to follow the advice in the link below to adjust the **PointCloud2** settings in RViz and then change the style to **points**.

https://github.com/IntelRealSense/realsense-ros/issues/975#issuecomment-552199690

ranjitkathiriya, 2021-05-20:
Yes, I have tested but it is working perfectly with the `rs_camera.launch` file. I am getting both colour depth and point cloud when I run this following launch file. 

The issue is when I `run rs_rgbd.launch` launch file at that I am facing the following issue as I maintained above. 

```
[ERROR] [1621499692.147319529]: Depth image has unsupported encoding [rgb8]
[ERROR] [1621499697.170020105]: Depth image has unsupported encoding [rgb8]
[ERROR] [1621499702.188284365]: Depth image has unsupported encoding [rgb8]
```

Not able to get point cloud below.  TO run this I have tried the following command but I think it is an internal code issue because it is working perfectly with rs_camera.launch.

ranjitkathiriya, 2021-05-20:
```
roslaunch realsense2_camera rs_rgbd.launch filters:=colorizer align_depth:=true      : Not Working   - No pointCloud - rgb8 Error
roslaunch realsense2_camera rs_rgbd.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR                                                   : Not Working  - No [point CLoud - rgb8 Error

roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR    : Working perfect

roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud align_depth:=true      : Working perfect

```

ranjitkathiriya, 2021-05-20:
Perfect now working! thanks

MartyG-RealSense, 2021-05-20:
Great news @ranjitkathiriya - thanks very much for the update!

ranjitkathiriya, 2021-05-20:
Sorry! @MartyG-RealSense ,

If I am using `roslaunch realsense2_camera rs_rgbd.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR`  at that time I am getting point of is_dense= true but I wanted is_danse:=false. I think `rs_rgbd.launch` I am getting empty points, yes I was getting the pointcloud but they were of unorganized cloud is dense, so is_dense = true.

MartyG-RealSense, 2021-05-20:
I believe that this subject is related to an earlier discussion about organized and unorganized point clouds that we had in December 2020.  In it was stated "An organized cloud has invalid points and so is_dense = false".

https://github.com/IntelRealSense/realsense-ros/issues/1563

You could try setting the point cloud filter parameter **ordered_pc** to be true.  This setting was added in ROS1 wrapper 2.2.22 and ROS2 wrapper 3.1.4.

[image]

ranjitkathiriya, 2021-05-20:
Thanks! got my answer.. was not knowing of this update. once again thanks

closed 2021-05-20 by ranjitkathiriya