realsenseai/realsense-ros #1883: what can i operate with IMU data and D435?

SU1997, 2021-05-20:
there is no T265 camera. But i have  imu data and a D435i camera,how to edit the source code to correctly run?

SU1997, 2021-05-20:
here are the ros topics:
/GLxxBatteryCmd
/GLxxBatteryMsg_msg
/SaveImage
/action_key_frame_lists
/adjust_status
/avoid_area
/avoid_mode
/avoid_status
/barcode_timeout
/battery
/battery_info
/behavior_status
/calib_process
/ceil_barc
/ceil_barc_offset
/ceil_barc_timeout
/charging_status
/cmd/zynq_cmd
/cmd_board
/cmd_vel
/cmd_vel_manual
/component_test
/config_robot_info
/control_action_mode
/control_mode
/diagnostics
/drivers/realsense/accel/imu_info0
/drivers/realsense/depth/camera_info0
/drivers/realsense/depth/image_rect_raw0
/drivers/realsense/depth/image_rect_raw0/compressed
/drivers/realsense/depth/image_rect_raw0/compressed/parameter_descriptions
/drivers/realsense/depth/image_rect_raw0/compressed/parameter_updates
/drivers/realsense/depth/image_rect_raw0/compressedDepth
/drivers/realsense/depth/image_rect_raw0/compressedDepth/parameter_descriptions
/drivers/realsense/depth/image_rect_raw0/compressedDepth/parameter_updates
/drivers/realsense/depth/image_rect_raw0/theora
/drivers/realsense/depth/image_rect_raw0/theora/parameter_descriptions
/drivers/realsense/depth/image_rect_raw0/theora/parameter_updates
/drivers/realsense/extrinsics/depth_to_infra10
/drivers/realsense/extrinsics/depth_to_infra20
/drivers/realsense/extrinsics/depth_to_robot0
/drivers/realsense/geekvision_realsense_manager/bond
/drivers/realsense/gyro/imu_info0
/drivers/realsense/imu0
/drivers/realsense/infra1/camera_info0
/drivers/realsense/infra1/image_rect_raw0
/drivers/realsense/infra1/image_rect_raw0/compressed
/drivers/realsense/infra1/image_rect_raw0/compressed/parameter_descriptions
/drivers/realsense/infra1/image_rect_raw0/compressed/parameter_updates
/drivers/realsense/infra1/image_rect_raw0/compressedDepth
/drivers/realsense/infra1/image_rect_raw0/compressedDepth/parameter_descriptions
/drivers/realsense/infra1/image_rect_raw0/compressedDepth/parameter_updates
/drivers/realsense/infra1/image_rect_raw0/theora
/drivers/realsense/infra1/image_rect_raw0/theora/parameter_descriptions
/drivers/realsense/infra1/image_rect_raw0/theora/parameter_updates
/drivers/realsense/infra2/camera_info0
/drivers/realsense/infra2/image_rect_raw0
/drivers/realsense/infra2/image_rect_raw0/compressed
/drivers/realsense/infra2/image_rect_raw0/compressed/parameter_descriptions
/drivers/realsense/infra2/image_rect_raw0/compressed/parameter_updates
/drivers/realsense/infra2/image_rect_raw0/compressedDepth
/drivers/realsense/infra2/image_rect_raw0/compressedDepth/parameter_descriptions
/drivers/realsense/infra2/image_rect_raw0/compressedDepth/parameter_updates
/drivers/realsense/infra2/image_rect_raw0/theora
/drivers/realsense/infra2/image_rect_raw0/theora/parameter_descriptions
/drivers/realsense/infra2/image_rect_raw0/theora/parameter_updates
/drivers/realsense/motion_module0/parameter_descriptions
/drivers/realsense/motion_module0/parameter_updates
/drivers/realsense/rgb_camera0/parameter_descriptions
[... 135 lines cut ...]
/sync_map_info
/sync_submap
/synchronized_points
/system_state
/target_map
/task4server
/task_action
/task_log
/task_status
/temp_path
/test_submap_map
/tf
/tf_static
/tmp_path
/unbalance_load_status
/updateFeatureMap
/updates_command
/updates_result
/vis_barc
/visualization_marker
/visualization_points
/vxslam_manager/bond
/waypoints
/weigh_reset
/window_points

MartyG-RealSense, 2021-05-20:
Hi @SU1997  Could you tell me more please about the problem that you are having?

For example, if you have a **D435i** camera and the IMU topics (accel and gyro) are not publishing then they can be published by adding the commands below onto the end of your roslaunch instruction.

**enable_gyro:=true enable_accel:=true**

For example:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true**

If it were a D435 camera that you have then IMU information cannot be obtained from that camera model because it does not have an IMU component inside it.

SU1997, 2021-05-20:
> Hi @SU1997 Could you tell me more please about the problem that you are having?
>
> For example, if you have a **D435i** camera and the IMU topics (accel and gyro) are not publishing then they can be [...]

thank you for your reply, i have D435i camera and recorded both depth data and imu data.
but the source code use T265 and D400 default,so my question is where to edit the source code to  match my data?
here is my rosbag info:

types:       nav_msgs/Odometry      [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/CameraInfo [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/Image      [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu        [6a62c6daae103f4ff57a132d6f95cec2]
             tf2_msgs/TFMessage     [94810edda583a504dfda3829e70d7eec]
topics:      /drivers/realsense/depth/camera_info0         909 msgs    : sensor_msgs/CameraInfo
             /drivers/realsense/depth/image_rect_raw0      909 msgs    : sensor_msgs/Image     
             /drivers/realsense/imu0                     11856 msgs    : sensor_msgs/Imu       
             /drivers/realsense/infra1/camera_info0        909 msgs    : sensor_msgs/CameraInfo
             /drivers/realsense/infra1/image_rect_raw0     907 msgs    : sensor_msgs/Image     
             /drivers/realsense/infra2/camera_info0        907 msgs    : sensor_msgs/CameraInfo
             /drivers/realsense/infra2/image_rect_raw0     907 msgs    : sensor_msgs/Image     
             /geekvision/galaxy/img                        605 msgs    : sensor_msgs/Image     
             /imu                                         3017 msgs    : sensor_msgs/Imu       
             /odom                                        3017 msgs    : nav_msgs/Odometry     
             /tf                                          9901 msgs    : tf2_msgs/TFMessage     (5 connections)
             /tf_static                                      1 msg     : tf2_msgs/TFMessage

MartyG-RealSense, 2021-05-20:
I am unable to answer questions related to the T265, and answered this case because your message said "there is no T265 camera". I can continue with the case though if it only involves the D435i. 

I do not understand your question about modifying the source code to match your data.   Could you provide more information about what you are trying to achieve, please?

SU1997, 2021-05-20:
> I am unable to answer questions related to the T265, and answered this case because your message said "there is no T265 camera". I can continue with the case though if it only involves the D435i.
>
> I [...]

yes,i mean i only have D435i,and i think i should change some settings in the source code and build it again.

MartyG-RealSense, 2021-05-20:
Thanks very much @SU1997 for the update - please let me know the outcome of your next build.  Good luck!

MartyG-RealSense, 2021-05-28:
Hi @SU1997  Do you have an update about this case that you can provide, please?  Thanks!

SU1997, 2021-05-28:
> Hi @SU1997 Do you have an update about this case that you can provide, please? Thanks!

thanks and i have solved this problem. im a rookie in ros, this took me several days to solve it but fortunately it works finally.thank you for your patiently reply.

MartyG-RealSense, 2021-05-28:
It is great to hear @SU1997 that you developed a solution.  :)

As your issue is solved, I will close this case.  Thanks again!

closed 2021-05-28 by MartyG-RealSense