realsenseai/realsense-ros #1885: Catkin Make error for a ROS Wrapper of Realsense Driver 2.41

ruddradev, 2021-05-20:
Hello,

I need some help to understand what the issue might be when I try to build the realsense wrapper for ROS Melodic, Realsense Driver Version 2.41.0. 

I used the files specified in - https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.21

The I put them in an empty catkin workspace src folder, and I try the commands provided in the readme - 

`catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release`

However I come across a few missing targets. Below is the part of the message that I encounter. Would appreciate any help.

```
CMake Error at /usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake:39 (set_property):
  set_property could not find TARGET realsense2::tm.  Perhaps it has not yet
  been created.
Call Stack (most recent call first):
  /usr/local/lib/cmake/realsense2/realsense2Targets.cmake:82 (include)
  /usr/local/lib/cmake/realsense2/realsense2Config.cmake:35 (include)
  realsense2_camera/CMakeLists.txt:43 (find_package)

CMake Error at /usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake:40 (set_target_properties):
  set_target_properties Can not find target to add properties to:
  realsense2::tm
Call Stack (most recent call first):
  /usr/local/lib/cmake/realsense2/realsense2Targets.cmake:82 (include)
  /usr/local/lib/cmake/realsense2/realsense2Config.cmake:35 (include)
  realsense2_camera/CMakeLists.txt:43 (find_package)

CMake Error at /usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake:49 (set_property):
  set_property could not find TARGET realsense2::uvc_fw.  Perhaps it has not
  yet been created.
Call Stack (most recent call first):
  /usr/local/lib/cmake/realsense2/realsense2Targets.cmake:82 (include)
  /usr/local/lib/cmake/realsense2/realsense2Config.cmake:35 (include)
  realsense2_camera/CMakeLists.txt:43 (find_package)

CMake Error at /usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake:50 (set_target_properties):
  set_target_properties Can not find target to add properties to:
  realsense2::uvc_fw
Call Stack (most recent call first):
  /usr/local/lib/cmake/realsense2/realsense2Targets.cmake:82 (include)
  /usr/local/lib/cmake/realsense2/realsense2Config.cmake:35 (include)
  realsense2_camera/CMakeLists.txt:43 (find_package)

-- Create Release Build.
-- realsense2_camera: 2 messages, 0 services
-- Configuring incomplete, errors occurred!
See also "/home/double/realsense_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/double/realsense_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

MartyG-RealSense, 2021-05-20:
Hi @ruddradev  May I first ask whether you have successfully installed librealsense SDK **2.41.0** and tested it as working correctly by running its **RealSense Viewer** program?

Once librealsense has been installed, and if ROS Melodic has also been installed, then the next step would be to build the **2.2.21** ROS wrapper.  The wrapper should not be installed before librealsense has first been successfuly installed.

ruddradev, 2021-05-21:
Hi @MartyG-RealSense,

Thanks for your response. It appears some cmake files from an earlier installation was causing the build to fail. I worked with our hardware provider to delete them and now the wrapper is building fine. We can close this issue now.

closed 2021-05-21 by ruddradev

MartyG-RealSense, 2021-05-21:
Excellent news @ruddradev that you found a solution - thanks very much for the update!