realsenseai/realsense-ros #1887: Update Results in Poor RVIZ Performance, Camera Issues

jbeck28, 2021-05-20:
Hello, 

I'm using a realsense D415, and Ubuntu 18.04 with kernel 5.4.0-73, and up until recently everything was working fine. 

I updated my local version of this repository (which I had cloned into my catkin workspace) using `git pull`. I have no idea what version I was using before, but I had originally cloned the repository some time in late 2019. 

Now, when I visualize the pointcloud in RViz (use rs_rgbd.launch, with some associated stuff), I get roughly 2fps whereas I was getting a solid 30 before. 

Furthermore, the application I use the camera with to detect chess boards now complains of NaN's, and is never able to find the chess board. There appears to be missing data in visualized point clouds. 

Any help would be appreciated, I believe this little piece of information will be helpful: 

`[ INFO] [1621548842.754968103]: RealSense ROS v2.3.0
[ INFO] [1621548842.755046059]: Built with LibRealSense v2.45.0
[ INFO] [1621548842.755140192]: Running with LibRealSense v2.45.0
`

Thanks in advance!

MartyG-RealSense, 2021-05-21:
Hi @jbeck28   As a starting point in investigating your case, could you look at the discussion in the link below please to see whether the advice is relevant to your particular issues?

https://github.com/IntelRealSense/realsense-ros/issues/1882

jbeck28, 2021-05-22:
I didn't see anything in particular that would help in that discussion, but I also could be ignorant. I've been using `rs_rgbd.launch` since I realized I needed an organized point cloud. Is there a substantive difference between `rs_camera` and `rs_rgbd` other than this organized vs unorganized pointcloud? I know this is a bit off topic, but I've not tried any other launch files for this reason, and that's mostly due to ignorance.

Another thing to note, I reverted librealsense2 back to 2.3, and the version of this package to 2.2.10 and performance has been restored. 

A bit of background before I explain this other thing I've noticed. About a year ago I recorded a bunch of poses on a 6 dof industrial robot arm that all make the camera (mounted to the end effector) able to see a chessboard that can be repeatably mounted. Before (and now that I've reverted to older versions) all of these poses were able to see all of the corners of the chessboard. When using version 2.45 of librealsense2 and the latest version of this repository with the same launch commands, the field of view seems to have shrunk to the point where I cannot see all of the chessboard in many of these poses. 

Before reverting to a previous version, I noticed that the nodelet was using some 50% of my CPU, now that is down to about 18%. This could be related to the low FPS problem.

I hope some of this information is actually useful... and I hope I answered your question. Thanks for your support on this!

- Josh

MartyG-RealSense, 2021-05-22:
Hi @jbeck28   Information about the difference between rs_launch and rs_rgbd is provided in the link below by a RealSense user and confirmed by a RealSense team member.

https://github.com/IntelRealSense/realsense-ros/issues/893

Which stream resolutions are you using in ROS wrapper 2.2.10 please?  A notable recent change in the wrapper is that from 2.2.22 onwards, a new profile system was introduced where custom stream configurations require three factors to be provided (stream width, stream height and FPS).  If those three factors are not provided then the requested configuration is deemed invalid and the default configuration of the particular camera model is applied instead at launch (for example, 848x480 at 30 FPS for depth  and 1280x720 at 30 FPS for color on D435 / D435i).

Before that, another previous notable change was in 2.2.15 onwards, where IMU, infrared, fisheye and pose topics were disabled by default and needed to be enabled in the launch in order for those topics to be published.

MartyG-RealSense, 2021-06-03:
Case closed due to no further comments received.

closed 2021-06-03 by MartyG-RealSense

jbeck28, 2021-06-07:
Hi @MartyG-RealSense 

Sorry for the egregious delay on this, I took a week off work and then last week was rather crazy. I'll continue troubleshooting this eventually, at which point I'll start commenting properly. Just wanted to keep you apprised, thanks for your support thus far. 

- Josh

MartyG-RealSense, 2021-06-07:
Thanks very much @jbeck28 for the update.  Please do feel free to continue your case when you are ready. Good luck!