realsenseai/realsense-ros #1888: rosbag does not record the full tf tree

Mikor-mkr, 2021-05-21:
Hi people,
thanks for taking the time to check my post. I launch t265 and d435 from 2 separate launch files. 
`roslaunch realsense2_camera rs_t265.launch  unite_imu_method:=linear_interpolation`

`roslaunch realsense2_camera test_v2.launch serial_no:=048122070613 initial_reset:=true align_depth:=true enable_sync:=true  filters:=temporal depth_width:=640 depth_height:=480 infra_width:=640 infra_height:=480 color_width:=640 color_height:=480 camera:=D435` 

When I run `rosrun tf view_frames`, the entire tf tree looks fine. But when I record the bag with the /tf and /tf_static, the tree is not complete, any idea why is that happening?

doronhi, 2021-05-23:
I would suspect it is a matter of timing between the publication of /tf and recording by the rosbag.
Could you try `roslaunch realsense2_camera rs_t265.launch unite_imu_method:=linear_interpolation tf_publish_rate:=1.0` ?

Mikor-mkr, 2021-05-23:
Thank you for your response, but I didn't have any success. What could I do to get you more info about my case?

closed 2026-10-23 by MartyG-RealSense