realsenseai/realsense-ros #1889: About hardware sync between two d435i

vislero, 2021-05-21:
I am trying to synchronize hardware between two d435i.

First,we connected the synchronization cable according to (https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration) and set the master and slave **inter_cam_sync_mode** as  #1451 ,then we can checked the pulse output of the master's sync pin.

However, when I record the rosbag and extract the image(topic : infra1/image_rect_roaw from cam1 and cam2) from the bag, the timestamp between the camera images is different as shown below.

[image]

Moreover, when calculating the offset between the timestamps,  it seems that the two offsets are not constant.

And we followed many issues and white paper (below)
	_- base(hardware sync): https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration
		- base2(gen locak scheme) : https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras
		- pin setup : https://vcl3d.github.io/VolumetricCapture/docs/hardware/rs2_hardware/
		- git issue 1 : https://github.com/IntelRealSense/realsense-ros/issues/1451
        - git issue 2 : https://github.com/IntelRealSense/realsense-ros/issues/1632_

but not worked welll...

and below is our launch script 

```
<launch>
  <arg name="serial_no_camera1"    			default="048322070608"/> 			<!-- Note: Replace with actual serial number -->
  <arg name="serial_no_camera2"    			default="048122071544"/> 			<!-- Note: Replace with actual serial number -->
  <arg name="camera1"              			default="master"/>		<!-- Note: Replace with camera name -->
  <arg name="camera2"              			default="slave"/>		<!-- Note: Replace with camera name -->
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>
  

  <arg name="json_file_path"      default=""/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="false"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="63"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>
[... 105 lines cut ...]
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      
    </include>
  </group>

    <rosparam> /master/stereo_module/inter_cam_sync_mode : 1 </rosparam>
    <rosparam> /slave/stereo_module/inter_cam_sync_mode : 2 </rosparam>
    
    <rosparam> /master/auto_exposure_priority: false </rosparam>
    <rosparam> /slave/auto_exposure_priority: false </rosparam>
    
    <rosparam> /master/auto_exposure : true </rosparam>
    <rosparam> /slave/auto_exposure : true </rosparam>
    
</launch>

```

could you please let me to know synchronize multiple d-435i? thank you.

MartyG-RealSense, 2021-05-21:
Hi @vislero  The paper https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration (sync modes 1 and 2, Master and Slave) involves hardware sync of **depth** frame timestamps rather than infrared.

Would it be possible to record a depth-stream bag and check the depth topic, please?

Also bear in mind that if hardware sync is working correctly then over time the timestamps will slowly drift apart.  If hardware sync is **not** working then they will remain perfectly aligned.  This subject is discussed in the section of the multiple camera paper linked to below, in the paragraph that is bolded **Now to the somewhat counter intuitive aspect of time stamps**.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-3-multi-camera-programming

doronhi, 2021-05-23:
Could you also try to turn the global_timestamp off?

vislero, 2021-05-25:
@MartyG-RealSense , @doronhi 

Thanks for the kind advice.
--------

Prior to answering, We had confirmed that librealsense and d-435i firmware version is latest, but realsense-ros version was v.2.2.18 
So, we updated the realsense-ros version to the latest(v2.3.0) and then proceeded this test again.

---------

0. First of all, the global timestamp is the key solution to this problem. When we turn off the global timestamp, we can get a very small offset between slave and master and the offsets are seem to constant.

1. We checked the offset depth frame in the rosbag and calculated the timestamp offset between the depth frames of both cameras. However, the results were the same when we tested them.

2. When **inter_cam_sync_mode** is turned on, the average and variance of the offset is smaller than when inter_cam_sync_mode is turned off. (Nearly 10000 times smaller) But couldn't we get the same timestamp data between these camera frames?

doronhi, 2021-05-25:
I didn't understand what you did in (1) or what is the point you try to convey. Could you rephrase?

2. What is the difference between the timestamps from 2 different devices now? Are you asking if they could be totally equal (diff == 0)?

vislero, 2021-05-25:
1. I understood  @MartyG-RealSense as saying that infrared stream may not synced but depth stream will be synced when we follow hardware sync method.

2. Yes your question is correct. I checked the infra1 and infra2 stream in each d-435i have totally equal timestamp. so i asked the infra between cameras could have the same timestamp.

thank you.

MartyG-RealSense, 2021-06-06:
Case closed due to no further comments received.

closed 2021-06-06 by MartyG-RealSense

iamchoking, 2022-01-14:
> First of all, the global timestamp is the key solution to this problem. When we turn off the global timestamp, we can get a very small offset between slave and master and the offsets are seem to constant.

Hello! I am sorry to ask this here, but what is the meaning of "Global Timestamp" in this context, and how does one turn it off?
Additionally, is this answer also relevant in ROS2 Foxy? If not, I will open another issue.

I am also having trouble syncing up two D435i's in ROS-foxy launch.