realsenseai/realsense-ros #1890: Pointcloud missing color on the sides

ivadimi, 2021-05-21:
**Description**
Pointcloud, visualized in rviz, has some missing or incorrect color on a side and a bit at the bottom.
[image]

**Launch file snippet**
`<include file="$(find realsense2_camera)/launch/rs_camera.launch">
<arg name="json_file_path" value="/home/user/catkin_ws/src/package/cfg/HighResHighAccuracyPreset.json"/>
<arg name="filters" value="pointcloud,temporal,spatial" />
<arg name="clip_distance" value="0.5" />
</include> `

What could be the potential cause of this? Thank you in advance.

MartyG-RealSense, 2021-05-21:
Hi @ivadimi   The RealSense ROS wrapper has a pointcloud filter setting called **allow_no_texture_points** that has the following description:

**************

The depth FOV and the texture FOV are not similar.  By default, pointcloud is limited to the section of depth containing the texture.  You can have a full depth to pointcloud, coloring the regions beyond the texture with zeros, by setting allow_no_texture_points to true.

**********

Could you try setting **allow_no_texture_points** to True, please by adding the command below to your roslaunch instruction:

**allow_no_texture_points:=true**

It is also worth mentioning that if you feel that you lose excessive detail from the image by using the High Accuracy preset then the **Medium Density** preset may help, as it provides as balance between accuracy and the amount of detail on the image.

ivadimi, 2021-05-21:
Hi @MartyG-RealSense .Thank you for your answer.

Unfortunately, I've already tried setting allow_no_texture_points to true and it didn't result in any change.
I will be able to try again on Tuesday and report again, in case it suddenly works.

MartyG-RealSense, 2021-05-21:
Okay, I look forward to continuing the case with you when you are ready.  Good luck!

ivadimi, 2021-05-25:
Just in case, I did a gold factory reset (for camera) and updated to latest ros-wrapper version.

Here is an updated snippet of a launch file and the corresponding results.
`    <include file="$(find realsense2_camera)/launch/rs_camera.launch"> 
        <arg name="json_file_path" value="$(arg HighResMedDensityPreset)"/>
        <arg name="filters" value="pointcloud,temporal,spatial" />
        <arg name="ordered_pc" value="true"/>
        <arg name="clip_distance" value="0.5" />
        <arg name="allow_no_texture_points" value="true"/>
    </include> `

[image]

I've also tried setting "ordered_pc" to true. It produced the following result: (now the texture flickers as opposed to being static)

[image]

Setting "alighed_depth" to true produces the desired result.

[image]

However I am not sure whether aligning depth to color will result in any distance measurement inaccuracies to the pointcloud.

MartyG-RealSense, 2021-05-25:
In general, aligning depth to color can have a positive influence on depth measurement, as it makes it easier to distinguish background pixels from foreground pixels.