realsenseai/realsense-ros #1893: can not change fps in D435i

Majed-Alsubaie, 2021-05-25:
i am trying to change fps to 15 hz by running:
`
roslaunch realsense2_camera rs_camera.launch     align_depth:=true     unite_imu_method:="linear_interpolation"     enable_gyro:=true      enable_accel:=true initial_reset:=true color_fps:=15  infra_fps:=15 depth_fps:=15`

i got  
```
started roslaunch server http://majed:43513/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 89 lines cut ...]
[ INFO] - /camera/realsense2_camera_manager: [1621934646.712350615] gyro stream is enabled - fps: 200
[ INFO] - /camera/realsense2_camera_manager: [1621934646.712373517] accel stream is enabled - fps: 63
[ INFO] - /camera/realsense2_camera_manager: [1621934646.712392614] setupPublishers...
[ INFO] - /camera/realsense2_camera_manager: [1621934646.714081332] Expected frequency for depth = 30.00000
[ INFO] - /camera/realsense2_camera_manager: [1621934646.728547072] Expected frequency for color = 30.00000
[ INFO] - /camera/realsense2_camera_manager: [1621934646.739767627] Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] - /camera/realsense2_camera_manager: [1621934646.750128050] Start publisher IMU
[ INFO] - /camera/realsense2_camera_manager: [1621934646.750741528] setupStreams...
[ INFO] - /camera/realsense2_camera_manager: [1621934646.750984142] insert Depth to Stereo Module
[ INFO] - /camera/realsense2_camera_manager: [1621934646.751019204] insert Color to RGB Camera
[ INFO] - /camera/realsense2_camera_manager: [1621934646.751040053] insert Gyro to Motion Module
[ INFO] - /camera/realsense2_camera_manager: [1621934646.751061004] insert Accel to Motion Module
 25/05 12:24:06,751 WARNING [140465646323456] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] - /camera/realsense2_camera_manager: [1621934646.807625619] SELECTED BASE:Depth, 0
[ INFO] - /camera/realsense2_camera_manager: [1621934646.817299356] RealSense Node Is Up!
[ WARN] - /camera/realsense2_camera_manager: [1621934646.846045953] 
 25/05 12:24:06,920 WARNING [140465252837120] (sensor.cpp:994) HID timestamp not found, switching to Host timestamps.
```

Where the frequencies still 30hz instead of 15 

_**[ INFO] - /camera/realsense2_camera_manager: [1621934646.710433655] **depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16**
[ INFO] - /camera/realsense2_camera_manager: [1621934646.711324488] **color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8****_ 

Thanks

doronhi, 2021-05-25:
You cannot re-define just part of a desired profile. If you re-define fps you must also define the width and the height:
`roslaunch realsense2_camera rs_camera.launch align_depth:=true unite_imu_method:="linear_interpolation" enable_gyro:=true enable_accel:=true color_fps:=15 infra_fps:=15 depth_fps:=15 color_width:=1280 color_height:=720 depth_width:=848 depth_height:=480`

If you are indeed interested in the infrared streams, you must also add the following parameters: `enable_infra1:=true enable_infra2:=true`

MartyG-RealSense, 2021-05-25:
@doronhi   Thanks very much!

@Majed-Alsubaie   The advice of @doronhi above is correct.  Since ROS wrapper 2.2.22 onwards, custom stream definitions should include three factors (width, height and FPS).  If these details are not provided then the launch deems the custom configuration invalid and instead applies the default configuration of the particular camera model being used.

In regard to enabling the infrared streams, they have been disabled by default since wrapper 2.2.15 onwards, hence the need to enable them in order for the infrared topics to be published.

MartyG-RealSense, 2021-05-31:
Hi @Majed-Alsubaie   Do you require further assistance with this case, please?  Thanks!

Majed-Alsubaie, 2021-06-01:
@MartyG-RealSense  @doronhi 
Thanks ,,, it works well

closed 2021-06-01 by Majed-Alsubaie

MartyG-RealSense, 2021-06-01:
Excellent news @Majed-Alsubaie - thanks very much for the update!