realsenseai/realsense-ros #1894: Using default NodeHandle instead of private NodeHandle for setting parameters

Marenix, 2021-05-25:
Hi. I've been going over the code for the wrapper, and noticed that in base_realsense_node.cpp in function publishTopics, you're giving function registerDynamicReconfigCb _node_handle (which is a default node handle). This function basically goes on to setting parameters. What I'm wondering is why you used a public node handle? As far as I know, it's recommended to use a private node handle when working with parameters, so I'm curious if there's a reason you used a public node handle specifically?

doronhi, 2021-05-26:
No reason that I am aware of. This is an old part of the code. If I am convinced there is a good enough reason to change I'll change.

Marenix, 2021-05-26:
It would make setting default parameters from .launch file like this possible (also, adding the appropriate lines in nodelet.launch.xml):
[image]
[image]

It might not be that important for dynamic parameters (like exposure in my example), since they can be set after the streams start, but the problem is with parameters like inter_cam_sync_mode which can't be set once the streams start. 
I tested the default code with this .launch file and parameters weren't changed (looked at the stream in rviz just to make sure the inter_cam_sync_mode wasn't set):
[image]
[image]

Then I changed to a private node handle in function publishTopics(passing _pnh to functions called in publishTopics instead of _node_handle), and the parameters were set (exposure to 500 and inter_cam_sync_mode to 4, no image in rviz):
[image]
[image]

doronhi, 2021-05-27:
That sounds very interesting. Could you submit a PR so I could test it more thoroughly?
It might take me a little while to get around to it but it sounds important.

Marenix, 2021-05-27:
I submitted a [PR#1901](https://github.com/IntelRealSense/realsense-ros/pull/1901).

RealSenseSupport, 2021-12-06:
@Marenix I saw you closed https://github.com/IntelRealSense/realsense-ros/pull/1901. Can this ticket also be closed? Thanks!

Marenix, 2021-12-06:
Yes, it can be closed. Thanks.

closed 2021-12-06 by Marenix