realsenseai/realsense-ros #1895: Can't start depth stream

aPonza, 2021-05-25:
Similarly to #1533, I'm receiving error `Depth stream start failure,1.62195e+12,Error,Hardware Error` from my D435. In realsense-viewer, the depth camera doesn't work either. As you can see from the following output, I already tried `initial_reset:=true`. The output from `rostopic echo /camera/depth/image_rect_raw` is empty too. I don't think I have other information that could help understand the issue that I could know before asking you. Can you help?

<details>
  <summary>roslaunch output</summary>

```
$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/ap/.ros/log/916d8bda-bd63-11eb-a746-d45ddf1a2746/roslaunch-ap-NUC7i7DNKE-7240.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ap-NUC7i7DNKE:38547/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 93 lines cut ...]
[ INFO] - [/camera/realsense2_camera_manager::FrequencyDiagnostics::FrequencyDiagnostics::34]: Expected frequency for depth = 30.00000
[ INFO] - [/camera/realsense2_camera_manager::FrequencyDiagnostics::FrequencyDiagnostics::34]: Expected frequency for color = 30.00000
[ INFO] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupStreams::1813]: setupStreams...
[ INFO] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupStreams::1835]: insert Depth to Stereo Module
[ INFO] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupStreams::1835]: insert Color to RGB Camera
[ INFO] - [/camera/realsense2_camera_manager::BaseRealSenseNode::SetBaseStream::2039]: SELECTED BASE:Depth, 0
[ INFO] - [/camera/realsense2_camera_manager::BaseRealSenseNode::publishTopics::256]: RealSense Node Is Up!
[ WARN] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setBaseTime::1787]: 
[ WARN] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupErrorCallback::228]: Hardware Notification:REC error,1.62195e+12,Error,Hardware Error
[ WARN] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupErrorCallback::228]: Hardware Notification:Depth stream start failure,1.62195e+12,Error,Hardware Error
[ WARN] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupErrorCallback::228]: Hardware Notification:Depth stream start failure,1.62195e+12,Error,Hardware Error
[ WARN] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupErrorCallback::228]: Hardware Notification:Depth stream start failure,1.62195e+12,Error,Hardware Error
[ WARN] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupErrorCallback::228]: Hardware Notification:Depth stream start failure,1.62195e+12,Error,Hardware Error
 25/05 16:15:03,371 WARNING [139679234316032] (backend-v4l2.cpp:1122) Frames didn't arrived within 5 seconds
[ WARN] - [/camera/realsense2_camera_manager::BaseRealSenseNode::setupErrorCallback::228]: Hardware Notification:Depth stream start failure,1.62195e+12,Error,Hardware Error
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

</details>

aPonza, 2021-05-25:
It came to me that issues started after updating software/firmware/library, unfortunately I don't quite remember which, but this could help.

MartyG-RealSense, 2021-05-25:
Hi @aPonza   A good starting point in investigating your case will be to get the **2.45.0** librealsense SDK and the RealSense Viewer working first, as this will provide a solid foundation for then getting the RealSense ROS wrapper working.  Could you provide information about the questions below, please?

 Are you using the Windows or Linux operating system?

Which computer / computing device are you using the camera with (Jetson, Raspberry Pi, desktop / laptop PC, etc)

MartyG-RealSense, 2021-05-26:
Thanks very much for the details.  Do you know which camera firmware driver is currently installed in your camera, please?  If you do not know then you can find it by left-clicking on the **Info** option in the options side-panel of the Viewer.

[image]

aPonza, 2021-05-27:
I'm on 05.12.13.50, sorry for the delay, had to come back to the office first.

aPonza, 2021-05-27:
BTW this is the log from:
1) Starting realsense-viewer
2) Enabling RGB stream
3) Enabling Stereo Module
4) Let some seconds pass
5) Closing the Stereo Module

<details>
  <summary>realsense-viewer log</summary>

```
11:25:19 [Info]  - librealsense version: 2.45.0

11:25:19 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Framebuffer size changed to 1855 x 1056
11:25:19 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Window size changed to 1855 x 1056
11:25:19 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Scale Factor is now 1
11:25:20 [Info]  - Intel RealSense D435 was selected as a default device
11:25:20 [Info]  - No online SW / FW updates available
11:25:20 [Info] /home/ap/git/librealsense/src/context.cpp:382 - Found 1 RealSense devices (mask 0xfe)
11:25:24 [Info]  - Starting streaming of Color...
11:25:24 [Info] /home/ap/git/librealsense/src/sensor.cpp:1416 - Request: RGB8 Color, 
Resolved to: YUYV Color, 
11:25:24 [Info] /home/ap/git/librealsense/src/linux/backend-v4l2.cpp:1569 - Video node was successfully configured to YUYV format, fd 9
11:25:24 [Info] /home/ap/git/librealsense/src/linux/backend-v4l2.cpp:1571 - Trying to configure fourcc YUYV
11:25:24 [Info] /home/ap/git/librealsense/src/metadata-parser.h:329 - Frame counter reset
11:25:24 [Info] /home/ap/git/librealsense/src/metadata-parser.h:329 - Frame counter reset
11:25:24 [Info] /home/ap/git/librealsense/src/metadata-parser.h:329 - Frame counter reset
11:25:25 [Info]  - Starting streaming of Depth...
11:25:25 [Info] /home/ap/git/librealsense/src/sensor.cpp:1416 - Request: Z16 Depth, 
Resolved to: Z16 Depth, 
11:25:25 [Info] /home/ap/git/librealsense/src/linux/backend-v4l2.cpp:1569 - Video node was successfully configured to Z16  format, fd 12
11:25:25 [Info] /home/ap/git/librealsense/src/linux/backend-v4l2.cpp:1571 - Trying to configure fourcc Z16 
11:25:27 [Info] /home/ap/git/librealsense/src/types.h:1104 - Depth stream start failure
11:25:27 [Error] /home/ap/git/librealsense/common/notifications.cpp:514 - Depth stream start failure
11:25:29 [Info] /home/ap/git/librealsense/src/types.h:1104 - Depth stream start failure
11:25:30 [Info] /home/ap/git/librealsense/src/types.h:1104 - Depth stream start failure
11:25:30 [Warn] /home/ap/git/librealsense/src/linux/backend-v4l2.cpp:1122 - Frames didn't arrived within 5 seconds
11:25:30 [Info] /home/ap/git/librealsense/src/types.h:1104 - Frames didn't arrived within 5 seconds
11:25:30 [Warn] /home/ap/git/librealsense/common/notifications.cpp:514 - Frames didn't arrived within 5 seconds
11:25:31 [Info] /home/ap/git/librealsense/src/types.h:1104 - Depth stream start failure
11:25:32 [Info]  - Stopping streaming
11:25:32 [Info] /home/ap/git/librealsense/src/linux/backend-v4l2.cpp:954 - V4L stream is closed
11:25:33 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
11:25:33 [Error] /home/ap/git/librealsense/common/notifications.cpp:514 - REC error
```
</details>

MartyG-RealSense, 2021-05-27:
It looks as though librealsense was built from source code with the kernel patching method (see the link below).  Is that correct, please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

If you did build using this method and there is a conflict with the Linux kernel, then building librealsense with the **RSUSB backend** method that bypasses the kernel may provide better results.

aPonza, 2021-05-27:
Yes it is. I'll uninstall the current build and retry with the flag `FORCE_RSUSB_BACKEND`, as I understand it.

MartyG-RealSense, 2021-05-27:
Yes.  Remember that you do not need to patch the kernel afterwards with the RSUSB method.  A list of instructions for a basic RSUSB installation without installing wrappers such as Python is described in the link below.

https://github.com/IntelRealSense/librealsense/issues/6432#issuecomment-631484711

aPonza, 2021-05-27:
Nada de nada, it's not working. The new log is attached:

<details>
  <summary> realsense-viewer log </summary>

```
12:06:48 [Info]  - librealsense version: 2.45.0

12:06:48 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Framebuffer size changed to 1855 x 1056
12:06:48 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Window size changed to 1855 x 1056
12:06:48 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Scale Factor is now 1
12:06:48 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:48 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:48 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:48 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Info] /home/ap/git/librealsense/src/context.cpp:382 - Found 1 RealSense devices (mask 0xfe)
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:49 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:50 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:50 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:50 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:50 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:50 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
12:06:50 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 300, error: No data available, number: 3d
[... cut ...]
12:07:14 [Info] /home/ap/git/librealsense/src/frame-archive.h:215 - The user was holding on to 2 frames after stream 0x0x6f96480 stopped
12:07:14 [Info] /home/ap/git/librealsense/src/frame-archive.h:215 - The user was holding on to 2 frames after stream 0x0x6f96480 stopped
12:07:15 [Info] /home/ap/git/librealsense/src/types.h:1104 - Depth stream start failure
```

</details>

Of course the "Physical Port" was `/sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video0` and now is `2-4-2`.

MartyG-RealSense, 2021-05-27:
Let's try a firmware update in the RealSense Viewer by going to the **More** option near the top of its options side-panel and selecting **Install Recommended Firmware** from its menu.

https://dev.intelrealsense.com/docs/firmware-update-tool

[image]

**Install Recommended Firmware** will install the recommended firmware for the particular RealSense SDK version that you have installed.

aPonza, 2021-05-27:
I don't have that option in my realsense-viewer, maybe because I'm on 2.45.0 and you're on 2.41.0? On the other hand, [this table](https://dev.intelrealsense.com/docs/firmware-releases) suggests I'm on the correct firmware version. I still tried to update from `rs-fw-update` with the downloaded binary from that same website, thinking it might've updated incorrectly last time. Nothing seems to have changed.

<details>
  <summary>FW update log</summary>

```
$ rs-fw-update -s 825312072645 -f Signed_Image_UVC_5_12_13_50.bin 

search for device with serial number: 825312072645

updating device: 
Name: Intel RealSense D435, serial number: 825312072645, update serial number: 822213023003, firmware version: 05.12.13.50, USB type: 3.2

firmware update started

firmware update progress: 100[%]

firmware update done

device 825312072645 successfully updated to FW: 05.12.13.50
```

</details>

MartyG-RealSense, 2021-05-27:
The image was an old one grabbed from a previous case to illustrate the option.  It is unusual if you do not have a **More** option in your side-panel as it was introduced in the Viewer a long time ago.   Would it be possible to provide an image of your Viewer window please?

aPonza, 2021-05-27:
There was definitely a tiny misunderstanding: I just don't have "Install Recommended Firmware", and instead I have "Check for Updates" (which btw returns `14:25:11 [Info]  - No online SW / FW updates available`

[image]

MartyG-RealSense, 2021-05-27:
I have the same menu for **More** in 2.45.0.  According to the 2.45.0 release notes, the **Install recommended firmware** option in the menu has been replaced with **Check for Updates**.  

If you left click on this new option, which checks online for updates, does anything happen?

aPonza, 2021-05-27:
> which btw returns 14:25:11 [Info] - No online SW / FW updates available

MartyG-RealSense, 2021-05-27:
Okay, thanks very much for checking.

Next, could you try running the **rs-capture** example program please to test whether it can stream depth.  you should be able to run it by inputting **rs-capture** into the Ubuntu terminal.

aPonza, 2021-05-27:
Thank you for guiding me through the process, I'm here thinking about other things I could do but can't think of anything much. Unfortunately rs-capture yields a black screen, and after some time (15000ms most likely) the terminal shows:

```
$ rs-capture 
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x21ad1b0):
    Frame didn't arrive within 15000
```

MartyG-RealSense, 2021-05-27:
At this point I will seek advice from Intel about your case.  Thanks for your patience!

MartyG-RealSense, 2021-05-27:
Hi @aPonza  I discussed your case with Intel and they provided advice.  

Can you recall any details of the configuration that you were using before the update that caused depth to stop working (librealsense version, firmware version, the method by which you installed llibrealsense when it was working, etc).  If you can, could you recreate that previous configuration please to test the camera with it and see whether depth is restored.  Thanks!

aPonza, 2021-05-28:
I have another D435 which sports firmware version 05.11.06.250 and actually streams depth at this very moment (i.e. with RSUSB configuration, but I could try going back to the old configuration to see if that was part of the issue). I suppose I could try on Monday to downgrade the firmware in the non-working camera to see if that fixes it, is that what you want me to try? I'd also go back to V4L to have the same configuration as before, not sure about the librealsense version though.

aPonza, 2021-05-28:
RSUSB seems to have introduced this warning I showed you in https://github.com/IntelRealSense/realsense-ros/issues/1895#issuecomment-849509760 about `10:56:30 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: No data available, number: 61`.

MartyG-RealSense, 2021-05-28:
If the **control_transfer returned** warning generates a small number of times during launch and stops, or at a rate of once or twice per minute after launch, then it can be safely ignored.  If it is generating continuously though then it can indcate that there is a serious communication problem with the device.

aPonza, 2021-05-31:
<details>
  <summary>:( it doesn't work, this is the log</summary>

```
14:33:43 [Info]  - librealsense version: 2.45.0

14:33:43 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Framebuffer size changed to 1855 x 1056
14:33:43 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Window size changed to 1855 x 1056
14:33:43 [Info] /home/ap/git/librealsense/src/rs.cpp:2610 - Scale Factor is now 1
14:33:43 [Info] /home/ap/git/librealsense/src/context.cpp:382 - Found 1 RealSense devices (mask 0xfe)
14:33:43 [Info]  - Intel RealSense D435 was selected as a default device
14:33:44 [Info]  - No online SW / FW updates available
14:33:44 [Info] /home/ap/git/librealsense/common/notifications.cpp:541 - Intel RealSense D435      (S/N 825312072645)
Current Version: 05.11.06.250
Recommended Version: 5.12.13.50
14:33:47 [Info]  - Starting streaming of Color...
14:33:47 [Info] /home/ap/git/librealsense/src/sensor.cpp:1416 - Request: RGB8 Color, 
Resolved to: YUYV Color, 
14:33:47 [Info] /home/ap/git/librealsense/src/uvc/uvc-streamer.cpp:28 - endpoint 84 read buffer size: 615424
14:33:47 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
14:33:47 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
14:33:52 [Info]  - Starting streaming of Depth...
14:33:52 [Info] /home/ap/git/librealsense/src/sensor.cpp:1416 - Request: Z16 Depth, 
Resolved to: Z16 Depth, 
14:33:52 [Info] /home/ap/git/librealsense/src/uvc/uvc-streamer.cpp:28 - endpoint 130 read buffer size: 815104
14:33:53 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:33:53 [Error] /home/ap/git/librealsense/common/notifications.cpp:514 - REC error
14:33:53 [Warn] /home/ap/git/librealsense/src/libusb/messenger-libusb.cpp:42 - control_transfer returned error, index: 768, error: No data available, number: 61
14:33:54 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:33:55 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:33:56 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:33:57 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:33:58 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:33:59 [Info] /home/ap/git/librealsense/src/types.h:1104 - Depth stream start failure
14:33:59 [Error] /home/ap/git/librealsense/common/notifications.cpp:514 - Depth stream start failure
14:33:59 [Info]  - Stopping streaming
14:34:00 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:34:01 [Info] /home/ap/git/librealsense/src/types.h:1104 - REC error
14:34:02 [Info] /home/ap/git/librealsense/src/types.h:1104 - Depth stream start failure
14:34:06 [Info]  - Stopping streaming
14:34:06 [Info] /home/ap/git/librealsense/src/frame-archive.h:215 - The user was holding on to 1 frames after stream 0x0x58d5c60 stopped
14:34:06 [Info] /home/ap/git/librealsense/src/frame-archive.h:215 - The user was holding on to 1 frames after stream 0x0x58d5c60 stopped
14:34:06 [Info] /home/ap/git/librealsense/src/frame-archive.h:215 - The user was holding on to 1 frames after stream 0x0x58d5c60 stopped
14:34:06 [Info] /home/ap/git/librealsense/src/frame-archive.h:215 - The user was holding on to 1 frames after stream 0x0x58d5c60 stopped
```

</details>

The only other difference would be RSUSB but then again the one at home works perfectly fine with the old firmware.

MartyG-RealSense, 2021-05-31:
Let's test the possibility that the default resolution settings are too much for this particular computer's USB system to handle by setting a low custom stream configuration that publishes depth only.

roslaunch realsense2_camera rs_camera.launch depth_width:=480 depth_height:=270 depth_fps:=30

aPonza, 2021-06-03:
:rocket: Yep, that seems to work in RViz, and testing again with realsense-viewer it also works up to 640x480. Weird, since the previous configuration used to work with that firmware. Anyways, it's most definitely a firmware issue then, isn't it? Thank you Marty, and thank you Intel! If I can help you further please ask away, otherwise this can be closed for me.

MartyG-RealSense, 2021-06-03:
You are very welcome @aPonza - thanks for the update!   As suggested, I will close the case as you are satisifed with the outcome.  Thanks again!

closed 2021-06-03 by MartyG-RealSense