realsenseai/realsense-ros #1902: Camera RealSense L515 get delayed timestamps.

PatrickVibild, 2021-05-27:
Hi all.

I have notice that after running the cameras for a while the camera topics generated by RealSense have its timestamp delayed.

I notice this issue after my Occupancy Map Updated throws me the error that my robotic frame and deepth frame times are to distant.

```
[ERROR] [1622124748.768053278]: Transform error: Lookup would require extrapolation into the past.  Requested time 1622120432.348351717 but the earliest data is at time 1622124738.770423889, when looking up transform from frame [edo_link_6] to frame [camera_side_depth_optical_frame]
```

I echoed my robot states and the depth image published from the RealSense Camera. And i get a 15s shift. (Left terminal shows camera_side/depth/image_rect_raw/header topic and right shows my robot_states/header topic)
[image]

I notice that the camera node will add some delay to the generated timestamps once they are executed for a while. In my system I have two Realsense L515 cameras, and when I run a fresh camera I can see that the timestamps correlates with the time of my machine.

[image]

In case that my launch files can help anyhow, here is the launch file I use to launch the cameras. I saw in previous issues that initial_reset might help with this issue, but in this case i have the same error. And after 2-3 min I can already notice that my generated octomap have certain delay in the system.

```
<launch>
    <include file="$(find main)/launch/side_camera_pose.launch"/>
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="camera" value="camera_side"/>
        <arg name="serial_no" value="f0191108"/>
        <arg name="filters" value="spatial,pointcloud"/>
        <arg name="initial_reset" value="true"/>
    </include>
</launch>
```

doronhi, 2021-05-30:
Could you try turning Global Timestamp off?

PatrickVibild, 2021-05-30:
> Could you try turning Global Timestamp off?

I was looking into the documentation and I did not found any parameters to set it off.

How do turn the global timestamp when launching the realsense nodelet?

doronhi, 2021-06-01:
To set global timestamp to the different modules see my response [here](https://github.com/IntelRealSense/realsense-ros/issues/1607#issuecomment-760128960)

PatrickVibild, 2021-06-01:
I disabled the global times

```
<launch>
    <include file="$(find main)/launch/top_camera_pose.launch"/>
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="camera" value="camera_top"/>
        <arg name="serial_no" value="f0190391"/>
        <arg name="filters" value="spatial,pointcloud"/>
        <arg name="initial_reset" value="true"/>
        <arg name="align_depth" value="true"/>
    </include>
    <rosparam>
        /camera_top/realsense2_camera/global_time_enabled: false
        /camera_top/realsense2_camera/l500_depth_sensor/global_time_enabled: false
        /camera_top/realsense2_camera/rgb_camera/global_time_enabled: false
    </rosparam>
    <node name="camera_node" pkg="camera_node_v4" type="camera_node.py" respawn="true"/>
</launch>
```

Output of the nodelet when I execute it.

```
started roslaunch server http://ros-Precision-5820-Tower.local:41889/

SUMMARY
========

PARAMETERS
 * /camera_side/realsense2_camera/accel_fps: -1
 * /camera_side/realsense2_camera/accel_frame_id: camera_side_accel...
 * /camera_side/realsense2_camera/accel_optical_frame_id: camera_side_accel...
 * /camera_side/realsense2_camera/align_depth: False
 * /camera_side/realsense2_camera/aligned_depth_to_color_frame_id: camera_side_align...
 * /camera_side/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_side_align...
 * /camera_side/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_side_align...
 * /camera_side/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_side_align...
 * /camera_side/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_side_align...
 * /camera_side/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_side_align...
 * /camera_side/realsense2_camera/allow_no_texture_points: False
 * /camera_side/realsense2_camera/base_frame_id: camera_side_link
 * /camera_side/realsense2_camera/calib_odom_file: 
 * /camera_side/realsense2_camera/clip_distance: -2.0
 * /camera_side/realsense2_camera/color_fps: -1
 * /camera_side/realsense2_camera/color_frame_id: camera_side_color...
 * /camera_side/realsense2_camera/color_height: -1
 * /camera_side/realsense2_camera/color_optical_frame_id: camera_side_color...
 * /camera_side/realsense2_camera/color_width: -1
 * /camera_side/realsense2_camera/confidence_fps: -1
 * /camera_side/realsense2_camera/confidence_height: -1
 * /camera_side/realsense2_camera/confidence_width: -1
 * /camera_side/realsense2_camera/depth_fps: -1
 * /camera_side/realsense2_camera/depth_frame_id: camera_side_depth...
 * /camera_side/realsense2_camera/depth_height: -1
 * /camera_side/realsense2_camera/depth_optical_frame_id: camera_side_depth...
 * /camera_side/realsense2_camera/depth_width: -1
 * /camera_side/realsense2_camera/device_type: 
 * /camera_side/realsense2_camera/enable_accel: False
 * /camera_side/realsense2_camera/enable_color: True
 * /camera_side/realsense2_camera/enable_confidence: True
 * /camera_side/realsense2_camera/enable_depth: True
 * /camera_side/realsense2_camera/enable_fisheye1: False
 * /camera_side/realsense2_camera/enable_fisheye2: False
 * /camera_side/realsense2_camera/enable_fisheye: False
 * /camera_side/realsense2_camera/enable_gyro: False
 * /camera_side/realsense2_camera/enable_infra1: False
 * /camera_side/realsense2_camera/enable_infra2: False
 * /camera_side/realsense2_camera/enable_infra: False
 * /camera_side/realsense2_camera/enable_pointcloud: False
[... 27 lines cut ...]
 * /camera_side/realsense2_camera/l500_depth_sensor/global_time_enabled: False
 * /camera_side/realsense2_camera/linear_accel_cov: 0.01
 * /camera_side/realsense2_camera/odom_frame_id: camera_side_odom_...
 * /camera_side/realsense2_camera/ordered_pc: False
 * /camera_side/realsense2_camera/pointcloud_texture_index: 0
 * /camera_side/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera_side/realsense2_camera/pose_frame_id: camera_side_pose_...
 * /camera_side/realsense2_camera/pose_optical_frame_id: camera_side_pose_...
 * /camera_side/realsense2_camera/publish_odom_tf: True
 * /camera_side/realsense2_camera/publish_tf: True
 * /camera_side/realsense2_camera/rgb_camera/global_time_enabled: False
 * /camera_side/realsense2_camera/rosbag_filename: 
 * /camera_side/realsense2_camera/serial_no: f0191108
 * /camera_side/realsense2_camera/tf_publish_rate: 0.0
 * /camera_side/realsense2_camera/topic_odom_in: odom_in
 * /camera_side/realsense2_camera/unite_imu_method: 
 * /camera_side/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.10

```

I can see that it takes around 6000 sequences to start have 1s delayed on the realsense node. Is something to do with the timestamps, when I check the depth image without any processing in RVIZ I can see that there is no delay and i actually see a realtime output of the camera. 

Is there anyhting else I can check on this?

doronhi, 2021-06-03:
I am sorry, I think I accidentaly mislead you. Global timestamp is actually off by default with the L515.
Indeed, it seems that turning global-timestamp off did not change anything. Try turning it on.
You can check the state of the flag using `rosrun rqt_reconfigure rqt_reconfigure`
What you experience seems as if the inner clock of the camera is a bit slower than the system's clock. If works correctly, this is one of the things the Global-timestamp mechanism should help with.

PatrickVibild, 2021-06-05:
Thanks dorohi,

I run the system for around 10 min and the camera was still on sync.

closed 2021-06-05 by PatrickVibild

bigbellmercy, 2021-06-28:
Hi @PatrickVibild, congrats for the resolution. I have the same problem. But it is not resolved for me when I used your launch file. Do you have some suggestion to resolve it?

bigbellmercy, 2021-06-29:
Hi, I am happy that I found the solution to resolve the time delay increasing problem of L515.

In the launch file of L515, after the parameters `global_time_enabled` are enabled as following, the problem disappeared:
```
    <rosparam>
        /camera1/l500_depth_sensor/global_time_enabled: true
        /camera1/motion_module/global_time_enabled: true
        /camera1/rgb_camera/global_time_enabled: true
    </rosparam>
```
, where `camera1` should be matched to yours.
For reference, this script is a little different to the above post(it didn't work for me).