realsenseai/realsense-ros #1903: Incredibly slow with librealsense 2.45, realsense_ros 2.3.0

bibbygoodwin, 2021-05-28:
Hi,

As this is my first realsense installation, I have no frame of reference, but launching `rs_rgbd.launch` or `rs_camera.launch` cause ~140% CPU usage by nodelet, and present frames at ~2fps in RViz. The entire computer (8-core 2.9GHz, 16GB ram) becomes almost unusable.

I'm on librealsense 2.45, realsense_ros 2.3.0. I installed with:
```
apt-get install -y librealsense2-dkms
apt-get install -y librealsense2-utils
apt-get install -y librealsense2-dev
apt-get install -y librealsense2-dbg
apt-get install -y ros-melodic-realsense2-camera
apt install -y ros-melodic-rgbd-launch
```

This seems to be a problem reported by others e.g. at #1887

MartyG-RealSense, 2021-05-28:
Hi @bibbygoodwin   Let's focus on the RealSense Viewer performance first.  Could you try disabling the Viewer's two **GLSL** options in its **Settings** interface please using the instructions in the link below to see whether doing so speeds up your computer.

https://github.com/IntelRealSense/librealsense/issues/8110#issuecomment-754705023

hardesh, 2021-06-02:
Hello,
Facing the same issue on Jetson Nano. The nodelet is taking around 30%-40% CPU now. Earlier it used to take 12%-13% CPU

MartyG-RealSense, 2021-06-02:
Hi @hardesh   Do you also experience a CPU usage increase in the RealSense Viewer like @bibbygoodwin does, please?

hardesh, 2021-06-02:
Hi @MartyG-RealSense,
I didn't check the CPU usage with realsense viewer. Will get back to you in a week
For now I've reverted back to librealsense v2.42 and realsense ros v2.2.22

MartyG-RealSense, 2021-06-02:
Thanks very much for the information @hardesh

MartyG-RealSense, 2021-06-14:
Hi @hardesh  Do you have an update about this case that you can provide, please?  Thanks!

hardesh, 2021-06-18:
Hi @MartyG-RealSense,
Sorry, haven't had a chance to try out the latest versions again. I Will let you know once I get the chance

MartyG-RealSense, 2021-06-18:
No problem at all @hardesh - I will keep this case open for a further time period.  Thanks for the update!

Giggus, 2021-08-13:
Hi, 

I have a similar issue here.

My SW setup is: 
- Ubuntu 20.04 
- kernel 5.8.0-63-generic
- ROS noetic
- librealsense2 2.48.0 
- ros-noetic-realsense2-camera (2.3.1)

My hardware is:
- MSI gaming laptop (MS-17F5)
- RAM : 15 GB 
- Intel® Core™ i7-10750H CPU @ 2.60GHz × 12 

My issue:
Launching "roslaunch realsense2_camera rs_rgbd.launch" I get **CPU 100%** just subscribing both image and depth topics on rviz. If I also publish the point cloud the CPU jumps to 105% for realsense2_camera_manager and 120% for rviz.

I am wondering if this is normal or if I am doing something wrong here.

I also tried (with no improvement):
- ROS melodic In a docker environment 
- I switched all CPU governors to "performance"
- Installed librealsense2 2.49.0 

Any suggestion would be very much appreciated. 

Cheers!

MartyG-RealSense, 2021-08-13:
Hi @Giggus  Could you try the alternative launch instruction below please for rs_camera.launch, which should generate an **ordered** pointcloud like rs_rgbd does.

**roslaunch realsense2_camera rs_camera.launch filters:=colorizer,pointcloud align_depth:=true pointcloud_texture_stream:=RS2_STREAM_COLOR initial_reset:=true ordered_pc:=true**

Giggus, 2021-08-13:
Hi @MartyG-RealSense thanks for the quick reply. 

By running your command I get CPU usage up to 140% for realsense2_camera_manager and 130% for rviz
[image]

MartyG-RealSense, 2021-08-13:
Use of the L515 model should not make a difference with ROS.  A standard first step though in diagnosing problems related to L515 is to unplug the USB cable from the base of the camera (not from the computer end), turn it around in the opposite orientation and re-insert it (USB-C cables are two-way insertion at the micro-sized end).  Is CPU usage the same afterwards?

Giggus, 2021-08-13:
If I plugged it reversed I can only get very low resolution streams - so I always plug in the orientation that gives me access to high resolution rgb and depth streams. 

Any other hints on what could it be causing this high CPU usage?

MartyG-RealSense, 2021-08-13:
Okay, you had the micro-sized end in the appropriate orientation already then if you could only access 320x240 when rotating the connector.  Thanks very much.

Does the L515's outer casing feel hot to the touch after the camera has been running for a few minutes?

Giggus, 2021-08-13:
Yes it does why?

MartyG-RealSense, 2021-08-13:
Rapid overheating can indicate that there is an electrical problem with the USB system that is heating up the camera.  This may be caused by a bad USB cable or a glitch on the USB port that the camera is plugged into.

Giggus, 2021-08-13:
what I observe is sometimes the full camera just stopping to work and then come back live. 

Do you think the heating is related to the high CPU usage?

MartyG-RealSense, 2021-08-13:
It is certainly abnormal that the camera casing is hot to the touch after only a few minutes of streaming.  Could you launch the camera in the RealSense Viewer, enable the depth stream and observe the temperatures (shown as yellow bars) in the **Controls** section of the 'L500 Depth Sensor' side-panel options and see what the temperature is after a few minutes?

[image]

Giggus, 2021-08-13:
After 10 minutes the temperature is 
[image]

MartyG-RealSense, 2021-08-13:
Thank you.  The internal temperatures are on the high side but not extremely so.  Do you have an alternative USB 3.2 port that you can try the camera in on the laptop?  My understanding is that if there is an electrical overheating issue on a port / cable then it affects devices with a high power draw rather than simple devices such as keyboard and mouse.

Giggus, 2021-08-13:
I already connected the device to all usb 3.2 ports and nothings changes

MartyG-RealSense, 2021-08-13:
I located a case in which similarly high processor usage such as 113% was occuring on a gaming laptop, whilst usage on a different desktop PC was less than 30%.  Curiously, usage was 41% when simply reading the topic with rostopic echo, but shot up to 113% when **RViz** was being used.

https://github.com/IntelRealSense/librealsense/issues/1130#issuecomment-484666530

Giggus, 2021-08-13:
Thanks for the link.. It seems to be exactly the same issue I have. 
No solution or temporary hack has been found yet? :(

MartyG-RealSense, 2021-08-13:
You could test whether setting RViz to use **compressed** data instead of raw makes a difference to the CPU usage.

[image]

Giggus, 2021-08-16:
Hi @MartyG-RealSense, after installing `image_transport_plugins` the CPU goes down to 100% for realsense_manager and to 30% for rviz (which is already good news).
[image]

If I subscribe to the pointcloud the CPU jumps back to crazy values
[image]

Might it be because a transport package for the pointcloud is required?

MartyG-RealSense, 2021-08-16:
Hi @Giggus  It's excellent to hear that you made positive progress.

**image_transport** is associated with transmission of compressed data and is not required by default for pointcloud generation.

Giggus, 2021-08-16:
I made some progress but using pointcloud is still not possible with such CPU consumption. I am trying to understand where such high CPU usage comes from..

MartyG-RealSense, 2021-08-16:
A RealSense ROS user in the link below had similarly high CPU usage with pointcloud and reduced it significantly by applying a Decimation Filter, at the cost of a reduction in resolution.

https://github.com/IntelRealSense/realsense-ros/issues/1964

Giggus, 2021-08-16:
Thanks @MartyG-RealSense. I tried but didn't see any difference. Do you think that decimation filter does actually apply also to the L515 camera (since depth is obtained by the embedded Lidar rather than from stereo cameras)?

MartyG-RealSense, 2021-08-16:
Although the Decimation filter does not appear in the list of depth post-processing filters in my RealSense Viewer, i did confirm that the **rs-post-processing** C++ SDK program could access and control decimation of the depth image with an L515.  The images below show an L515 image with the Decimation filter's **Filter Magnitude** setting set to '2' (upper image) and '4' (lower image).

[image]

[image]

Giggus, 2021-08-17:
Hi @MartyG-RealSense , thanks for the reply!!
I tried to add the decimation filter but I didn't see any change to my pointcloud. 
I am launching the nodelets with rs_rgbd.launch, might this be the issue?

For the moment the only solution I found to reduce the CPU usage is to reduce the color_fps to 6 (I can't decrease the depth_fps with the L515 apparently). 

Cheers!

MartyG-RealSense, 2021-08-17:
Thanks very much @Giggus - yes, depth and infrared FPS is fixed at 30 on the L515.

MartyG-RealSense, 2021-08-23:
Hi @Giggus  Do you require further assistance with your question above, please?  Thanks!

Giggus, 2021-08-23:
Hi @MartyG-RealSense, thanks for your assistance. 

For the moment I "solved" the problem by reducing the color frame rate to 6 fps (consequently the point cloud rate reduces as well). This result is not optimal since the same script loads the CPU a third on a different PC. So I do think there is still a problem (maybe PC related). But at least with current configs I can keep working with some CPU resources still available. 

Cheers!

MartyG-RealSense, 2021-08-23:
You are very welcome, @Giggus - if you are satisifed with the outcome of this case, please feel free to use the **Close issue** button under the comment writing box to close the case.  Thanks again!

MartyG-RealSense, 2021-08-23:
@hardesh   You also had problems on this case - is your problem now resolved, please?  Thanks!

hardesh, 2021-08-23:
Hi @MartyG-RealSense, sorry i haven't tried installing librealsense 2.45. v2.42 with realsense-ros 2.2.22 is giving us good results, so we're sticking to that

MartyG-RealSense, 2021-08-23:
Okay, as long as you are satisfied with your current setup with librealsense 2.42.0 and wrapper 2.2.22 then that's totally fine.

MartyG-RealSense, 2021-08-23:
As everyone is satisifed with their outcomes, I will close the case.  You are welcome to add further comments after closure if you have additional enquiries abut the topics covered in this dscussion.  Thanks again!

closed 2021-08-23 by MartyG-RealSense