realsenseai/realsense-ros #1907: Control Transfer Error returned  error. roor: No data available

JeongeunJPark, 2021-06-01:
[image]

Hello, I setup realsense 2.0 sdk with ros-melodic(desktop-full), ubuntu 18.04.
When we try to start intel d435i using command  **roslaunch realsense2_camera rs_camera.launch** , 
We were stucked with the problem.

I tried to some methods to solve this problem for days, but they did not work.
Could you let me know how to solve this problems?

-- 
what i tried to before are...
1. roslaunch realsense2_camera rs_camera.launch enable_sync:=true (doesn't work)
2. roslaunch realsense2_camera rs_camera.launch initial_reset:=true (doesn't work)
3. I thought the problem was related with hardware(usb port), I checked my coumputer's usb port(USB 3.1)
4. I changed camera resolution 1280*720 to 840*480(doesn't work) 
5. I found there is a another problem to figure out Frame error, 
[image]
-> So I fixed this Error, but the one occured before is not being fixed.

Camera and Computer Environment is here.
What can I do for solving this problem?
[image]
[image]
[image]

MartyG-RealSense, 2021-06-01:
Hi @JeongeunJPark  Could you provide the launch log that is generated after launching with **roslaunch realsense2_camera rs_camera.launch** please?  Thanks!

JeongeunJPark, 2021-06-01:
Hello, @MartyG-RealSense ! 
This is the log. 

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ssac23-ROG-Strix-G712LW-G712LW:36667/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 65 lines cut ...]
[ INFO] [1622534082.662049331]: RGB Camera was found.
[ INFO] [1622534082.662241091]: Motion Module was found.
[ INFO] [1622534082.662269978]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1622534082.662294137]: num_filters: 0
[ INFO] [1622534082.662307702]: Setting Dynamic reconfig parameters.
[ WARN] [1622534083.023161176]: Param '/camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1622534083.034182055]: Param '/camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ INFO] [1622534083.057338442]: Done Setting Dynamic reconfig parameters.
[ INFO] [1622534083.057707873]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1622534083.058589886]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1622534083.058609106]: setupPublishers...
[ INFO] [1622534083.060176621]: Expected frequency for depth = 30.00000
[ INFO] [1622534083.081187574]: Expected frequency for color = 30.00000
[ INFO] [1622534083.092456434]: setupStreams...
[ INFO] [1622534083.103254275]: insert Depth to Stereo Module
[ INFO] [1622534083.103288612]: insert Color to RGB Camera
[ INFO] [1622534083.150563448]: SELECTED BASE:Depth, 0
[ INFO] [1622534083.156357344]: RealSense Node Is Up!
[ WARN] [1622534083.279919963]: 
 01/06 16:54:43,280 WARNING [140042970326784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/06 16:54:43,330 WARNING [140042970326784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/06 16:54:43,381 WARNING [140042970326784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 01/06 16:54:43,431 WARNING [140042970326784] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

As you can see, the last error occurs repeatedly.

MartyG-RealSense, 2021-06-01:
If the **control_transfer returned** warning is generating continuously without stopping then this can indicate a serious communication problem with the device.

Let's first test whether the USB system of your computer is being over-burdened by launching with depth only at a low resolution using a custom camera configuration.  Please try inputting the roslaunch instruction below to see what effect, if any, it has on the number of control_transfer returned warnings.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30**

JeongeunJPark, 2021-06-01:
Unfortunately, I used this command **roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30**, then, nothing is changed.

What can I do for check hardware section(usb connection) or software installation? 
I checked my usb connection using **lsusb** command.

MartyG-RealSense, 2021-06-01:
The launch log can also provide an indication of the USB connection status.  This would not indicate how the stream would perform during launch though - it would just inform about which type of connection (USB 2 or 3) the camera is being detected as being on.

[image]

Does the camera perform normally in the RealSense Viewer?

JeongeunJPark, 2021-06-02:
Hello, @MartyG-RealSense ! 

Oh, the log was just viewing what kind of connection is done... I understand it.

Yes, I usually use RealSense Viewer to confirm our camera's data.
What I really want is just take the frame data(RGBD) from the Camera(Intel d435) and make our autonomous driving robot can move depending on the data's value.
So I will use the camera's data with ros-melodic, qt creator, etc....

I am very sorry to bother you. 
I could not tell you exactly what I stucked wtih my poor english skills!  

Then What would i do to make better this error?

MartyG-RealSense, 2021-06-02:
It is no bother at all, I assure you.  And your English is very good.   :)

In RViz, what have you currently got **Fixed Frame** set to?  The RViz image that you kindly provided earlier suggests that it may have been set to **map**.

Could you try setting it to **camera_link** if you have not done this already, please?  Does setting it to **camera_link** improve the results?

[image]

MartyG-RealSense, 2021-06-14:
Case closed due to no further comments received.

closed 2021-06-14 by MartyG-RealSense

JeongeunJPark, 2021-06-15:
Hello,
MartyG-RealSense!

How have you been?
I am so sorry to do not reply soon.

Our team urgently supported other tasks, so We couldn't check the realsense
camera data.
Fortunately, We solve the problem with your suggestion and We can check our
depth data and RGB data using Intel d435i camera!

Thank you for your help.
I hope you have a great day.

Sincerely,
Jeongeun Park.

MartyG-RealSense, 2021-06-15:
It is no trouble at all @JeongeunJPark - it's excellent to hear that your problem was solved.  Thanks very much for the update!

And I am fine, thank you.  :)