realsenseai/realsense-ros #1908: Depth image Noise

jacklu333333, 2021-06-01:
Hi, I'm using realsense l515 module for my robot, and I was hoping it can provide robust mapping results with the rtabmap. However, I am experienced high noise with the depth images.

I was wondering the following question:
1. What might be the reason in the environment that induces the noise for the l515 depth module?

2. Is there any way to low the noise problem by reconfiguration?

Thanks in advance

jacklu333333, 2021-06-03:
Hi, after serval experiments, I found that it seems the L515 was unable to detect the glass in my environment. Is there any way I can reduce the noise?

RealSenseCustomerSupport, 2021-06-30:
----------------------------------------------

Yes, sometimes it's appearing difficult for L515 to deal with some material, i.e. glass, at some view angles causing less reflection of laser. If that's a static object, you probably need to combine the depth data by viewing the object from multiple angles.

Thanks

closed 2021-07-14 by RealSenseCustomerSupport