realsenseai/realsense-ros #1909: Disable image stream to ROS master

alex-stonebridge, 2021-06-01:
Hello, I have a ROS master PC and a slave PC with d435 camera. I use rs_rgbd.launch fila and I want my camera to publish local camera topics via nodelet. However, for some reason, I can subscribe to these topics on my master PC and see the contents. Most likely it takes a lot of network resources and I would like to prevent it. Is there any way to do that?

MartyG-RealSense, 2021-06-01:
Hi @alex-stonebridge  Have you tried **unsubscribing** from the topics, please?  Links to official ROs documentation with information about doing so can be found here:

**C++**
https://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber(c++)

**Python - unregister()**
http://docs.ros.org/en/electric/api/rospy/html/rospy.topics.Topic-class.html#unregister

The link below also suggests that it is possible to do so from the RViz interface.

https://answers.ros.org/question/354792/rclcpp-how-to-unsubscribe-from-a-topic/?answer=359820#post-id-359820

Alternatively, if you do not need a local slave to be publishing then a method that has been looked at in the past is to use a **mux** node to set up two publishers (one that sends data and one that does not) and switch between the publishers to start / stop data transmission.

https://github.com/IntelRealSense/realsense-ros/issues/1195#issuecomment-626750400

alex-stonebridge, 2021-06-01:
@MartyG-RealSense Thank you for your attention. However, now I am not sure if I understand the ROS architecture correctly. So, if I see slave`s topics on my master PC that doesn`t mean that the data is already being transported? Thus, only when I subscribe to the topic it will start to transport the data?

MartyG-RealSense, 2021-06-01:
Subscription implies to me that the master PC would have to delberately subscribe to the source that is publishing the data in order to retrieve if, rather than the data being forcibly transmitted from the source (the slave PC) to the master without the master's permission.

I will seek the advice of @doronhi the RealSense ROS wrapper developer for confirmation one way or the other.

Hi @doronhi  If a slave computer is publishing data, would a master PC receive that data automatically or would the master have to deliberately subscribe to the publisher on the slave before the data could be transmitted from the slave PC to the master PC?  In other words, if the master does not subscribe to the publisher then the data cannot be transported?  Thanks!

doronhi, 2021-06-03:
The master PC will not receive the data if it is none of the nodes running on it is registered on that topic. This is a general ROS concept. See the answer to [this](https://answers.ros.org/question/37965/how-does-ros-handle-messages-between-3-or-more-nodes-across-2-machines) question.
I hope it answers your question as well.

closed 2021-06-03 by alex-stonebridge