realsenseai/realsense-ros #1910: Change rostopic output image dimensions

greendev32, 2021-06-01:
Hi, is there any way to make the rostopics /camera/color/image_raw and /camera/depth/image_rect_raw publish this data with different image dimensions? Mine was publishing both of these with the size 320x240. I changed USB cables and now it's suddenly publishing with the size 640x360.

I'd like to go back to the dimensions it used to have, because it cannot find the depth correctly anymore when the images are at this size. I noticed that when using the realsense-viewer, it can find the depth when the image is zoomed out, but loses that ability as the dimensions increase when zooming in. I believe that is happening in the ROS implementation too.

Thanks

MartyG-RealSense, 2021-06-02:
Hi @greendev32   Could you confirm for me which RealSense camera model you are using, please?   Thanks!

greendev32, 2021-06-02:
> Hi @greendev32 Could you confirm for me which RealSense camera model you are using, please? Thanks!

I'm using a d435i. Also I meant to mention I was using the rs_camera.launch file to create those topics - if that helps

MartyG-RealSense, 2021-06-02:
Thanks very much for the clarification.  

Could you check the camera name in the options side-panel of the RealSense Viewer to see the number beside the D435i camera name.  If it has the number **2.1** beside it instead of 3.2 then this could indicate that the camera is being detected as being on a USB 2 connection after the USB cable was changed.  This has been known to happen if a USB 2 cable has unknowingly been attached to the camera instead of a USB 3 cable.  Doing so would limit the capabilities and supported resolutions of the camera compared to the full access to its capabilities that it has when detected as being on a USB 3 connection.

greendev32, 2021-06-03:
Thanks for the info! I had tried it with a USB 2 cable before switching to the USB 3 one I'm using now, so I'm guessing that was it. Launching with the arguments you gave helped a lot and it gives me correct values at a distance, further away than 50cm. It still can't match what I'm getting in the realsense viewer however - when closer than 50cm it starts outputting 0cm, while the viewer looking at the same object can go down to about 18cm. I tried to mimic the settings in the viewer but there are so many, I'm not sure which setting(s) are allowing the viewer to output at close range. Do you have any advice on this?

MartyG-RealSense, 2021-06-10:
Sincere apologies for the delay in responding further. 

On the D435i model the default minimum depth sensing distance is 0.1 meters (10 cm), so you should certainly be able to get much closer than 50 cm.  As the ROS wrapper does not have a minimum distance setting (it defines maximum distance with the **clip_distance** setting).

Could you try 848x480 depth resolution instead of 1280x720 please?  My thinking is that as minimum depth distance can be reduced by using a lower resolution, using 1280x720 might be increasing the minimum distance.  On the D435i model, the optimal depth accuracy resolution is 848x480 (it is 1280x720 on the D415 model).

MartyG-RealSense, 2021-06-16:
Hi @greendev32  Do you require further assistance with this case, please?  Thanks!

greendev32, 2021-06-17:
I tried using different resolutions including 848x480, but still wasn't able to get the depth to show up at close distances like it did in the viewer. I ended up saving off the configuration from the viewer and pointing to it in the launch file. That fixed my problem! Thanks for everything.

MartyG-RealSense, 2021-06-17:
Excellent news that you found a solution @greendev - thanks very much for the update!  :)

closed 2021-06-28 by MartyG-RealSense