realsenseai/realsense-ros #1911: Jetson nano Fps of topics and IMU direction  / D435i

Majed-Alsubaie, 2021-06-03:
Hi i have a weird  issue where i set topics to be published with 15 FPS but that does not work and topics published with 2  , 15 and 30 hz

`roslaunch realsense2_camera rs_camera.launch align_depth:=true unite_imu_method:="linear_interpolation" enable_gyro:=true enable_accel:=true color_fps:=15 infra_fps:=15 depth_fps:=15 color_width:=1280 color_height:=720 depth_width:=848 depth_height:=480 initial_reset:=true
... logging to /home/kfupm/.ros/log/53ecfde0-c49d-11eb-a067-0242cc161f6e/roslaunch-kfupm-desktop-29817.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kfupm-desktop:39893/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 848
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 181 lines cut ...]
average rate: 16.772
	min: 0.015s max: 0.433s std dev: 0.12571s window: 50
average rate: 15.021
	min: 0.015s max: 0.435s std dev: 0.13444s window: 60
average rate: 16.010
	min: 0.015s max: 0.448s std dev: 0.12967s window: 75
average rate: 15.746
	min: 0.015s max: 0.500s std dev: 0.13204s window: 93
average rate: 15.287
	min: 0.015s max: 0.500s std dev: 0.13422s window: 107
^C
average rate: 15.603
	min: 0.015s max: 0.500s std dev: 0.13260s window: 110
`

**the main issue is that the /camera/aligned_depth_to_color/image_raw publisheds with low rate 1.8 hz  ** and please notice  /camera/depth/image_rect_raw published with 30 hz .
This issue appears when i use Jetson nano and all work fine when i use my laptop 

The second issue is that the IMU topic give me y-diraction to ground instead of z or neg-z 

`linear_acceleration: 
  x: 0.00724035618121
  y: -8.98433915158
  z: 2.00167468906
linear_acc`

MartyG-RealSense, 2021-06-03:
Hi @Majed-Alsubaie  You have defined **infra_fps* in the roslaunch instruction but not infra_height and infra_width.  The missing height and width definitions would cause the custom configuration to be identified as invalid, and the default configuration of the particular camera model being used would be applied instead.

Also,the infra topics need to be enabled as they are disabled by default.  They can be enabled with **infra1_enabled:=true infra2_enabled:=true**

Please try the roslaunch instruction below.

**roslaunch realsense2_camera rs_camera.launch align_depth:=true enable_gyro:=true enable_accel:=true unite_imu_method:="linear_interpolation" color_width:=1280 color_height:=720 color_fps:=15 infra1_enabled:=true infra2_enabled:=true infra_fps:=15 depth_width:=848 depth_height:=480 depth_fps:=15 initial_reset:=true**

Majed-Alsubaie, 2021-06-04:
Hello @MartyG-RealSense ,
The issue is not related to the Infra topics.
Actually i am not looking for  Infra topics and i include **infra_fps:=15**  by mistake.
Still the **/camera/aligned_depth_to_color/image_raw** published with low rate (2 hz ) where the expected rate 15 hz

Thanks

Majed-Alsubaie, 2021-06-04:
` roslaunch realsense2_camera rs_camera.launch align_depth:=true  color_width:=1280 color_height:=720 color_fps:=15 depth_width:=848 depth_height:=480 depth_fps:=15 initial_reset:=true
`
LED TO  :

**rostopic hz /camera/aligned_depth_to_color/image_raw = 
average rate: 5.147
	min: 0.042s max: 0.636s std dev: 0.04236s window: 691**

**rostopic hz /camera/color/image_raw =
average rate: 13.833
	min: 0.014s max: 0.172s std dev: 0.07426s window: 12

**

**rostopic hz /camera/depth/image_rect_raw =
average rate: 6.693
	min: 0.001s max: 0.221s std dev: 0.07990s window: 59

**

MartyG-RealSense, 2021-06-04:
Alignment is a processing-intensive operation, so the capabilities of your computer / computing device may have a bearing on performance.  

Next, could you try disabling in ROS an option called **Auto-Exposure Priority** please.  If Auto-Exposure is enabled and the RGB option **Auto-Exposure Priority** is disabled then it should force the FPS to be maintained at a constant rate.  For example, if depth and color have both been set to 15 FPS then both streams should be forced to run at 15 FPS instead of the FPs rate varying.

https://github.com/IntelRealSense/realsense-ros/issues/1349#issuecomment-681695259

Majed-Alsubaie, 2021-06-05:
@MartyG-RealSense ,Thanks
I think the problem related to the POWER. because Jetson Nano's USB is not providing enough power .
And I solved it by using Powered-USB hub [https://www.amazon.com/gp/product/B00DQFGH80/ref=ewc_pr_img_4?smid=ATVPDKIKX0DER&th=1](url).
However, i notice sometime the camera instead of taking the power from USB-Hub , it absorbed the power from Jetson Nano  port.
So any idea how to force camera to use USB-HUB's power ?

and yes please can you  comment on the second issue where the IMU topic consider the Y-direction to the ground/down

MartyG-RealSense, 2021-06-05:
I have not heard of a device being able to draw power from a port other than the one that it is connected to.  I researched the subject and it suggested that if a powered hub's mains power is turned off or cut off then a device may be able to continue on the computer power if it is not a high power-drawing device such as an external hard disk drive or recordable dvd drive.  

I do not know of any (easy) method to fix that if the electricity supply to the powered hub is inactive.  Conceivably if there was an issue with the stability of the local mains electricity supply then you could add an Uninterruptible Power Supply (UPS) pack to your computer system that instantly kicks in to provide power when the mains power fails.  It may not be worth the expense though unless a problem with the local power supply is confirmed and it is critically important that power is not drawn from the Jetson.

The Y-accel IMU value is not related to z-distance (depth).  It is a measurement of downward acceleration under the force of gravity.

MartyG-RealSense, 2021-06-11:
Hi @Majed-Alsubaie  Do you require further assistance with this case, please?  Thanks!

Majed-Alsubaie, 2021-06-12:
Hi @MartyG-RealSense ,,Thanks and sorry for responding late.
I did disable  Auto-Exposure Priority and enable Auto-Exposure , still i got  :
`rostopic hz /camera/depth/image_rect_raw
subscribed to [/camera/depth/image_rect_raw]
	min: 0.001s max: 0.528s std dev: 0.12417s window: 141
average rate: 27.254
	min: 0.001s max: 0.528s std dev: 0.12394s window: 171
average rate: 27.381
	min: 0.001s max: 0.528s std dev: 0.12387s window: 201
`
For /camera/color/image_raw , i got it with 15fps but depth not yet

MartyG-RealSense, 2021-06-13:
If you are still using **align_depth:=true** are you able to acheve 15 FPS for both if you test removing align_depth:=true from the roslaunch instruction?

Majed-Alsubaie, 2021-06-13:
@MartyG-RealSense .
i have just tried ,,,Yes i am able to achieve constant 15 FPS when i disable **align_depth:=true **

Majed-Alsubaie, 2021-06-13:
For Alignment , In  Realsense-Viewer ,I Believe GPU is used so it works fine
But In ROS wrapper is not , so  Alignment processing done using CPU with low FPS
How can i make ROS wrapper used GPU ,(Build  it with CUDA) ?
I am using Jetson Nano , **OpenCv 4.5.1  | compile cuda YES**

MartyG-RealSense, 2021-06-13:
**Doronhi** the RealSense ROS wrapper developer comments about CUDA in ROS in the link below.  

https://github.com/IntelRealSense/realsense-ros/issues/1177#issuecomment-674291080

My understanding of that discussion is that Doronhi advises that the RealSense ROS Debian packages do not include CUDA support.  CUDA support can instead be built in librealsense using **-DBUILD_WITH_CUDA=true**, as mentioned in the instructions on Intel's Jetson installation page.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

**********

The link below is the case of a Jetson user who achieved 30 FPS when using align_depth:=true if they disabled auto-exposure and set a manual exposure within a certain range.

https://github.com/IntelRealSense/realsense-ros/issues/1357

Disabling auto-exposure and setting a manual exposure value within a certain range is a known alternative to the FPs enforcemement method of enabled auto-exposure and disabled auto-exposure priority.  The maths of selecting an exposure value are described by a RealSense team member in the link below.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

Majed-Alsubaie, 2021-06-14:
Hi, Thanks @MartyG-RealSense 
i have a good news after succeed  on building ROS wrapper with CUDA. All my issues disappeared
I think All jetson's users should install  ROS wrapper with CUDA.
Thank you again 
+I would  like to suggest  using ENU convention with IMU in ROS wrapper.

closed 2021-06-14 by Majed-Alsubaie

MartyG-RealSense, 2021-06-14:
That is excellent news @Majed-Alsubaie - thanks very much for the update and your suggestions for Jetson users  :)