realsenseai/realsense-ros #1912: L515 configuration failing

RGring, 2021-06-04:
Hi,
I use the L515 with ROS melodic. I use the following launch file to get maximum resolution for depth and rgb data.
`<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="l515"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="1280"/>
  <arg name="fisheye_height"      default="720"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="1024"/>
  <arg name="depth_height"        default="768"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="1280"/>
  <arg name="infra_height"       default="720"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="1920"/>
  <arg name="color_height"        default="1080"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="6"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
      <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>

      <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
      <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
      <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>

      <arg name="depth_width"              value="$(arg depth_width)"/>
      <arg name="depth_height"             value="$(arg depth_height)"/>
      <arg name="enable_depth"             value="$(arg enable_depth)"/>

      <arg name="color_width"              value="$(arg color_width)"/>
      <arg name="color_height"             value="$(arg color_height)"/>
      <arg name="enable_color"             value="$(arg enable_color)"/>

      <arg name="infra_width"              value="$(arg infra_width)"/>
      <arg name="infra_height"             value="$(arg infra_height)"/>
      <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
      <arg name="enable_infra2"            value="$(arg enable_infra2)"/>

      <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>`

Unfortunately, the configuration fails.

I get the following stdout
`
[ INFO] [1622800462.493163647]: Device with serial number f0246262 was found.

[ INFO] [1622800462.493216206]: Device with physical ID 1-2-39 was found.
[ INFO] [1622800462.493260621]: Device with name Intel RealSense L515 was found.
[ INFO] [1622800462.493891647]: Device with port number 1-2 was found.
[ INFO] [1622800462.493933403]: Device USB type: 2.1
[ WARN] [1622800462.493961158]: Device f0246262 is connected using a 2.1 port. Reduced performance is expected.
[ INFO] [1622800462.496327402]: getParameters...
[ INFO] [1622800462.566399232]: setupDevice...
[ INFO] [1622800462.566446837]: JSON file is not provided
[ INFO] [1622800462.566477165]: ROS Node Namespace: l515
[ INFO] [1622800462.566512737]: Device Name: Intel RealSense L515
[ INFO] [1622800462.566558018]: Device Serial No: f0246262
[ INFO] [1622800462.566591887]: Device physical port: 1-2-39
[ INFO] [1622800462.566621727]: Device FW version: 01.05.05.00
[ INFO] [1622800462.566651492]: Device Product ID: 0x0B64
[ INFO] [1622800462.566679242]: Enable PointCloud: Off
[ INFO] [1622800462.566706162]: Align Depth: On
[ INFO] [1622800462.566726764]: Sync Mode: On
[ INFO] [1622800462.566795329]: Device Sensors: 
[ INFO] [1622800462.581948889]: L500 Depth Sensor was found.
[ INFO] [1622800462.586481084]: RGB Camera was found.
[ INFO] [1622800462.586704954]: Motion Module was found.
[ INFO] [1622800462.587185384]: num_filters: 1
[ INFO] [1622800462.587217767]: Setting Dynamic reconfig parameters.
[ INFO] [1622800462.890762268]: Done Setting Dynamic reconfig parameters.
[ WARN] [1622800462.890956379]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: 1024, Height: 768, FPS: 30, Format: Z16
[ WARN] [1622800462.891018674]: Using default profile instead.
[ INFO] [1622800462.891306240]: depth stream is enabled - width: 320, height: 240, fps: 30, Format: Z16
[ WARN] [1622800462.891427744]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 1920, Height: 1080, FPS: 6, Format: None
[ WARN] [1622800462.891474792]: Given stream configuration is not supported by the device!  Stream: Confidence, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: None
[ INFO] [1622800462.891838174]: setupPublishers...
[ INFO] [1622800462.895660053]: Expected frequency for depth = 30.00000
[ INFO] [1622800462.934061382]: setupStreams...
[ INFO] [1622800462.945827959]: insert Depth to L500 Depth Sensor
[ INFO] [1622800462.986689405]: SELECTED BASE:Depth, 0
[ INFO] [1622800462.987343673]: RealSense Node Is Up!
`

Finally only a depth image of resolution 240 x 320 is published. RGB data is missing completely. Furthermore, it says it is using USB port 2.1, although it is attached to USB 3.0.

What is the problem here? I appreciate help.

doronhi, 2021-06-07:
Hi,
The Linux system can sometimes, accidentally, detect a USB 3.1 connection as a USB 2.1 connection if the cable was attached slowly or loosely but usually the issue is caused by a low-quality USB cable.
In any case, the fact that the device is recognized by the system as being connected through a USB 2.1 port can explain the issues with the configuration.

RGring, 2021-06-07:
Indeed, when the connection was USB 3.1, the configuration worked.

closed 2021-06-07 by RGring