realsenseai/realsense-ros #1913: Passing config_file arg at launch results on failing to parse the argument

TSC21, 2021-06-04:
Trying `ros2 launch realsense2_camera rs_launch.py config_file:="'<abs_path_to_file>/d435i.yaml'"` results on:

```sh
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: '<abs_path_to_file>/d435i.yaml'
```

I believe the current logic in `rs_launch.py` using the substitutions with `LaunchConfiguration` is faulty. I tried some different approach but no luck so far. I believe this needs a fix.

TSC21, 2021-06-04:
Found the solution in https://github.com/IntelRealSense/librealsense/issues/8863#issuecomment-823026235. Issuing a PR for it.

MartyG-RealSense, 2021-06-04:
Thanks very much @TSC21 - if you create a PR, please post a link to it here when it is submitted.  Thanks again!

TSC21, 2021-06-05:
> Thanks very much @TSC21 - if you create a PR, please post a link to it here when it is submitted. Thanks again!

@MartyG-RealSense here: https://github.com/IntelRealSense/realsense-ros/pull/1915. Happy to help!

MartyG-RealSense, 2021-06-05:
Great, thanks @TSC21 - I have added an **Enhancement** tag to this case to highlight that this case https://github.com/IntelRealSense/realsense-ros/issues/1913 should be kept open whilst the associated PR is active.  Thanks again!

doronhi, 2021-06-30:
Fixed in #1942

closed 2021-06-30 by doronhi