realsenseai/realsense-ros #1914: D435i Error (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

yaodongC, 2021-06-05:
**I keep getting this error and no RGB and point cloud data in rviz,**
ERROR [546459083136] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!
WARNING [547289534848] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
**realsense-vier give me error:**
Invalid Value in rs2_get_option(options:0x7f88027820, option:Emitter On Off):
hwmon command 0x7d failed. Error type: No data to return (-21).

Setup:
librealsense: V2.31.0
realsense-ros:V2.2.11
NVIDIA Jetson AGX Xavier [16GB]
L4T 32.2.3 [ JetPack 4.2.3 ]
CUDA 10.0.326
OpenCV version: 3.3.1

yaodongC, 2021-06-05:
**Note: latest librealsense and realsense-ros will crash on my machine**
**Note: I cmpile librealsense and realsense-ros from source using scripts from jetsonhacks**

MartyG-RealSense, 2021-06-05:
Hi @yaodongC   Your librealsense SDK and RealSense ROS wrapper versions are correctly matched, and it is my understanding that CUDA 10 is correct for use with Nvidia Jetson.  JetPack 4.2.3 is also verified by Intel for use with Jetson AGX.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

It is possible that your camera firmware version could be an issue though.  If firmware **5.12.13.50** is installed in the camera then it should be used with librealsense SDK **2.39.0** or newer.  This is because internal changes were introduced in the firmware from version 5.12.8.200 onwards.

There are much newer SDK versions available, going up to 2.45.0 at the time of writing this (an SDK that is compatible with firmare 5.12.13.50).  If you need to use SDK 2.31.0 however then I would recommend downgrading the firmware to **5.12.7.100** (the version before the internal changes to the firmware were made).  Alternatively, if you go to the **More** option near the top of the RealSense Viewer's options side-panel and select the **Install Recommended Firmware** option from its menu then it should install the firmware version recommended for SDK 2.31.0 (which is firmware **5.12.1.0**).

yaodongC, 2021-06-05:
Thanks for the advice!
I have upgrade librealsense SDK and realsense-ros to latest version
And I still didi not get data in ROS
See #1916

closed 2021-06-05 by yaodongC