realsenseai/realsense-ros #1916: No data in rostopics (messenger-libusb.cpp:42) control_transfer returned error

yaodongC, 2021-06-05:
**Both realsense-viewer and realsense-ros keep giving this error and no data in published rostopic**
(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

**Note** 
I can see RGB and depth point cloud in realsense-viewer but no data in ROS
librealsense is compiled from source and **realsense-ros installed from apt** 
**realsense-ros latest compiled from source will crash** 

**Setup:**
**Model: D435i**
- firmware version: 05.12.13.50
- USB type: 3.2

**NVIDIA Jetson AGX Xavier [16GB]**
- librealsense: V2.45.0
- realsense-ros:latest
- L4T 32.2.3 [ JetPack 4.2.3 ]
- CUDA 10.0.326
- OpenCV version: 3.3.1

yaodongC, 2021-06-05:
**realsense-ros latest compiled from source will crash with the following error:**
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/yaodong/Desktop/catkin_ws_vsalm/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE

yaodongC, 2021-06-05:
[image]

yaodongC, 2021-06-05:
05/06 11:09:15,754 WARNING [547087290752] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 05/06 11:09:15,805 WARNING [547087290752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

yaodongC, 2021-06-05:
**Launch file setup**
  <arg name="depth_width"         default="-1"/>
  <arg name="depth_height"        default="-1"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="-1"/>
  <arg name="color_height"        default="-1"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="-1"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="-1"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

MartyG-RealSense, 2021-06-05:
Hi @yaodongC  If depth data is being successfully displayed in RViz, could you try changing the **Color Transformer** setting of RViz from "Intensity" to "RGB8" and also set the style to **points**.

yaodongC, 2021-06-06:
The problem is not RVIZ. I have use **rostopic echo**  to listen to rostopics published by realsense, and there are no data published (excepts for 1 topic: camera/depth/image_rect_raw)

yaodongC, 2021-06-06:
rs_alignned_depth.launch (default setting) will also give this error:

(synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

MartyG-RealSense, 2021-06-06:
Thank you for the additional information.  Would it be possible to update your JetPack to a modern version?   SDK 2.45.0 supports Jetpack 4.4.1 and 4.5 (32.4.4 and 32.5)

yaodongC, 2021-06-06:
I cannot. Can you recommend an SDK version for my jetcpack. And [docs ](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md)in realsense-SDK has explicitly stated it was tested on my jectpack version.

MartyG-RealSense, 2021-06-06:
The most recent questions on the librealsense GitHub forum about JetPack 4.2.3 were in mid to late 2019.   SDK **2.31.0** from December 2019 may therefore be an appropriate match to that age of JetPack version. 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.31.0

JetPack 4.2.3 is verified as a minimum for use with the official Intel instructions for installing from source code on Jetson though, so it may be best to try that method first.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend

eyildirir, 2021-06-11:
I can confirm this issue in SDK 2.45.0 on a Jetson Nano with JetPack 4.5.1.

Executing `roslaunch realsense2_camera rs_camera.launch` to launch the node, and then subscribing to both color and depth image_raw nodes, one of the nodes doesn't publish anything if the other one is publishing. 

Note that I don't have a barrel connector for power, it will arrive in roughly 3 days. I will test again then and post the results here.

MartyG-RealSense, 2021-06-11:
Thanks very much for your report @eyildirir - I look forward to your test results with a barrel connector.  Good luck!

eyildirir, 2021-06-14:
@MartyG-RealSense The barrel connector is in. 5V 4A, but the issue still happens when I try to run the D455 camera with the `filters:=pointcloud` parameter. Color feed seems to stop receiving data after a few seconds, cannot recover.

Here are the ROS logs for reference:

[camera-realsense2_camera-3-stdout.log](https://github.com/IntelRealSense/realsense-ros/files/6650683/camera-realsense2_camera-3-stdout.log)
[master.log](https://github.com/IntelRealSense/realsense-ros/files/6650684/master.log)
[roslaunch-slip-cv-14319.log](https://github.com/IntelRealSense/realsense-ros/files/6650685/roslaunch-slip-cv-14319.log)
[rosout.log](https://github.com/IntelRealSense/realsense-ros/files/6650686/rosout.log)
[rosout-1-stdout.log](https://github.com/IntelRealSense/realsense-ros/files/6650687/rosout-1-stdout.log)

MartyG-RealSense, 2021-06-14:
What version of CUDA is your Nano using, please?  It is my understanding that Nano requires CUDA 10, based on what somebody who wanted to use CUDA 9 on Bank was told by an Nvidia forum moderator in the link below.

https://forums.developer.nvidia.com/t/jetson-nano-install-different-version-of-cuda/72475

It is also worth noting that librealsense only supports up to JetPack **4.5** at the time of writing this, whilst you have listed your JetPack version as 4.5.1

https://github.com/IntelRealSense/librealsense/pull/8366

closed 2021-06-26 by MartyG-RealSense