realsenseai/realsense-ros #1920: D435i USB3.2 ROS2 Foxy : Infra sensors not recongnized : (Infrared, 0) sensor isn't supported by current device! -- Skipping...

aagrawa3-dev, 2021-06-07:
Hi All,
I am stuck with this issue while trying to visualize Infra images in RVIZ. When I am launching realsense node, it ends up showing warning that infra sensors not avaliable for the device. But thats not true as I can see infra images in realsense-viewer.

More detail:

Below image showing infra images with realsense-viewer.
 
[image]

But the same does not work when launching realsense node with infra params enabled.
We get this while launching node:

```
$ ros2 launch realsense2_camera rs_launch.py  serial_no:='_0443220*****' align_depth:=true enable_infra1:=true enable_infra2:=true

[INFO] [launch]: All log files can be found below /home/**/.ros/log/2021-06-06-09-58-05-423207-host-367

[INFO] [launch]: Default logging verbosity is set to INFO

[INFO] [realsense2_camera_node-1]: process started with pid [369]

[realsense2_camera_node-1] [INFO] [1622973485.544040372] [RealSenseCameraNode]: RealSense ROS v3.1.4

[realsense2_camera_node-1] [INFO] [1622973485.544081399] [RealSenseCameraNode]: Built with LibRealSense v2.44.0

[realsense2_camera_node-1] [INFO] [1622973485.544087397] [RealSenseCameraNode]: Running with LibRealSense v2.44.0

[realsense2_camera_node-1] [INFO] [1622973485.704050198] [RealSenseCameraNode]: Device with serial number 044322073299 was found.

[realsense2_camera_node-1]

[realsense2_camera_node-1] [INFO] [1622973485.704074545] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video6 was found.

[realsense2_camera_node-1] [INFO] [1622973485.704078924] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.

[realsense2_camera_node-1] [INFO] [1622973485.704228604] [RealSenseCameraNode]: Device with port number 2-3 was found.

[realsense2_camera_node-1] [INFO] [1622973485.704237105] [RealSenseCameraNode]: Device USB type: 3.2

[realsense2_camera_node-1] [INFO] [1622973485.705121592] [RealSenseCameraNode]: getParameters...

[realsense2_camera_node-1] [INFO] [1622973485.705200754] [RealSenseCameraNode]: 707

[realsense2_camera_node-1] [INFO] [1622973485.706370365] [RealSenseCameraNode]: setupDevice...

[realsense2_camera_node-1] [INFO] [1622973485.706380500] [RealSenseCameraNode]: JSON file is not provided

[realsense2_camera_node-1] [INFO] [1622973485.706386534] [RealSenseCameraNode]: Device Name: Intel RealSense D435I

[realsense2_camera_node-1] [INFO] [1622973485.706390805] [RealSenseCameraNode]: Device Serial No: 044322073299

[realsense2_camera_node-1] [INFO] [1622973485.706394997] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video6

[realsense2_camera_node-1] [INFO] [1622973485.706399614] [RealSenseCameraNode]: Device FW version: 05.12.13.50

[realsense2_camera_node-1] [INFO] [1622973485.706403701] [RealSenseCameraNode]: Device Product ID: 0x0B3A

[realsense2_camera_node-1] [INFO] [1622973485.706407484] [RealSenseCameraNode]: Enable PointCloud: Off

[realsense2_camera_node-1] [INFO] [1622973485.706411233] [RealSenseCameraNode]: Align Depth: On

[realsense2_camera_node-1] [INFO] [1622973485.706414907] [RealSenseCameraNode]: Sync Mode: On

[realsense2_camera_node-1] [INFO] [1622973485.706422891] [RealSenseCameraNode]: Device Sensors:

[realsense2_camera_node-1] [INFO] [1622973485.708211741] [RealSenseCameraNode]: Stereo Module was found.

[realsense2_camera_node-1] [INFO] [1622973485.712912351] [RealSenseCameraNode]: RGB Camera was found.

[realsense2_camera_node-1] [INFO] [1622973485.713034245] [RealSenseCameraNode]: Motion Module was found.
[... 369 lines cut ...]
    Color         320x180       @ 60Hz     RGBA8
    Color         320x180       @ 60Hz     BGR8
    Color         320x180       @ 60Hz     YUYV
    Color         320x180       @ 30Hz     RGB8
    Color         320x180       @ 30Hz     Y16
    Color         320x180       @ 30Hz     BGRA8
    Color         320x180       @ 30Hz     RGBA8
    Color         320x180       @ 30Hz     BGR8
    Color         320x180       @ 30Hz     YUYV
    Color         320x180       @ 6Hz      RGB8
    Color         320x180       @ 6Hz      Y16
    Color         320x180       @ 6Hz      BGRA8
    Color         320x180       @ 6Hz      RGBA8
    Color         320x180       @ 6Hz      BGR8
    Color         320x180       @ 6Hz      YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format
    Accel        N/A            @ 250Hz    MOTION_XYZ32F
    Accel        N/A            @ 63Hz     MOTION_XYZ32F
    Gyro         N/A            @ 400Hz    MOTION_XYZ32F
    Gyro         N/A            @ 200Hz    MOTION_XYZ32F

```

MartyG-RealSense, 2021-06-07:
Hi @aagrawa3-dev   As a starting point in investigating your case, it is worth highlighting that you have a mismatch between your librealsense SDK and RealSense ROS wrapper versions.  It is important to match a particular RealSense ROS wrapper version as closely as possible with the librealsense SDK version recommended for it in the wrapper release notes.  

The ROS2 wrapper **3.1.4** that you are using was designed for SDK **2.42.0**.  

There is not a ROS2 wrapper specifically for SDK 2.44.0 though.  In that situation, I would recommend using a wrapper that supports the nearest previous SDK version which is **3.1.5** (designed for SDK 2.43.0).

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.5

In regard to enabling the infra topics on ROS2: another RealSense ROS user experienced this issue in April 2021.  Their solution was to set enable_infra1 and enable_infra2 to '1' instead of 'true'.

https://github.com/IntelRealSense/realsense-ros/issues/1287#issuecomment-814118459

aagrawa3-dev, 2021-06-08:
Hi @MartyG-RealSense 
Thanks for quick reply. I will investigate about sdk and wrapper version mismatch.

About: #1287 (comment) The propsed solution is not working.
With command:
```ros2 launch realsense2_camera rs_launch.py serial_no:='_044322****' enable_infra1:=1 infra_rgb:=true```
it fails:
```
[RealSenseCameraNode]: Failed to set parameter: enable_infra1. expected [bool] got [integer]
```

A snapshot from rs_launch.py (not changed by me):
```
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'infra_width',                  'default': '-1', 'description': 'infra width'},
                           {'name': 'infra_height',                 'default': '-1', 'description': 'infra width'},
                           {'name': 'enable_infra1',                'default': 'true', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'true', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'fisheye_width',                'default': '-1', 'description': 'fisheye width'},
                           {'name': 'fisheye_height',               'default': '-1', 'description': 'fisheye width'},
                           {'name': 'enable_fisheye1',              'default': 'true', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye2',              'default': 'true', 'description': 'enable fisheye2 stream'},
                           {'name': 'confidence_width',             'default': '-1', 'description': 'depth image width'},        
```

MartyG-RealSense, 2021-06-08:
Does it work if you use just **enable_infra:=true** instead of enable_infra1 and enable_infra2 please?

aagrawa3-dev, 2021-06-08:
It does not work. Also somewhere I read that enable_infra is parameter for L515 type cameras. 

About version mismatch topic:
I was able to run the infra stream, using another command:
```
$  ros2 run realsense2_camera realsense2_camera_node --ros-args -p filters:=colorizer
[INFO] [1623179195.228057467] [RealSenseCameraNode]: RealSense ROS v3.1.4
[INFO] [1623179195.228106022] [RealSenseCameraNode]: Built with LibRealSense v2.44.0
[INFO] [1623179195.228111829] [RealSenseCameraNode]: Running with LibRealSense v2.44.0
[INFO] [1623179195.287420500] [RealSenseCameraNode]: Device with serial number 044322073299 was found.

[INFO] [1623179195.287460555] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video6 was found.
[INFO] [1623179195.287470670] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[INFO] [1623179195.287677704] [RealSenseCameraNode]: Device with port number 2-3 was found.
[INFO] [1623179195.287685306] [RealSenseCameraNode]: Device USB type: 3.2
[INFO] [1623179195.288490127] [RealSenseCameraNode]: getParameters...
[INFO] [1623179195.288580622] [RealSenseCameraNode]: 707
[INFO] [1623179195.289795979] [RealSenseCameraNode]: setupDevice...
[INFO] [1623179195.289805581] [RealSenseCameraNode]: JSON file is not provided
[INFO] [1623179195.289811304] [RealSenseCameraNode]: Device Name: Intel RealSense D435I
[INFO] [1623179195.289815522] [RealSenseCameraNode]: Device Serial No: 044322073299
[INFO] [1623179195.289819633] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video6
[INFO] [1623179195.289825518] [RealSenseCameraNode]: Device FW version: 05.12.13.50
[INFO] [1623179195.289828822] [RealSenseCameraNode]: Device Product ID: 0x0B3A
[INFO] [1623179195.289831763] [RealSenseCameraNode]: Enable PointCloud: Off
[INFO] [1623179195.289834830] [RealSenseCameraNode]: Align Depth: Off
[INFO] [1623179195.289837893] [RealSenseCameraNode]: Sync Mode: On
[INFO] [1623179195.289845136] [RealSenseCameraNode]: Device Sensors:
[INFO] [1623179195.291641593] [RealSenseCameraNode]: Stereo Module was found.
[INFO] [1623179195.296481399] [RealSenseCameraNode]: RGB Camera was found.
[INFO] [1623179195.296675955] [RealSenseCameraNode]: Motion Module was found.
[INFO] [1623179195.296685981] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1623179195.296690513] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1623179195.296693890] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[INFO] [1623179195.296697196] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[INFO] [1623179195.296700800] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1623179195.296704474] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[INFO] [1623179195.296709997] [RealSenseCameraNode]: Add Filter: colorizer
[INFO] [1623179195.297928884] [RealSenseCameraNode]: num_filters: 1
[INFO] [1623179195.297941067] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[WARN] [1623179195.425572245] [RealSenseCameraNode]: Failed to set parameter:stereo_module.auto_exposure_limit = 0[1, 165000]
parameter 'stereo_module.auto_exposure_limit' could not be set: Parameter {} doesn't comply with integer range.
[WARN] [1623179195.425836481] [RealSenseCameraNode]: Failed to set parameter:stereo_module.auto_gain_limit = 0[16, 248]
parameter 'stereo_module.auto_gain_limit' could not be set: Parameter {} doesn't comply with integer range.
[INFO] [1623179195.427604601] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[INFO] [1623179195.428058047] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[INFO] [1623179195.428198672] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[INFO] [1623179195.428375490] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[INFO] [1623179195.428776451] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[INFO] [1623179195.428793970] [RealSenseCameraNode]: gyro stream is enabled - fps: 400
[INFO] [1623179195.428827558] [RealSenseCameraNode]: accel stream is enabled - fps: 250
[INFO] [1623179195.428834384] [RealSenseCameraNode]: setupPublishers...
[INFO] [1623179195.431282338] [RealSenseCameraNode]: setupStreams...
[WARN] [1623179195.535011573] [RealSenseCameraNode]:
 08/06 19:06:35,533 WARNING [140509551982336] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 08/06 19:06:35,534 WARNING [140508982638336] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.5/0003:8086:0B3A.0004/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable
[INFO] [1623179195.585973268] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[INFO] [1623179195.586974531] [RealSenseCameraNode]: RealSense Node Is Up!

```
Here interesting to note that the warning is misleading and even though the stream is enabled it keeps coming.
```
[INFO] [1623179195.296685981] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
```
[... cut ...]
[INFO] [1623179195.428198672] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[INFO] [1623179195.428375490] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
```

So, to summarize:
```ros2 run realsense2_camera realsense2_camera_node --ros-args -p filters:=colorizer```  --> works

``` ros2 launch realsense2_camera rs_launch.py  infra_rgb:=true enable_infra1:=true enable_infra2:=true ```   --> not working

Can you please help me figure out what is wrong with rs_launch file or which exact param needed here ?

MartyG-RealSense, 2021-06-08:
In regard to your use of the **infra_rgb** condition, the ROS2 wrapper documentation states "This feature is only supported by Realsense sensors with RGB streams available from the infra cameras".  The D435i does not support RGB from the left infrared sensor as the left and right IR sensors are both monochrome.  A D415 or D455 is required to obtain color from left IR.

aagrawa3-dev, 2021-06-08:
OK Thanks.
After a long trial and error I got this list of inter-dependent params which I need to either pass on cmd line or just enable them inside rs_launch.py.

**ros2 launch realsense2_camera rs_launch.py depth_width:=640 depth_height:=480 depth_fps:=30.0 infra_width:=640 infra_height:=480 infra_fps:=30.0 infra_rgb:=false enable_sync:=true align_depth:=true**

The fps, height and width parameters can be changed based on possible values in rs-enumerate-devices.

 
Question:

1. 

filters:='colorizer' can not be used (reason unknown) with above cmd, it will stop the display-stream of depth image. Although from launch console all looks healthy. Could you please investigate why?

 

1. 

align_depth - this should actually project depth view with rgb image and infra-image, but seems not working as expected. Can you please cross check the behavior? - depth-align-raw-image, depth-aligned-infra-image

MartyG-RealSense, 2021-06-09:
1.  Advice about using filters with rs_launch.py in the ROS2 wrapper is provided in the link below.  It suggests to define **colorizer** without quotation marks around it.

**ros2 launch realsense2_camera rs_launch.py filters:=temporal,colorizer depth_qos:=SYSTEM_DEFAULT**

https://github.com/IntelRealSense/realsense-ros/issues/1710#issuecomment-782829843

2.  The ROS2 wrapper documentation states that using **align_depth** will publish an aligned depth to color topic **/camera/aligned_depth_to_color/image_raw** - it does not mention infrared.

aagrawa3-dev, 2021-06-09:
I tried colorizer filter without ' ', and result is as below:
Command: 
```
s$ ros2 launch realsense2_camera rs_launch.py   depth_width:=640 depth_height:=480 depth_fps:=30.0 infra_width:=640 infra_height:=480 infra_fps:=30.0  infra_rgb:=false enable_sync:=true align_depth:=true filters:=colorizer
[INFO] [launch]: All log files can be found below /home/aagrawa3/.ros/log/2021-06-09-10-23-47-249901-demohost-124
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [126]
[realsense2_camera_node-1] [INFO] [1623234227.856380199] [RealSenseCameraNode]: RealSense ROS v3.1.4
[realsense2_camera_node-1] [INFO] [1623234227.856416476] [RealSenseCameraNode]: Built with LibRealSense v2.44.0
[realsense2_camera_node-1] [INFO] [1623234227.856422519] [RealSenseCameraNode]: Running with LibRealSense v2.44.0
[realsense2_camera_node-1] [INFO] [1623234227.917810581] [RealSenseCameraNode]: Device with serial number 044322073299 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1623234227.917891326] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video6 was found.
[realsense2_camera_node-1] [INFO] [1623234227.917920241] [RealSenseCameraNode]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO] [1623234227.918728185] [RealSenseCameraNode]: Device with port number 2-3 was found.
[realsense2_camera_node-1] [INFO] [1623234227.918770929] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1623234227.922241938] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1623234227.922614339] [RealSenseCameraNode]: 707
[realsense2_camera_node-1] [INFO] [1623234227.923801918] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1623234227.923812605] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1623234227.923817975] [RealSenseCameraNode]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO] [1623234227.923821583] [RealSenseCameraNode]: Device Serial No: 044322073299
[realsense2_camera_node-1] [INFO] [1623234227.923824821] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video6
[realsense2_camera_node-1] [INFO] [1623234227.923828562] [RealSenseCameraNode]: Device FW version: 05.12.13.50
[realsense2_camera_node-1] [INFO] [1623234227.923831845] [RealSenseCameraNode]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO] [1623234227.923834943] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1623234227.923840546] [RealSenseCameraNode]: Align Depth: On
[realsense2_camera_node-1] [INFO] [1623234227.923844374] [RealSenseCameraNode]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1623234227.923852623] [RealSenseCameraNode]: Device Sensors:
[realsense2_camera_node-1] [INFO] [1623234227.925636137] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1] [INFO] [1623234227.930525331] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1623234227.930653648] [RealSenseCameraNode]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1623234227.930663068] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1623234227.930667697] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1623234227.930671276] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1623234227.930674639] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1623234227.930678808] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1623234227.930682896] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1623234227.930689589] [RealSenseCameraNode]: Add Filter: colorizer
[realsense2_camera_node-1] [INFO] [1623234227.931060983] [RealSenseCameraNode]: num_filters: 1
[realsense2_camera_node-1] [INFO] [1623234227.931068883] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1623234227.991040099] [RealSenseCameraNode]: Failed to set parameter:stereo_module.auto_exposure_limit = 0[1, 165000]
[realsense2_camera_node-1] parameter 'stereo_module.auto_exposure_limit' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1623234227.991540966] [RealSenseCameraNode]: Failed to set parameter:stereo_module.auto_gain_limit = 0[16, 248]
[realsense2_camera_node-1] parameter 'stereo_module.auto_gain_limit' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1623234227.992849024] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1623234227.993263356] [RealSenseCameraNode]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1623234227.993377461] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[... cut ...]
[realsense2_camera_node-1] [INFO] [1623234669.205364651] [RealSenseCameraNode]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1623234669.325136142] [RealSenseCameraNode]:

```

Output:

[image]

Now if I remove infra settings from cmd line, then colorizer filter starts working. So clearly infra topics have issue with colorizer filter.

Command:
```
$ ros2 launch realsense2_camera rs_launch.py   depth_width:=640 depth_height:=480 depth_fps:=30.0 infra_rgb:=false enable_sync:=true align_depth:=true filters:=colorizer 
```
Output:
[image]

MartyG-RealSense, 2021-06-09:
Looking at the RViz image directly above and comparing the order of the left-side images to the image descriptions on the right side, it looks as though the topics **aligned_depth_to_infra1** and **Infra2/image_rect_raw** are not being published in RViz (which would be consistent with the infra topics being removed from the roslaunch instruction in that particular test, especially as they appear in the second to last image where the infra topics were enabled in roslaunch).

I will seek the advice of @doronhi the RealSense ROS wrapper developer on this particular question.

Hi @doronhi  Are there known circumstances where the infra topics may stop working if a colorizer filter is applied?  Thanks!

aagrawa3-dev, 2021-06-10:
Also a room for improvement:
Below 2 in single run do not make sense, or at least confuse the user:
```
...
[RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
...
[RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
..
```

doronhi, 2021-06-13:
Those lines do make sense:
The first one says that Infra-**0** is not supported (This sensor is available for L515 for instance).
The second one says that Infra**1** is enabled.

This is confusing, I agree. It could be that the wrapper should not relate to sensors the current device does not have and the user didn't ask for.

aagrawa3-dev, 2021-06-14:
> Those lines do make sense:
> The first one says that Infra-**0** is not supported (This sensor is available for L515 for instance).
> The second one says that Infra**1** is enabled.
>
> This is confusing, [...]

Thanks for reply. Could you also comment on below question or how can I enable infra with colorizer filter: 
Hi @doronhi Are there known circumstances on the ROS2 wrapper where the colorizer filter may stop working if the infra topics are enabled? Thanks!

doronhi, 2021-06-15:
I am not aware of any connection between the infra topics and the colorizer filter which works on the depth images.

aagrawa3-dev, 2021-06-18:
Thanks for info. Closing this issue.

closed 2021-06-18 by aagrawa3-dev