realsenseai/realsense-ros #1921: Noisy vertical lines infra image

fiorano10, 2021-06-07:
We're running 4 cameras connected to 4 separate USBs, running librealsense v2.42, ros v2.22. Occasionally  see the camera load up like this, and the only way to fix this is to relaunch the cameras.
[image]

MartyG-RealSense, 2021-06-07:
Hi @fiorano10  When you say that the problem occurs when the camera "loads up", do you mean that it occurs immediately after launch please?  If so, does adding **initial_reset:=true** to the end of the roslaunch instruction to reset the camera at launch make a positive difference?

If the distortion occurs randonmly at some point after launch: there was a librealsense case involving multiple cameras this year when the infrared image would become noisy and the depth image break up at random times during both the day and night.  The RealSense user in that case found that problem no longer occurred if the official USB cables supplied with the camera were used instead of their own choice of cable.  Are you using the official 1 meter cables please?

https://github.com/IntelRealSense/librealsense/issues/8880

MartyG-RealSense, 2021-06-14:
Hi @fiorano10  Could you provide an update regarding the previous advice given in the comment above please?  Thanks!

closed 2021-06-28 by MartyG-RealSense