realsenseai/realsense-ros #1925: Downgread librealsense version from v2.47.0 to v2.45.0

ranjitkathiriya, 2021-06-10:
Hello there,

My PC has auto-upgraded to v2.47.0 librealsense version and I am getting an error when I am executing realsense camera.

Error:
```
roslaunch realsense2_camera rs_camera.launch 
... logging to /home/kuka-ai/.ros/log/33b1364a-c9f7-11eb-9790-6d260f2a1ae8/roslaunch-kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU-46428.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:44837/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [46451]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 33b1364a-c9f7-11eb-9790-6d260f2a1ae8
process[rosout-1]: started with pid [46476]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [46483]
process[camera/realsense2_camera-3]: started with pid [46484]
[ INFO] [1623334914.012416947]: Initializing nodelet with 16 worker threads.
[ERROR] [1623334914.059900790]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library /home/kuka-ai/ai_catkin/devel/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.45: cannot open shared object file: No such file or directory)
[ERROR] [1623334914.060002689]: The error before refreshing the cache was: Failed to load library /home/kuka-ai/ai_catkin/devel/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.45: cannot open shared object file: No such file or directory)
[FATAL] [1623334914.060557933]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 46484, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/kuka-ai/.ros/log/33b1364a-c9f7-11eb-9790-6d260f2a1ae8/camera-realsense2_camera-3.log].
log file: /home/kuka-ai/.ros/log/33b1364a-c9f7-11eb-9790-6d260f2a1ae8/camera-realsense2_camera-3*.log
```

MartyG-RealSense, 2021-06-10:
Hi @ranjitkathiriya   That is an unusual error.  Typically, if the librealsense version has been upgraded without the RealSense ROS wrapper being rebuilt afterwards then I would expect the log to say "Built with 2.45.0, running with 2.47.0".

If librealsense 2.47.0 was installed with Debian packages then you could initiate a **purge** to uninstall all librealsense related packages.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

ranjitkathiriya, 2021-06-10:
`sudo apt-get purge librealsense2` Yes I did uninstall. 

I am installing it from this link https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md, how can I install the 2.45.0 version only. I tried installing from this link every time it is installing the new version.

I also tried these steps for installing v2.45.0

If you were using the Debian packages then follow the installation guide to remove the current version, then re-install Librealsense SDK using an explicit version/build:

- List all the available versions with apt-cache showpkg librealsense2

- Select and install the specific version from the list. For instance, to install the demos run:

sudo apt install librealsense2=2.45.0-0~realsense0.4552 librealsense2-gl=2.45.0-0~realsense0.4552 librealsense2-utils=2.45.0-0~realsense0.4552

- *Make sure to replace tbd with the actual build number.

While installing the utils package I am getting this error

```
> sudo apt install librealsense2-utils=2.45.0-0~realsense0.4552
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been mo
ved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-utils : Depends: librealsense2 (= 2.45.0-0~realsense0.4552) but 2.47.0-0~realsense0.4806 is to be installed
                       Depends: librealsense2-gl (= 2.45.0-0~realsense0.4552) but 2.47.0-0~realsense0.4806 is to be installed
                       Depends: librealsense2-net (= 2.45.0-0~realsense0.4552) but 2.47.0-0~realsense0.4806 is to be installed
E: Unable to correct problems, you have held broken packages.
```

MartyG-RealSense, 2021-06-10:
Assuming that you are starting at the beginning of the package installation instructions after having purged the previos packages:

If you have registered the server key and added the server to the list of repositories, have you then installed the **librealsense2-dkms** package?

ranjitkathiriya, 2021-06-10:
> Assuming that you are starting at the beginning of the package installation instructions after having purged the previos packages:

Yes, I have followed the steps from the beginning. 

> have you then installed the librealsense2-dkms package?

Yes, I have installed it but It is taking the latest version 2.47.0.

```
(ros1_envs) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/ai_catkin$ sudo apt-get install librealsense2-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  libglfw3
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  librealsense2-udev-rules
The following NEW packages will be installed:
  librealsense2-dkms librealsense2-udev-rules
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/4,282 kB of archives.
After this operation, 23.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Selecting previously unselected package librealsense2-dkms.
(Reading database ... 339950 files and directories currently installed.)
Preparing to unpack .../librealsense2-dkms_1.3.18-0ubuntu1_all.deb ...
Unpacking librealsense2-dkms (1.3.18-0ubuntu1) ...
Selecting previously unselected package librealsense2-udev-rules:amd64.
Preparing to unpack .../librealsense2-udev-rules_2.47.0-0~realsense0.4806_amd64.deb ...
Unpacking librealsense2-udev-rules:amd64 (2.47.0-0~realsense0.4806) ...
Setting up librealsense2-udev-rules:amd64 (2.47.0-0~realsense0.4806) ...
Setting up librealsense2-dkms (1.3.18-0ubuntu1) ...
Loading new librealsense2-dkms-1.3.18 DKMS files...
Building for 5.8.0-55-generic
```

MartyG-RealSense, 2021-06-10:
As a new RealSense ROS wrapper version tends to be released quite soon after a new librealsense version, maybe you could wait for a wrapper that is compatible with 2.47.0 to be released by Doronhi the RealSense ROS wrapper developer and then build that wrapper on top of 2.47.0.

doronhi, 2021-06-13:
Hi,
A new realsense2_camera will be out next week, hopefully, but debian packages for a new release tend to be delayed some more by ros-distro sync schedule. Since you have the librealsense2 version 2.47.0 debain I assume you built the realsense2_wrapper itself from source and won't be waiting for the debian package anyway.

I am not aware of any relevant API changes in the new librealsense2 2.47.0 version so rebuilding realsense2_camera after installing librealsense2 version 2.47.0 should work fine
(i.e. 
```
cd catkin_ws
rm -rf build/ install/
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
```
)

If you wish to revert back to librealsense2 v2.45.0, a solution to the issue you are having is manually installing the dependencies to the correct version. That is:
```
sudo apt install librealsense2=2.45.0-0~realsense0.4552
sudo apt install librealsense2-gl=2.45.0-0~realsense0.4552
sudo apt install librealsense2-net=2.45.0-0~realsense0.4552
sudo apt install librealsense2-utils=2.45.0-0~realsense0.4552
```
And maybe some more dependencies, if you more error message starting with `The following packages have unmet dependencies:`

ranjitkathiriya, 2021-06-14:
I tried to revert back but I am getting these errors.

```
kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~$ sudo apt install librealsense2=2.45.0-0~realsense0.4552
[sudo] password for kuka-ai: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2
kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~$ sudo apt install librealsense2-gl=2.45.0-0~realsense0.4552
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-gl
kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~$ sudo apt install librealsense2-net=2.45.0-0~realsense0.4552
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-net
kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~$ sudo apt install librealsense2-utils=2.45.0-0~realsense0.4552
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-utils
kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~$
```

weizhuowang, 2021-06-14:
Replace 4552 with 4551.

ranjitkathiriya, 2021-06-14:
Same Error and 551 is for melodic afaik,
```
kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~$ sudo apt install librealsense2-utils=2.45.0-0~realsense0.4551
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package librealsense2-utils
```

sudo apt install librealsense2=2.45.0-0~realsense0.4552 librealsense2-gl=2.45.0-0~realsense0.4552 librealsense2-utils=2.45.0-0~realsense0.4552

MartyG-RealSense, 2021-06-14:
Thanks very much for the update @ranjitkathiriya - if you are going to wait for a week then I will keep this case open for a further time period.   Good luck!

garcia-g, 2021-06-18:
@MartyG-RealSense thank you for your support.

Have you considered a more permanent solution to this issue? This is a recurring problem when the versions change and these 1-week breaks are a drawback for system stability.

We would like to avoid building from source since there is a deb package or having to create an apt caching server just to solve this so please consider it seriously.

Can you time better the releases or you find a way to keep the previous versions?

MartyG-RealSense, 2021-06-18:
Hi @garcia-g  A suitable solution for you may be to build using the ROS wrapper's **Method 1** if possible, so that a matching librealsense and ROS wrapper version are installed from packages together at the same time instead of installing librealsense and the ROs wrapper separately.  

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

Whilst using Method 1 means that you would likely not be installing the very latest librealsense version but it should ensure that your SDK and wrapper versions are always synced.

ranjitkathiriya, 2021-06-18:
Hello @garcia-g,

I am building for source because I am using a real sense-viewer to visualize depth and based on that I am implementing it to an algorithm. If you are not using real sense viewer as I do. Then I would suggest directly follow method one as suggested by @MartyG-RealSense.

Advantages you will get if you are not building from the source.

wasteful of time (because you have to do many more things than simply apt-get install),
wasteful of resources (as your computer now compiles something which was already compiled for you), and
wasteful of effort (as users tend to run into problems and then post new questions about how to build packages from source)

garcia-g, 2021-06-20:
Thank you @MartyG-RealSense and @ranjitkathiriya for your kind advice.

I will try the ROS distribution (Method 1) but that doesn't solve my problem entirely. The issue remains as I will have no control on the updates since the ROS distribution also removes the packages once the latest version is released, which, if unlucky, will render my system unusable. Since we use docker and CI, we can survive but it's not the ideal situation. In docker, we use pinned versions of the packages for stability, so not having the previous versions will cause the CI to fail.

I'm wondering if someone else uses docker with pinned versions and how they solve these issues 🤔 

About building the code, @ranjitkathiriya you're right: it's too much waste for something that should be solved more easily.

MartyG-RealSense, 2021-06-20:
You are very welcome @garcia-g 

I can recall a past case in which someone needed control over when and how updates occurred.    It was suggested that a custom **fork** of the official RealSense GitHub was created on the person's own GitHub account.

[image]

doronhi, 2021-06-21:
Sorry to update that due to some delays and since librealsense2 v2.48.0 is expected very soon it is possible that the next realsense2_camera version will be delayed by a couple of days and maybe skip librealsense2 v2.47.0.

MartyG-RealSense, 2021-06-21:
Thanks very much @doronhi for the information about the release schedule.

garcia-g, 2021-06-21:
> You are very welcome @garcia-g
> 
> I can recall a past case in which someone needed control over when and how updates occurred. It was suggested that a custom **fork** of the official RealSense GitHub was created on the person's own GitHub account.
> 
> [image]

@MartyG-RealSense Thank you for the tip, I'll consider that option too.

MartyG-RealSense, 2021-06-27:
Hi @ranjitkathiriya  Do you require further assistance with this case, please?  Thanks!

ranjitkathiriya, 2021-06-28:
No! thanks.

closed 2021-06-28 by ranjitkathiriya

MartyG-RealSense, 2021-06-28:
Thanks very much @ranjitkathiriya for the update!

ranjitkathiriya, 2021-06-28:
> if you are going to wait for a week then I will keep this case open for a further time period. Good luck!

Actually, I thought to keep this case open, but now it's okay!